betaflight/betaflight #9762: Anti Gravity GUI update

selimmch, 2020-05-03:
**Describe the bug**
The Betaflight 10.7 configurator is not updating the Anti Gravity gain in the GUI when changed by hand, while it is actually changing the Gain in the CLI.

**To Reproduce**
Change the gain value in the PID tab then hit save. The GUI is not updated. Go to the CLI and get the Anti Gravity gain value, it is set to the value you had written.

**Expected behavior**
The configurator must update the GUI when the value set by hand is saved.

**Flight controller configuration**
 ``` 
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 29 2020 / 08:08:49 (b480103d4) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405STD, version: c4140206, date: 2020-05-02T03:17:47Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF405STD
manufacturer_id MTKS
mcu_id 00450029324d500720303256
signature 

# name: Morfeus

# feature
feature -SOFTSERIAL
feature -DYNAMIC_FILTER
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# serial
serial 0 1024 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 3 2 115200 57600 0 115200
serial 4 2048 115200 57600 0 115200

# master
set acc_calibration = 18,6,-30,1
set fpv_mix_degrees = 30
set serialrx_provider = CRSF
set dshot_bidir = ON
set yaw_motors_reversed = ON
set small_angle = 180
set gyro_1_align_yaw = 1800
set name = Morfeus

profile 0

# profile 0
set iterm_relax_cutoff = 8
set p_pitch = 48
set i_pitch = 45
set d_pitch = 50
set f_pitch = 130
set p_roll = 52
set i_roll = 47
set d_roll = 50
set f_roll = 130
set p_yaw = 52
set i_yaw = 70
set d_min_roll = 44
set d_min_pitch = 37

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 100
set thr_expo = 20
set roll_expo = 20
set pitch_expo = 20
set roll_srate = 55
set pitch_srate = 55
set yaw_srate = 60
set tpa_breakpoint = 1400

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: I2C_SCL 3
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: SERIAL_TX 5
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

**Setup / Versions**
MATEK F405 STD

**Additional context**
Add any other context about the problem here.

Asizon, 2020-05-04:
I cant reproduce it, works fine for me. What values are you trying to put? You diff doesnt show these configured values, also, did you make a clean flashing with full chip erase? And what is you SO??

selimmch, 2020-05-04:
I'm on Windows 10 and I just did another clean flashing with full chip erase, the problem remains.

I put a gain of 2 and when I hit save, the display switched to 3.5. Here is a screen of the PID tab and the CLI.

[image]

[image]

Asizon, 2020-05-04:
Thanks and good catch. Definitely this function breaks when you try to put a value less than default 3.5. Fixing now.

selimmch, 2020-05-04:
You're welcome