betaflight/betaflight #9763: GPS and Battery Level issue

Gerardo961, 2020-05-03:
** I have just switched to BF4.2. and Everytime I arm my two quads it shows that more than half battery has been drained even though it is fully charged. It wasn't happening in BF4.1.   Also it doesn't displays the GPS info at first arm. I have to reboot the FC one or two times from the OSD  and then the GPS info get displayed in the OSD correctly. This doesn't happen in BF 4.1 
**

diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Apr 29 2020 / 08:09:54 (b480103d4) MSP API: 1.43
# config: manufacturer_id: IFRC, board_name: IFLIGHT_F722_TWING, version: bc19b7dc, date: 2020-04-01T04:37:12Z

# start the command batch
batch start

board_name IFLIGHT_F722_TWING
manufacturer_id IFRC

# name: Armattan

# feature
feature -SOFTSERIAL
feature GPS
feature TELEMETRY

# map
map EATR1234

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 4 2 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 46 1 1800 2100 0 0
aux 2 26 0 1800 2100 0 0
aux 3 35 2 1775 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745    0
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E U FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY    0    0 5732 5769 5806 5843    0    0
vtxtable powerlevels 4
vtxtable powervalues 14 20 26 30
vtxtable powerlabels 25 100 400 1.0

# master
set gyro_lowpass2_hz = 375
set gyro_notch1_hz = 200
set gyro_notch1_cutoff = 180
set gyro_notch2_hz = 200
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 300
set dyn_lpf_gyro_max_hz = 750
set acc_calibration = -33,-33,43,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_scale = 200
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 15
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 420
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_auto_baud = ON
set gps_rescue_initial_alt = 40
set gps_rescue_descent_dist = 100
set gps_rescue_ascend_rate = 400
set gps_rescue_min_sats = 6
set pid_process_denom = 2
set osd_warn_rssi = ON
set osd_vbat_pos = 2454
set osd_rssi_pos = 2265
set osd_vtx_channel_pos = 2099
set osd_ah_pos = 2126
set osd_gps_lon_pos = 2081
set osd_gps_lat_pos = 2113
set osd_gps_sats_pos = 2243
set osd_home_dir_pos = 2318
set osd_warnings_pos = 14729
set debug_mode = GYRO_SCALED
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_freq = 5740
set gyro_2_align_yaw = 900
set name = Armattan

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 255
set dterm_lowpass2_hz = 225
set anti_gravity_gain = 7000
set iterm_rotation = ON
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 51
set i_pitch = 95
set d_pitch = 42
set p_roll = 55
set i_roll = 88
set d_roll = 44
set p_yaw = 62
set i_yaw = 92
set d_min_roll = 24
set d_min_boost_gain = 40

rateprofile 0

# end the command batch
batch end

# 

FC IFLIGHT_F722_TWING v2.1 (Succex F7 V2.1)
VTX Unify PRO 32 HV

[image]

zigomatichub, 2020-05-04:
In BF4.2 RC1, the HOT start seems NOK (Hot start is after the first battery and maintened by a small battery on your GPS board).
IN BF: At Hot start, GPS stay at 0 and blincking with La and Lo value other then 0.00000.
Sometimes blinking with between 0 and other GPS sat value. 
Sometimes blinking 0 SAT during till battery is dead or removed.
It's a sporadic issue. 
My workaround is:
After put the battery, wait 5 sec to see the issue then go to smartaudio, just rebooting the FC (save and exit without change anything).  At reboot GPS appear and directly I can start to fly (my min sat value set at 6 instead of 8).
This problem appear starting with BF4.2 RC1 with a Matek F722-SE and GPS Matek http://www.mateksys.com/?portfolio=m8q-5883

Seems link to you issue as well.

Gerardo961, 2020-05-04:
Hello!,
That's exactly the issue I have.
When first plugging the battery there isn't any GPS icon in the OSD. Is all gone. I have to reboot the FC as you mentioned and then all appears ok.

The second issue is that lipos always show up as half depleted when they are not. :(

Regards

mikeller, 2020-05-04:
@Gerardo961: Looks like your voltage meter calibration is off - recalibrate the voltage meter: https://github.com/betaflight/betaflight/blob/master/docs/Battery.md#configuration

hdcrb, 2020-05-04:
I have the same issue with no gps info being shown, but it's resolved by arming and disarming in my case.

tn2009, 2020-05-04:
Same issue with GPS. I solved this problem like this. on the ports tab, set the port speed for GPS to 57600. And GPS works perfectly no reboots from OSD or Configurator are required.

Gerardo961, 2020-05-04:
I already have it in 57600 and same issue still persist. I need to reboot from OSD in order for the GPS to show up.

tn2009, 2020-05-04:
> I already have it in 57600 and same issue still persist. I need to reboot from OSD in order for the GPS to show up. Enviado desde Outlook Mobile<https://aka.ms/blhgte>
> […](#)

there is a bug with GPS in 4.2 RC1.

ggmaniack, 2020-05-04:
@Gerardo961 To fix the battery issue, raise the Max Cell Voltage (& perhaps calibrate the voltage meter). If your battery slightly exceeds it, BF will think that it is a battery with one more cell.

zigomatichub, 2020-05-04:
Checked goggles recording,  I see quiet some issues with the GPS but difficult for me to say something about that as I don't know how GPS is managed in BF (blinking, home point etc) .
I can fly with 5 GPS number blinking but gps_rescue_min_sats = 6 which is weird.
GPS baud is on Auto on configurator in configuration tab while GPS Baud are set to 57600 in ports Tab. 
About battery, I didn't notice issues till now.

comsoft1908, 2020-05-06:
Had the same issue since moving to BF4.2 RC1 - I assume the problem is the auto baud detection.
I have 3 BN220 and all were set to different baud rates from the factory. 

What worked for me: set the baud rate to 57600 via u-center, save the config with u-center. Go into betaflight configurator: disable auto-baud, enabled auto-config and set baudrate to 57600 in ports tab.

This way betaflight does not need to guess the baud rate, which fixes the problem.

DarkSoulRL, 2020-05-09:
Same problem. Matek m8q. Not always detected. Rebooting does not always help. The barometer also sometimes disappears. And several times it was in flight. After flashing stable 4.1.6 All devices are detected at startup and work stably.

zigomatichub, 2020-05-09:
I think one workaround is to put Off Baudrate AUTO in configuration tab and put correct value in Port tab. cfr the info upper

Gerardo961, 2020-05-09:
Hello Guys,
Here is a Video of my issue. You can see how I have to reboot the FC many times in order for the GPS to show up in the OSD. 
https://youtu.be/p1J5bzGDuHE

hdcrb, 2020-05-09:
fixed baud rate did not fix this for me at least

closed 2020-06-17 by stale[bot]