betaflight/betaflight #9766: Runaway Takeoff Prevention Happening Only On Bidirect Dshot

baileyfpv, 2020-05-04:
**Describe the bug**
When using bidirectional dshot with the right looptimes (4k/4k and dshot 300) when I give a half throttle punch with my quad it pulls up the error runaway and the quad does endless backflips into the ground until I disarm.

**To Reproduce**
not sure if its my hardware or not, I cant use rpm filtering, the error doesn't happen with is disabled

**Expected behavior**
during a throttle punch of around half throttle causes it to trigger runaway.

**Flight controller configuration**
```Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
Matek F405 MK2 Bought from Kiwiquads NZ
TBS Source One V3 Frame
TBS Unify Pro HV
Runcam Swift 2
SPEDIX GS35-4 4IN1 35A 3-4S ESC DHOT 1200 BLHELI32 - running the latest rev.
dogcom 1300 4s batts
latest betaflight revision, I can confirm that different versions of 4.x aren't the issue

**Additional context**
I know there was a Issue on it awhile back but it doesn't resolve the issue for me,

And a minor this is that only one motor is reversed in BLHELI 32 and I have to remap the motors because the esc is facing sideways so my XT60 is coming out the side, not the back

baileyfpv, 2020-05-04:
It didnt let me include my BBL but here is a csv of one, and a .zip of two bbl, "this" and "this one"

[this.BBL.csv.zip](https://github.com/betaflight/betaflight/files/4577898/this.BBL.csv.zip)
[Archive.zip](https://github.com/betaflight/betaflight/files/4577905/Archive.zip)

baileyfpv, 2020-05-05:
@mikeller whats the command for the betaflight dump, I deleted it and i cant find it

mikeller, 2020-05-05:
@baileyfpv: `diff`

closed 2020-05-05 by baileyfpv