betaflight/betaflight #9768: Redpine protocol does not bind with 4.2 RC1

schugabe, 2020-05-05:
**Describe the bug**
I tried to bind 4.2 RC1 with the redpine protocol with a multiprotocol module. As far as I can tell the board enters the bind mode (green led solid) but the bind does never finish.

If I switch the quad+module to D16 the bind works.

**To Reproduce**
1. Select redpine as protocol on the configuration tab
2. Go to receiver tab + press bind button (or use cli)
3. Configure redpine protocol on the transmitter and start bind
4. Bind does not finish

**Setup / Versions**
- FC is from betafpv with MATEKF411RX target
- Transmitter is Frsky qx7s with opentx 2.3.7
- Multiprotocol module irangex irx4 plus flashed with firmware 1.3.0.91

brycedjohnson, 2020-05-18:
Hey @schugabe Let me see if I can repo with my multimodule and betaflight with those versions.

isokasakissoja, 2020-06-06:
Having the same issue on BF 4.2RC4, CrazybeeF4FR, and Jumper T8SG V2.

brycedjohnson, 2020-06-09:
@schugabe, slow for very late reply.  Can you try and bind in fast mode?  I switched to only bind in fast mode and the TX should switch to fast during bind.   But apparently with with a matek411 (crazybee board) on RC6 and 1.3.0.91 in fast mode it binds.  In slow more it does not.  I may have missed a checkin on multimodule to fix that.

brycedjohnson, 2020-06-09:
@isokasakissoja Which deviation version are you running on the Jumper?  

My pull request never actually got into the main branch, so I'll have to bug those guys to get it pulled in and I'll do a custom build with the latest deviation changes as well.

isokasakissoja, 2020-06-09:
@brycedjohnson I'm using the test version of deviation found on your RCGroups thread. And if I was unclear, I meant the board lights on the Crazybee flight controller behaving differently on redpine. I also tried binding on both fast and slow mode, with proper frequency tuning, etc.

schugabe, 2020-06-09:
@brycedjohnson I can confirm that binding in fast mode works but slow does not work. I know that I tried both variants before and none worked. Maybe there was a different issue...

But now I tried to bind in fast mode multiple times and worked every time. But never in slow mode.

Do you think there is something to do in the betaflight code or on the tx side? Otherwise I'd close this ticket.

closed 2020-07-18 by stale[bot]