betaflight/betaflight #9769: Remaping bug

LastRizes, 2020-05-05:
**Describe the bug**
Remaping does not work

**To Reproduce**
set the signal cable to tx 1.resource SERIAL_TX 1 NONE.  Remap the motor 2 to A09

**Expected behavior**
Working esc

**Flight controller configuration**
 ``` # version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 Apr 29 2020 / 08:09:54 (b480103d4) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722
manufacturer_id MTKS

# resources
resource MOTOR 2 A09
resource SERIAL_TX 1 NONE

# master
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

```
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: ADC_BATT
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: FREE
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Matek F722-STD, banggood
 - Jumpter T1 receiver, AKK FX2 Ultimate mini, runcam eagle 2 pro, ;
 - VTX to PDB, signal vtx to signal of FC, CAM to cam at the FC, Receiver to 2 RX, 2-nd esc to TX1.

**Additional context**
Add any other context about the problem here.

mikeller, 2020-05-05:
According to the resource list that you posted motor 2 is mapped to C07 - connect it there any it will work.

LastRizes, 2020-05-05:
C07 is signal port? But my signal port is broken

LastRizes, 2020-05-05:
> According to the resource list that you posted motor 2 is mapped to C07 - connect it there any it will work.

BTW, everything worked in the last version of the BF(4.1.6 and later).

mikeller, 2020-05-06:
@LastRizes: But motor 2 is mapped to C07, so that's where you have to connect to if you want to use it:

> C07: MOTOR 2

LastRizes, 2020-05-06:
> @LastRizes: But motor 2 is mapped to C07, so that's where you have to connect to if you want to use it:
> 
> > C07: MOTOR 2

Yeah, I can see that, but I need to put it on A09.

mikeller, 2020-05-06:
@LastRizes: `resource motor 2 a09`, and an appropriate definition for timer and DMA should do that.

jflyper, 2020-05-06:
@LastRizes 
It is NOT a bug; just saying you want to use PA9 as motor output does not make it work. Like @mikeller said, you have to assign a timer channel to PA9.

Try this in CLI
```
timer a9 af1
resource serial_tx 1 none
resource motor 2 a9
```

closed 2020-05-11 by LastRizes