betaflight/betaflight #9770: Iflight F7 and F4 Board can't handle 4.1 - drone flips - deathroll - uncontrolled yaw

rashmandashfpv, 2020-05-05:
1st FC: F4 of the Nazgul5 series
FW: 4.0 - drone flies fine -> same drone, no crash -> installed 4.1 -> drone flips when i apply throttle 
-> degraded back to 4.0 -> drone still flips.

2nd FC: F7 of the Cidora series
FW: 4.0 - drone flies fine -> same drone, no crash -> installed 4.1 -> drone flips when i apply throttle 
-> degraded back to 4.0 -> drone still flips.

3rd FC: iFlight SucceX-D F7 TwinG V2.1
FW: 4.0 - drone flies fine -> same drone, no crash -> FC swaped - FW 4.1 preinstalled -> drone flips when i apply throttle 
-> i saved me the time and do not degraded back to 4.0.
 
Any Ideas? 
If u need further information please ask!!!

etracer65, 2020-05-05:
Since you're the only one having this problem and repeatedly having it occur it would imply that you are doing something wrong.

Check with the manufacturer to ensure you're using the correct target(s).

Check that the movement of the 3D graphic on the Setup tab exactly matches the movement of the quad (move it around). If not, fix the board alignment settings.

REMOVE PROPS! On the motors tab spin each motor one at a time and make sure the position and the rotation match the graphic at the top. Correct any motors that are incorrect.

Make sure your props are on in the correct orientation and position.

This is not a setup/configuration support site. Recommended resources:
Betaflight Slack: https://slack.betaflight.com
Facebook Group: https://www.facebook.com/groups/betaflightgroup/

rashmandashfpv, 2020-05-05:
Thx for the fast Reply!!! REMOVE PROPS! On the motors tab spin each motor one at a time and make sure the position and the rotation match the graphic at the top. Correct any motors that are incorrect.                                That was it!!!!                 The UI dos not Change the direction of the propspin. BUT the UI spindirection must match the physical spindirection.                 -!!!-THANK YOU-!!!-                Gesendet von Mail für Windows 10 Von: Bruce LuckcuckGesendet: Dienstag, 5. Mai 2020 19:52An: betaflight/betaflightCc: rashmandashfpv; AuthorBetreff: Re: [betaflight/betaflight] Iflight F7 and F4 Board can't handle 4.1 - drone flips - deathroll - uncontrolled yaw (#9770) Since you're the only one having this problem and repeatedly having it occur it would imply that you are doing something wrong.Check with the manufacturer to ensure you're using the correct target(s).Check that the movement of the 3D graphic on the Setup tab exactly matches the movement of the quad (move it around). If not, fix the board alignment settings.REMOVE PROPS! On the motors tab spin each motor one at a time and make sure the position and the rotation match the graphic at the top. Correct any motors that are incorrect.Make sure your props are on in the correct orientation and position.This is not a setup/configuration support site. Recommended resources:Betaflight Slack: https://slack.betaflight.comFacebook Group: https://www.facebook.com/groups/betaflightgroup/

closed 2020-05-05 by etracer65