betaflight/betaflight #9772: SPI RX FRSKY WITH D16 EU-LBT NO WORK WITH JESC RPM FILTER

Zanth11, 2020-05-06:
Hello,
I have a mobula6 that flies well. I wanted to run ESC under JESC in 48khz and since the radio link cuts a few meters. If I switch back to BHeliS I have no problem. I tried in dshoot300 with gyro and loop at 4k, I always have rx loss randomly. 

I tried without rpm filter, I tried without bidirectional dshoot, the problem persists. Jesc does not work with spi rx d16 frsky in betaflight 4.1.1

My mobula6 is flash crazybeef4fr 4.1.1. I read on the forum that the flash of the esc in 48khz disturbed the signal of the spi rx in d16, the problem is it fixed in a more recent version of betaflight or will it be fixed in version 4.2?

Thank you all and sorry for the mistakes I am French and really bad in English. :)

MJ666, 2020-05-06:
If this is really interference of the 48kHz JESC switching frequency likely no update of the BF (including the SPI RX) firmware will help. The only thing which could help will be to change JESC switching frequency to something wich will not disturb. One thing could make the problem less is waving the motor wires and move the antenna more far away from ESCS but options are limited with the given size of the mobula.

Zanth11, 2020-05-06:
Ok, so if we want to use esc in 48khz we have to stay in d8. I thought it was a software bug since it works well in d8 and not in d16.

Asizon, 2020-05-06:
Umm, I think these issues related to spi rx protocol are fixed in 4.2

Zanth11, 2020-05-06:
I see to another thread on rcgroup the problem has been reported since 4.x and I tried 4.1.5 and it still bug. I have not seen an announcement regarding 4.2 :(

Where is patchnote for 4.2 ?

Thx u all

etracer65, 2020-05-07:
If it's a problem with the higher ESC PWM frequency causing RF interference then there's nothing we can do about it in the firmware.

Zanth11, 2020-05-07:
I do not know if it is an interference or latency problem, I noticed that before the loss rx I had some lag on my orders

etracer65, 2020-05-07:
@Zanth11 Then try testing different options to zero in on the problem. So far you know that this is good:
- BLHeli_S 16.7; 24KHz PWM

And that this is bad:
- JESC 48KHz PWM with bidirectional DSHOT enabled

So try:
- JESC 24KHz PWM with bidirectional DSHOT enabled
- JESC 24KHz PWM with bidirectional DSHOT **disabled**
- JESC 48KHz PWM with bidirectional DSHOT **disabled**

Zanth11, 2020-05-07:
I have try JESC 48Khz with bidirectional Dshoot disabled, no work
I have  short rang and random loss rx.

I try JESC 24Khz tomorrow if the weather is nice.

Thanks

Gozzz, 2020-05-09:
I have no problems running rpm filtering/jesc/d16 on 4.1.6 with crazybeef4.

Zanth11, 2020-05-09:
Hello, u dont run spi with frsky-x-lbt.

According to several reading threads, it seems that the problem does not appear on the d16 fcc mode. This is why I think more of a betaflight software bug than an interference problem between jesc 48khz and the spi rx.

mikeller, 2020-05-09:
@Zanth11: If the problem does not manifest when using one ESC firmware (BLHeli_S 16.7; 24KHz PWM) but not when using another one (JESC 48KHz PWM) then this is a strong indicator that this isn't a bug in Betaflight.

Either you do the tests that @etracer65 recommended above, or this should be closed as a hardware problem.

closed 2020-06-17 by stale[bot]