betaflight/betaflight #9773: LED Limit

yngndrw, 2020-05-06:
The limit of 32 LEDs has been discussed in the past (#1174, end of 2016) but with the prevalence of F7 flight controllers, I'd like this limit be re-considered.

A limit of 32 is inconvenient when using strips of 8 LEDs ([For example these boards](https://www.banggood.com/2X-Mateksys-2812-LED-Strip-Slim-Board-5V-for-RC-Drone-FPV-Racing-p-1286966.html) which are nice for arms as the data input and output are on the same side) because it allows for example 4 boards, one per arm. This doesn't allow any extra for the body of the quad.

A limit of 40 would allow for a 5th strip of 8 LEDs or 2x 4 LED strips for the body without an especially large change to the overhead.

etracer65, 2020-05-06:
> but with the prevalence of F7 flight controllers

This argument holds no weight. There are still far more F4 flight controllers in use than F7.

You can already connect banks of LEDs in parallel. So you could use a 16 LED chain (8 front, 8 back) on each side and connect them in parallel. Then off the end of one of those chains you can connect another 16 LEDs - giving you a total of 48. The only limitation is that the left/right sides (first 16 LEDs) will display the same colors/patterns.

mikeller, 2020-05-06:
@yngndrw: This is not so much a problem of resource consumption, as it is one of added complexity - unfortunately the way LED_STRIP information is exchanged with configurator was done in a bad way without creating a whole new  way to configure LEDs for firmware going forward, and forcing all configuration apps to support two ways for doing this from now on.
Since you can drive more than 32 LEDs by connecting some of them in parallel I think there is not enough demand for this change to justify adding this complexity and then having to maintain it.

yngndrw, 2020-05-06:
That's a shame, I know you can parallel them but it's a shame to have individually addressable LEDs and then just group them all together. On the flip side, I suppose that has to be balanced against the configuration and animation options that are available and what you'd be able to do with them if they were all individual. I fully appreciate the maintenance issue.

I might parallel some of them and see what I can do with them given the available options, if I'm not happy with that I'll build a separate controller for them to animate them as I want. I prefer the APA102/SK9822/Dotstar driver ICs anyway so I'd consider switching anyway.

Thanks both.

closed 2020-05-06 by yngndrw