betaflight/betaflight #9774: 4.2.0-RC1 freeze after arm/disarm without throttle

noah95, 2020-05-06:
**Describe the bug**
4.2.0-RC1 hang/esc reboot after arm-disarm cycle without applying throttle.

**To Reproduce**
Arm the craft and disarm shortly after without applying throttle. Now I can't arm again, the OSD is stuck on the post-flight summary screen and after ~10 sec the ESCs beep as if it were a restart and I can arm again. CMS is also not working during that lock phase.

**Expected behavior**
Fast arm/disarm cycling should work? With prearm toggling of course.

**Flight controller configuration**
```
# 
# Building AutoComplete Cache ... Done!
# 
# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 29 2020 / 08:08:49 (b480103d4) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name FURYF4OSD
manufacturer_id DIAT
mcu_id 00400046374e501120353355
signature 

# name: NoHo

# resources
resource I2C_SCL 1 NONE
resource I2C_SDA 1 NONE
resource PINIO 1 B07

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1 19200 57600 0 115200

# aux
aux 0 0 0 1300 1700 0 0
aux 1 13 0 1700 2100 0 0
aux 2 26 0 1300 1700 0 0
aux 3 31 1 1700 2100 0 0
aux 4 35 3 900 1300 0 0
aux 5 36 1 1700 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set gyro_notch1_hz = 400
set gyro_notch2_hz = 200
set acc_calibration = -35,33,89,1
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set ibata_scale = 115
set small_angle = 180
set pid_process_denom = 2
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1000
set osd_tim1 = 2561
set osd_vbat_pos = 66
set osd_tim_1_pos = 2485
set osd_vtx_channel_pos = 2145
set osd_current_pos = 439
set osd_mah_drawn_pos = 2198
set osd_craft_name_pos = 2475
set osd_gps_speed_pos = 343
set osd_gps_lon_pos = 482
set osd_gps_lat_pos = 450
set osd_gps_sats_pos = 386
set osd_home_dir_pos = 99
set osd_home_dist_pos = 101
set osd_altitude_pos = 375
set osd_debug_pos = 2400
set osd_power_pos = 2177
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2165
set osd_battery_usage_pos = 2152
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set debug_mode = D_MIN
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5769
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set gyro_1_align_yaw = 1800
set name = NoHo

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 0
set dterm_lowpass_hz = 0
set throttle_boost = 0
set p_pitch = 58
set i_pitch = 72
set d_pitch = 49
set f_pitch = 106
set p_roll = 54
set i_roll = 68
set d_roll = 45
set f_roll = 101
set p_yaw = 52
set i_yaw = 72
set f_yaw = 82
set d_min_roll = 26
set d_min_pitch = 28

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: FREE
B07: PINIO 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller Mamba F405 Mk2

If you can give me hints on how to debug this, let me know if I can provide more feedback.

etracer65, 2020-05-07:
Please try testing with a very basic configuration. Just enough to arm. No resource mapping, vtxtable, etc. If that works then try reintroducing your settings in steps to see if you can identify when the problem starts.

noah95, 2020-05-07:
The issue arises when the Blackbox erase mode is assigned to the same switch as prearm. 

- Blackbox erase = same range as prearm -> issue persists
- Blackbox erase mode not assigned -> issue is gone
- Blackbox erase on different switch: 
  - no erase, fast arm-disarm works
  - short toggle to erase mode and back (<30s), arming fails, board reboots

**To boil it down**: If switching to blackbox erase for <30s, arming fails afterwards and board reboots ~10s after.

Asizon, 2020-05-07:
@noah95 Blackbox erases take some time and aplies a board reboot when it have finished

noah95, 2020-05-07:
Oh okay, then I misunderstood the >30s! I thought you'd have to stay in the mode for >30s to initiate an erase. Thanks for the assistance and sorry for the false alarm.

closed 2020-05-07 by noah95