betaflight/betaflight #9776: No RSSI in BF when using Frsky XSR FPORT ACCESS 2.1, BF LUA and RC Control working

marekwiecek, 2020-05-06:
Problem Description

BF does not display RSSI in OSD, shows 0% in set-up screen when RSSI is supplied only via FPORT.
When configured to transmit RSSI via a channel, then it works but it is supposed to work without having to use the channel for it. Initial set-up done as based on https://oscarliang.com/setup-frsky-fport/:

1. Flash everything
2. Connect R-XSR smartport to TX 2 on the board
3. Configure receiver to fport

set serialrx_halfduplex = ON
set serialrx_inverted = ON
save
4. remove rssi source channel 

tried also with tlm_halfduplex ON/OFF and tlm_halfduplex ON/OFF - no change

HW:
transmitter: taranis x9lite, internal RF, flashed with ISRM-N_2.1.0.frsk
rx: R-XSR flashed with RXSR-ACCESS_2.1.0.frsk, connected to TX2 pad (not SBUS)
fc: mamba F722S Target: DIAT/MAMBAF722(STM32F7X2),

SW:
BF: 4.2 RC1
Configurator: 10.7.0-RC1 (091ce5b5)
BF Lua is working with this config, I can see values from FC config and I can set them from the radio LUA scripts
RSSI and other sensors are re-discovered
RSSI shows on Taranis 

Diff

```# version```
```# Betaflight / STM32F7X2 (S7X2) 4.2.0 Apr 29 2020 / 08:09:54 (b480103d4) MSP API: 1.43```
```# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: a5269145, date: 2020-04-09T22:32:24Z```

```# start the command batch```
batch start

board_name MAMBAF722
manufacturer_id DIAT

```# name: Drone```

```# feature```
feature -SOFTSERIAL
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature MOTOR_STOP
feature GPS

```# beeper```
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

```# serial```
serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 0 57600 57600 0 115200
serial 4 2 115200 57600 0 115200
serial 5 1 115200 57600 0 115200

```# aux```
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 13 2 1700 2100 0 0
aux 3 28 1 900 1300 0 0
aux 4 35 2 1300 1700 0 0

```# master```
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -14,22,202,1
set min_check = 1010
set max_check = 1990
set rc_smoothing_derivative_type = PT1
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_device = NONE
set dshot_idle_value = 700
set dshot_bidir = ON
set failsafe_switch_mode = STAGE2
set failsafe_procedure = GPS-RESCUE
set bat_capacity = 1550
set ibata_scale = 115
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_set_home_point_once = ON
set gps_rescue_angle = 40
set gps_rescue_descent_dist = 50
set gps_rescue_ground_speed = 1000
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 6
set deadband = 5
set yaw_deadband = 5
set tlm_halfduplex = OFF
set osd_vbat_pos = 2337
set osd_rssi_pos = 2509
set osd_rssi_dbm_pos = 2413
set osd_tim_2_pos = 2444
set osd_flymode_pos = 2455
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2401
set osd_craft_name_pos = 77
set osd_gps_speed_pos = 2177
set osd_gps_lon_pos = 2048
set osd_gps_lat_pos = 2065
set osd_gps_sats_pos = 2209
set osd_home_dir_pos = 2113
set osd_home_dist_pos = 2145
set osd_warnings_pos = 14634
set osd_avg_cell_voltage_pos = 2369
set osd_battery_usage_pos = 2089
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set osd_stat_min_rssi_dbm = ON
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 800
set name = Drone

profile 0

```# profile 0```
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dyn_lpf_dterm_curve_expo = 6
set dterm_lowpass2_hz = 195
set vbat_sag_compensation = 100
set feedforward_transition = 40
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 60
set throttle_boost_cutoff = 10
set p_pitch = 69
set d_pitch = 51
set f_pitch = 114
set p_roll = 63
set d_roll = 47
set f_roll = 108
set p_yaw = 68
set d_yaw = 30
set f_yaw = 108
set d_min_roll = 31
set d_min_pitch = 34
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40

rateprofile 0

```# rateprofile 0```
set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 102
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

marekwiecek, 2020-05-06:
This is NOT a BF issue. I did a bit more research and found an existing issue with the ACCESS 2.1 firmware from Frsky for R-XSR. Issue link here:

https://github.com/FrSkyRC/Firmware-Test/issues/36

once I tested a beta version they provided RXSR_ACCESS_2.1.0_20200506 on 3 different R-XSRs on BF 4.0 (on F3), 4.1 (on F4), 4.2RC1 (on F7) running on different boards it started working as expected. This is to be closed.

closed 2020-05-06 by marekwiecek

mikeller, 2020-05-07:
@marekwiecek: Thanks for supplying the information - I am sure this will be affecting other users as well.

marekwiecek, 2020-05-07:
Glad to pay it forward - happy flying!