betaflight/betaflight #9778: iFlight SucceX-D Mini f722 - Motor 2 DMA configured incorrectly on default configuration

JagerBombastic, 2020-05-07:
Hi,
I seem to have found a issue with the default config for the iFlight SucceX-D Mini f722. it uses the same target as the non-mini version which I also have but the following issue is not present on the non mini board.

The issue is that motor 2 will not work with bi-directional dshot, nor will it connect to BLheli_32 suite with the default config file. but it does spin up with dshot 300, 600 and PWM.

by default:
dma pin C06 0

With it changed to the following it starts working just fine.
dma pin C06 1

tested with betaflight 4.1.6 and blheli_32 32.7

mikeller, 2020-05-07:
@JagerBombastic: Can I get you to please post a `diff` of the configuration that works on the 'normal' board and does not work on the mini, plus a `resource show all` with this configuration from both boards?

JagerBombastic, 2020-05-07:
Please disregard the above I found it to not be correct after some more investigating.
It seems if I connect a runcam racer to the tx3 and rx3 pads of this board it knocks out the ability for esc 2 to communicate with the flight... very strange...

closed 2020-06-13 by stale[bot]