betaflight/betaflight #9781: F.Port broken in build #2188

haplm, 2020-05-08:
**Describe the bug**
F.Port is broken in last BF build (#2188 - 07-05-2020 13:43), but working in 4.2.0 RC1. Receiver is correctly configured and connected to Tx, but BF doesn't see it.

I'm using F4 with 'uninverted SmartPort' hacked receiver.

Of course I'm not sure about it, but looking at history of commits, #9771 is a suspect :-)

**To Reproduce**
Flash (last) build #2188, configure F.Port receiver, nothing happens

**Expected behavior**
Receiver is able to communicate with FC

**Flight controller configuration**
```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 May  7 2020 / 11:48:41 (b2e99c665) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id AIRB
mcu_id 003600223436511338363536
signature 

# serial
serial 2 64 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# master
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set pid_process_denom = 4
set gyro_1_align_yaw = 2700

# save configuration
save
#
```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 2
A03: MOTOR 4
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERIAL_TX 12
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 3
B01: MOTOR 1
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERIAL_TX 12
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH3: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: ADC 2
DMA2 Stream 4: DSHOT_BITBANG 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

# 
```

**Setup / Versions**
 - OMNIBUS AIO F4 V5 Flight Control
 - R-XSR, TBS UNIFY PRO 5G8 HV

**Additional context**
4.1.6 works as well

aowi7280, 2020-05-08:
I have the same issue with F.Port on Frsky qx7 and R9m/R9mm receiver. Confirmed build 2187 works OK. 
# Betaflight / STM32F405 (S405) 4.2.0 May  6 2020 / 14:23:35 (061f37965) MSP API: 1.43
# config: manufacturer_id: AIRB, board_name: OMNIBUSF4SD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

Asizon, 2020-05-09:
Again here???

Asizon, 2020-05-09:
@haplm Maybe you better have this open, and specify that there is a solution in https://github.com/betaflight/betaflight/pull/9771 , so we do not forget it

haplm, 2020-05-09:
Flashing FportTestHex from #9771 fixes the problem (on the bench). Will fly it hopefully later today  to give final confirmation.

haplm, 2020-05-09:
I have managed to test it in flight as well, and I can confirm that it works. Closing

PR #9785 (2020-05-13): Fix the FrSky CRC fix from #9771 (sorry)
This has been tested (thanks to @Asizon). The changes to the previous PR hadn't been tested, only the original PR, sorry.

Fixes #9781
closed 2020-05-09 by haplm