betaflight/betaflight #9782: No NTSC or Pal choices in OSD Tab in 4.2 RC1

dynofpv, 2020-05-08:
matekf411 betafpv F4 2-4S AIO Brushless Flight Controller 20A V2.0

missing video format menu, in configurator OSD Tab  

conf 10.7.0 RC1
BF 4.2 RC1
[4.2_BTFL_cli_Dyno_FPV 75X_20200508_101550.txt]
(https://github.com/betaflight/betaflight/files/4600533/4.2_BTFL_cli_Dyno_FPV.75X_20200508_101550.txt)
# diff all

# version
# Betaflight / STM32F411 (S411) 4.2.0 Apr 29 2020 / 08:09:11 (b480103d4) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF411, version: be0c9205, date: 2020-01-15T19:44:32Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF411
manufacturer_id MTKS
mcu_id 006a003d3437510a34363836
signature 

# name: Dyno FPV 75X

# feature
feature -SOFTSERIAL
feature -AIRMODE
feature LED_STRIP

# beacon
beacon RX_LOST
beacon RX_SET

# map
map RETA1234

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 6,7::A:0
led 1 7,7::A:0
led 2 7,6::CT:0
led 3 6,6::CT:0

# aux
aux 0 0 0 900 1300 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1700 2100 0 0
aux 4 28 1 900 1700 0 0
aux 5 35 3 1300 1700 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 3
vtxtable powervalues 0 1 2
vtxtable powerlabels 25 25 200

# master
set gyro_lowpass2_hz = 125
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 100
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 250
set acc_calibration = -171,91,-317,1
set min_check = 1008
set max_check = 1998
set fpv_mix_degrees = 17
set serialrx_provider = CRSF
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_poles = 12
set bat_capacity = 300
set yaw_motors_reversed = ON
set small_angle = 180
set osd_units = IMPERIAL
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_cap_alarm = 350
set osd_rssi_pos = 2336
set osd_link_quality_pos = 2400
set osd_rssi_dbm_pos = 2368
set osd_tim_2_pos = 2455
set osd_flymode_pos = 2443
set osd_craft_name_pos = 2121
set osd_avg_cell_voltage_pos = 2432
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 8
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5917
set gyro_1_align_yaw = 1800
set name = Dyno FPV 75X

profile 0

# profile 0
set vbat_pid_gain = ON
set p_pitch = 69
set d_pitch = 57
set f_pitch = 105
set p_roll = 63
set d_roll = 53
set f_roll = 99
set p_yaw = 68
set f_yaw = 99
set d_min_roll = 35
set d_min_pitch = 38
set auto_profile_cell_count = 3
set idle_min_rpm = 36

profile 1

# profile 1
set vbat_pid_gain = ON
set p_pitch = 78
set d_pitch = 65
set f_pitch = 105
set p_roll = 71
set d_roll = 60
set f_roll = 99
set p_yaw = 77
set f_yaw = 99
set d_min_roll = 39
set d_min_pitch = 43
set auto_profile_cell_count = 2
set idle_min_rpm = 25

profile 2

# restore original profile selection
profile 1

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 85
set pitch_expo = 85
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82

rateprofile 1

# rateprofile 1
set rates_type = ACTUAL
set roll_rc_rate = 10
set pitch_rc_rate = 10
set yaw_rc_rate = 10
set roll_expo = 85
set pitch_expo = 85
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82

rateprofile 2

# rateprofile 2
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 210
set yaw_expo = 10
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
#

Asizon, 2020-05-08:
Like the above warning message in configurator says, the osd chip has not detected, follow the instructions in this warning to make it work

Asizon, 2020-05-08:
@mikeller @etracer65 maybe it is more complex than we think, what about allows clicking on ntsc/pal selector and font manager, but push a warning popup if not osd chip available. It only have one button to cancel/go back without changes, but in this way the user is "obligated" to read these instructions.

dynofpv, 2020-05-08:
ok, I read the instructions. But, I did not disconnect usb before plugging in battery. all is good, sorry for the false report.

PR #9807 (2020-05-27): Fixed reporting of configured / detected OSD device.
Fixes #9782.

This fixes the reporting of the configured OSD device in MSP and CLI `status` for boards that do not power the MAX7456 chip when powered over USB, by keeping a record of the configured OSD type in this case.

Unfortunately, for this to work, the Unified Target configurations for these boards will need to have `set osd_displayport_device = MAX7456` added, as the `AUTO` setting results in the OSD type falling through to MSP if no OSD chip is detected (see betaflight/unified-targets#270).
closed 2020-05-08 by dynofpv