betaflight/betaflight #9786: [FEATURE REQUEST] Add center trim to rxrange command

joshuabardwell, 2020-05-09:
The rxrange command currently sets endpoints per channel, but channel center setting is shared by all four control channels. Users setting endpoints with rxrange may not be able to set channel centers with subtrims for the same reason they can't set channel endpoints with the transmitter. Propose to add an additional parameter to rxrange to set channel center.

so instead of "rxrange 0 1000 2000" it would be "rxrange 0 1000 1500 2000".

etracer65, 2020-05-09:
If we did this the "center" would have to go at the end of the command. like `rxrange 0 1000 2000 1500` with the last parameter bing optional. Can't change the ordering of the parameters for an existing command as that would catch everybody out.

Also I'm not sure if this would work. It would be possible to set a non-symmetrical range above/below the center. Right now the logic works by taking the input value from the receiver and scaling is relatively from the entered `rxrange` to the 1000-2000 range. The channel center is always still 1500. Then to all the rest of the code the input looks like 1000/1500/2000 scaling. Then all channels are evaluated agains the single `mid_rc` value.

Even if we had `mid_rc` values for each channel they would still be in the context of the scaled 1000-2000 range and not the pre-rxrange input.

Sounds like a big project.

mikeller, 2020-05-10:
@joshuabardwell: I am not aware of any currently sold hardware that would need this. Are you?

joshuabardwell, 2020-05-29:
More fundamentally -- if there is a reason to adjust endpoints in the CLI, then the same reason should justify adjusting center in the CLI. If the argument is, "let them do it in the transmitter," then why does rxrange exist at all? If the argument is, "let them do it in the CLI," then shouldn't they also be able to do centers.

@etracer65 my thinking of the logic would be that the channel values would be distributed linearly between center and endpoint rather than from endpoint to endpoint.

So if you change center from 1500 to 1600, you would have 400 values from center to right endpoint, and 600 values from center to left endpoint.

mikeller, 2020-05-31:
@joshuabardwell: At least for Spektrum, the range is actually 1020 to 2020, with a symmetrical center of 1520 - no need for users to do anything but set `rxrange 0 1020 2020` (https://www.lakesawyerhawks.org/pdf/RC_Tech_Tips/Servo_Center_and_Range_for_Spektrum_Tx_by_Andy_and_Others.pdf).
Do you have any pointers to documentation for FlySky? It would be unusual if they had an asymmetrical center with the min - center range being different from the center - max range, and stick movement below or to the left of the center resulting in less craft movement than at the top / to the right.

Re changing the endpoints in CLI and not in the UI, this is something that was this way forever since Betaflight forked from cleanflight - we could add it to the UI, but I am not sure if this will be helpful for novice users - the defaults are working as they are for most users, and changing the RX ranges will cause it to stop working in most cases.

BrandonsBakedBeans, 2020-07-29:
> @joshuabardwell: I am not aware of any currently sold hardware that would need this. Are you?
Spektrum DX6i's can be hard to setup with the current RXRANGE command. One of mine went out yesterday, requiring me to bind everything to the second one, but the gimbals were wildly different values (there was a recall on these unit for the pots, one of my units was sent in and the other was not). Setting my endpoints in the CLI with the current command will lead to center values being up to 20 microseconds off. Additionally, setting subtrim on this radio will move the endpoints. The fix is to override the endpoints using the transmitter's travel adjustment menu, but some of the economy radios by this brand do not have that option. I have a Spektrum dsm2 transmitter that came with a helicopter (model hp6dsm), but has no screen (only dip switches) so this unit would affected (assuming they use similar components, this was not tested).
So, while an overwhelming majority of radios can be adjusted to fit the range of travel accurately, maybe it would still be beneficial to move the remaining configuration from the radio to the configurator for unification of procedure, if nothing else.

mikeller, 2020-08-01:
@BrandonsBakedBeans:

> there was a recall on these unit for the pots, one of my units was sent in and the other was not

It sounds to me like you are proposing that this is used as a workaround for hardware that was shipped defective and needs to be returned / fixed by the manufacturer. I do not think that encouraging manufacturers to sell defective hardware 'because Betaflight will fix it' is an approach that is in the users' best interest.

BrandonsBakedBeans, 2020-08-01:
@mikeller:
> It sounds to me like you are proposing that this is used as a workaround for hardware that was shipped defective and needs to be returned / fixed by the manufacturer. I do not think that encouraging [...]

While my motivations for bringing this up is bad hardware design, my argument for the feature is more for uniformity in setup from user to user.

nightflasher, 2021-05-18:
My first radio was a spektrum DX6i and it was more than hard to get the values right, so I have up on it in two ways: 1) didn't touch the rxrange anymore, 2) sold the transmitter. Flysky fs-i6x never had this issue, but came back with the nirvana, but could easily fixed in the transmitter.
But today, there is a new kid around: ExpressLRS
It has this one "multi position switch", which doesn't have 1500...maybe this is motivation enough?

joshuabardwell, 2022-09-18:
@mikeller Is there any interest in re-visiting this?

I was reminded of this again today when helping someone whose channel centers were not correct. He was using a controller that did not have easily-adjustable center subtrim or endpoints. If he could have used rxrange, it would have been easy to fix the issue. As is, there wasn't an easy fix other than, buy a new controller.

My basic question remains: if the argument is, "adjust your endpoints and subtrim in the controller," then why does rxrange exist at all; if rxrange exists then we accept that the FC has some interest in adjusting endpoints; at which point, why does it not also adjust centers?

closed 2026-09-26 by blckmn