betaflight/betaflight #9787: gps_rescue_min_dth lower than 50 meters

Xedos9er, 2020-05-09:
Is there any chance to enable lower values for gps_rescue_min_dth than 50m, maybe even set to 0? It's really annoying to see the quad getting dropped when failsafe happens flying behind trees in short range (<50m). In many situations the quad would have been saved when GPS rescue would have engaged. Please enable setting lower than 50m !!

etracer65, 2020-05-09:
This has been discussed multiple times and from a safety perspective has always been rejected. You shouldn't be experiencing a failsafe at less than 50m. Fix your radio link.

closed 2020-05-09 by etracer65

Xedos9er, 2020-05-09:
You can experience a 
FS at 50m, especially flying NLOS behind trees. Happened several times on different RX.

etracer65, 2020-05-09:
That is not normal. You need to consider repairing or replacing your radio equipment.