betaflight/betaflight #9792: LiPo Battery Cell Monitor - sensor support

Simmy07, 2020-05-10:
**Introduction**
I would like to propose new feature in order to improve battery capacity estimation along with notifying user about this.

**Is your feature request related to a problem? Please describe.**
User estimate battery usage either using timer, entire pack voltage or pack voltage plus current consumption which gives rough consumed energy. This brings disadvantages like a knowledge when battery is low for 2s, 3s (and so on) or a need to remember which battery pack is currently used in a case of consumed energy. There may be also old batteries, which do not provide full capacitance.
Users also often use cheap cell's monitors which not only does not provide accurate readings, but they do not provide any feedback besides embedded buzzers. Others existing solutions provide only WiFi access to the battery status or are designed to work only with *FrySky* transmitters.

**Describe the solution you'd like**
I am developing a *LiPo Battery Cell Monitor* board which can monitor voltage of each cell independently. While battery voltage is a poor estimate of remaining capacity I believe it is way better than above solution. This board uses comparator to extract exact cell's voltage. Later those readings can be presented via serial port (I2C advised due to UART ports being to precious; SPI would consume ADC). Board could present per-cell voltage, entire pack voltage and (the most important) the weakest cell voltage. The weakest cell voltage is actually the one that could be presented via OSD, telemetry (this for example solves issues with cheap *FlySky FS-i6* transmitters), low-battery warning buzzer, on-board status LED flashing and LED strip patterns to user. In order to save main CPU cycles, *LiPo Battery Cell Monitor* can smooth out readings itself either by using average or some sort of filtering.

[image]

Currently, *LiPo Battery Cell Monitor* is designed for use with 2s-4s battery packs. It is possible to extend it for use with 2s-6s battery packs by introducing one more comparator and redesigning the board. MCU has enough ADC's to support such solution. In a later stages of the development a current sensor could be added along with a voltage SAG calculation. If attaching this sensor permanently to the battery would be needed, there could be added an transistors or an analog switch to minimize further power consumption.

squshy7, 2020-05-12:
Curious on your thoughts on how this would physically look...are you envisioning a device that plugs in to the balance lead (and possibly in line with the main leads for current sensing)? If this is the case, how would users neatly attach the data wires to the FC without being too fragile or potentially ripping pads off the FC (being that batteries are nearly always external to the electronics stack)? Is it possible that this could be designed to communicate over bluetooth to the FC, alleviating the wiring issues?

michal-konopinski, 2020-05-12:
Look at BattGO, very nice solution, only one more wire, but propertiary :(

Simmy07, 2020-05-13:
@squshy7 I was actually thinking about small board that you can plug into balancing leads of the battery. I don't think this would be a problem to have 4 wires coming from the FC to the *LiPo Battery Cell Monitor* board - there can always be an easy to attach connectors in case you don't want to use it *right now*. 

Actually, I like the PDB integration idea, however I didn't proposed any since my drone has integrated PDB with frame and my plane has no PDB at all. But surely there could be two versions, a PDB one and standalone one.

As for ' just keep in mind that the only thing it solves is being unable to read individual cell voltages' - I believe it can be more reliable than just measuring battery capacitance during a fly (but I'm not an expert). In order to correctly estimate battery capacity including issues with charging old/damaged cells, this board would have to be attached to the battery *always*.

@michal-konopinski There are actually few solutions out there, but like you said, they are either big, proprietary or designed to work only with receivers of one producer. Also, I'm not enough hardware skilled to design a device that would be constantly attached to the battery.

mikeller, 2020-05-16:
I think your idea makes some sense - having individual cell voltages will definitely be an advantage - it means more information about the in-flight battery state, and it will allow diagnosing things that are currently impossible / hard to diagnose, like otherwise good packs with one cell slowly dying.
I think trying to build this as something that goes in between the balance connector and the flight controller is a good approach - the commercial solutions that are available all have the drawback (for the user) that they are permanently attached to the battery, which means that the user has to spend extra money for every battery equipped with this - having to spend the money only once per craft will be cheaper.

closed 2026-09-26 by blckmn