betaflight/betaflight #9794: change PID via Blackbox Explorer

BavariaAirview, 2020-05-10:
It´s not a comfortalbe way to review BB Logs and have to jump to Betaflight to change values. 
Even when you still are in massstorage mode to check varius Logs. To write a letter I´m far too lazy. 

The Idea I had to change this:  You are goning to open Logs in BBE. When you like to change something, you´ll open the PID informations and click on the value you like to change. 
In the Backround BBE creates a txt with CLI commands to change these values. 
On reboot BF checks the storage, set the values, deletes the file, and reboots again.

etracer65, 2020-05-10:
Blackbox viewer doesn't have any of the infrastructure to connect or communicate with the flight controller.

BavariaAirview, 2020-05-11:
Yeah, I thought it is like this.  There for I brought the idea to create a txt to store cli commands on the BBE storage. This Commands could be read by the FC and change the settings.

etracer65, 2020-05-11:
First the Blackbox viewer doesn't contain all the validation or rules to be able to correctly generate valid values for the PIDs. Further these rules change between firmware versions so BBE would need to have all the historical rules in place as well. Next the flight control has no ability to "read files" from the flash storage - it only writes streaming data to the flash. Lastly, onboard flash for example cannot be written to from your desktop as it's a read-only emulated filesystem.

BavariaAirview, 2020-05-20:
All right, then close this issue, please.

closed 2020-05-27 by BavariaAirview