betaflight/betaflight #9796: Holybro KakuteHDV Can’t Remap Motor Outputs

vicewize, 2020-05-10:
Holybro Kakute HDV when updating to new 4.2 candidate and trying to reassign motor outputs (my ESC is mounted backwards) the CLI states that the motors are in use still after freeing them. So I have tried saving after this simple change then the flight controller will not be recognized by computer at all (won’t even show up as a device). I have to put the FC in DFU mode and reflash to allow it to connect. Has worked fine on all previous versions of 4.1. I’m running a mac with  Betaflight configurator 10.7.0-RC1 on Mac OS Mojave 10.14.6. 

Never had this issue before and I have recreated it 3 times (by reflashing) one time by making no changes and gone directly to CLI to reassign first thing still happens.

‘ ‘ ‘
# version
# Betaflight / STM32F745 (S745) 4.2.0 Apr 29 2020 / 08:09:34 (b480103d4) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7HDV, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7HDV
manufacturer_id HBRO
mcu_id 003100533236510b31393330
signature 

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# master
set gyro_lowpass2_hz = 325
set dyn_notch_width_percent = 0
set dyn_notch_q = 225
set dyn_notch_min_hz = 90
set dyn_lpf_gyro_min_hz = 260
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = -128,210,-272,1
set baro_hardware = AUTO
set rc_smoothing_auto_smoothness = 9
set serialrx_provider = CRSF
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT300
set motor_pwm_rate = 480
set small_angle = 180
set pid_process_denom = 2
set gyro_1_align_yaw = 2700

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set thr_mid = 38
set thr_expo = 36
set rates_type = ACTUAL
set roll_rc_rate = 12
set pitch_rc_rate = 10
set yaw_rc_rate = 11
set roll_expo = 65
set pitch_expo = 50
set yaw_expo = 60
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

ratCurrently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: PREINIT 3
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 3
E10: FREE
E11: MOTOR 4
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 5
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 5
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

Kakute F7 HDV with DJI System running TBS Crossfire nano receiver. Hobbywing 60 Amp 4in1 ESC.

Also the configurator has a weird glitchy graphic effect going on and acts somewhat strange but totally useable. Havent tried this on windows in my virtual machine yet but will. 

Thx

etracer65, 2020-05-10:
Just do:
```
resource MOTOR 1 NONE
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
```
and then remap them to the correct pins if you don't want to see the warning that doesn't really matter in this case.

vicewize, 2020-05-11:
I did that that’s not the issue it’s after I remap them even after freeing them when I hit save and it reboots, the FC is no longer recognized as a device at all. Only in DFU. Maybe my configurator is possibly corrupted I’ll try it on windows and let you know. Thank you

vicewize, 2020-05-11:
Ok I found the issue by searching the configurator issues and only seams to effect MacOS. Apparently its in the firmware and tested by upload a unified target posted. Heres the page: 

https://github.com/betaflight/betaflight-configurator/issues/2006

etracer65, 2020-05-11:
@vicewize You are referencing a completely unrelated thing in regards to your issue. Remapping motors takes place in the CLI. The CLI is not affected by the issue your referred to. Can you please be clear what your issue is? If it's a poorly worded duplicate of the referenced problem then it will be closed as a duplicate.

vicewize, 2020-05-14:
Sorry if that was confusing (that was after a long day, apologies). Basically when I went to remap the motors in the CLI, after I would type "save" and it would reboot (like usual), the FC would no longer be recognized as a device at all by my mac, the only way I could get it back was by putting the FC into DFU and then it would be recognized, and I could reflash which I repeated multiple times. I believe it was probably any change I made in the CLI would case this issue after issuing the "save" command, it just happens that the first thing I did was remap motors so I believed it was remapping that was causing the issue, which is unlikely. I flashed the newest release today and the problem no longer exists, it reboots perfectly and is instantly recognized as it normally would after remapping or doing any CLI change and saving. So this issue now is irrelevant and fixed. Thanks and I appreciate everyones work.

closed 2020-05-14 by vicewize

teahwttam, 2020-09-25:
I'm having the exact same issue but on windows 10.
I have to put the FC in DFU mode and reflash after I remap  motors.