betaflight/betaflight #9799: No uarts in port tab - iFlight SucceX-D F7

Hexenmeister1, 2020-05-11:
1. DJI Remote control was correctly connected but it did not work at all.
2. In Setup tub I clicked "Reset Settings" and everything disappeared (all information from sensors and in general it looks like FL is dead)
3. "Connect" button works but not all the time but if I use "Update Firmware" it does not work at all I have an error: failed to open serial port.
It would be really nice to restore everything in default state. For example, override Firmware or get correct default settings. Could you help with this problem?
-------------------------------------------------------------------------------------------
# 
# Building AutoComplete Cache ... Done!
# 
# dump
###WARNING: NO CUSTOM DEFAULTS FOUND###

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.1 Nov 15 2019 / 12:57:05 (1e5e3d369) MSP API: 1.42
# manufacturer_id: IFRC   board_name: IFF7_TWIN_G   custom defaults: NO

# start the command batch
batch start

board_name IFF7_TWIN_G
manufacturer_id IFRC

# name: -

# resources
resource BEEPER 1 NONE
resource MOTOR 1 NONE
resource MOTOR 2 NONE
resource MOTOR 3 NONE
resource MOTOR 4 NONE
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 NONE
resource SERIAL_TX 4 NONE
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 NONE
resource SERIAL_RX 2 NONE
resource SERIAL_RX 3 NONE
resource SERIAL_RX 4 NONE
resource SERIAL_RX 5 NONE
[... 953 lines cut ...]
set rateprofile_name = -
set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 65
set tpa_breakpoint = 1250
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end
----------------------------------------------------------------------------------------------

etracer65, 2020-05-11:
You're using the wrong target. Refresh (with full chip erase) and use the `IFLIGHT_F722_TWING` target.

Hexenmeister1, 2020-05-11:
its not. It looks like "Reset Settings" does not work correct.
On the Firmware page I found config link [https://raw.githubusercontent.com/betaflight/unified-targets/master/configs/default/IFRC-IFLIGHT_F722_TWING.config](url) and with CLI overwrite existing. Now I have default state with latest firmware.
Do you have any idea why DJI Remote Controller does not work? I found different advices but till current moment have not resolved this problem.

etracer65, 2020-05-11:
The fact that it says: `custom defaults: NO` in your diff tells me you didn't flash the correct target. You probably flashed the generic `STM32F7X2` target and pasted in a config. That won't survive a reset to defaults. Flash the target I told you and when you connect the first time say "yes" to applying custom defaults.

Hexenmeister1, 2020-05-11:
maybe you are right. I tried different things to run Remote Controller.
Thank you for help.
 Could you give me one more tip about Remote Controller? The default manual do not resolve this problem.

closed 2020-05-11 by Hexenmeister1

etracer65, 2020-05-11:
This is not DJI support. I believe they have a manual and a support group.