betaflight/betaflight #9801: STM32F7X2 (S7X2) error w/ DMA assignment

gasrovescio, 2020-05-13:
I flashed the firmware 4.1.2 of my FC iFlight SucceX F7 TwinG V2.1 and all the UART ports have disappeared except for the USB-C port.
So I tried to restore my CLI that I had saved to see if I solved the problem but reports this error: # dma pin C08 1
### ERROR: INVALID DMA OPTION FOR PIN C08: '1' ###

What should I do now?

[BTFL_cli_20200514_003207.txt](https://github.com/betaflight/betaflight/files/4625082/BTFL_cli_20200514_003207.txt)

gasrovescio, 2020-05-13:
$M>em
Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# #

# # diff

# # version

# # Betaflight / STM32F7X2 (S7X2) 4.1.2 Dec 22 2019 / 02:21:16 (89051e256) MSP API: 1.42

# # custom defaults: YES

# # start the command batch

# batch start
Command batch started

# # name: Titan DC5

# # resources

# resource BEEPER 1 C15

Resource is set to C15

# resource MOTOR 1 C08

Resource is set to C08

# resource MOTOR 2 C06

Resource is set to C06

# resource MOTOR 4 C07

Resource is set to C07

# resource MOTOR 5 B06

Resource is set to B06

# resource MOTOR 6 B07

Resource is set to B07

# resource MOTOR 7 B01

Resource is set to B01

# resource MOTOR 8 B00

Resource is set to B00

# resource LED_STRIP 1 A01

Resource is set to A01

# resource SERIAL_TX 4 C10

Resource is set to C10

# resource SERIAL_TX 5 C12

Resource is set to C12

# resource SERIAL_TX 6 NONE
[... 268 lines cut ...]
dashboard_i2c_bus set to 2
# set pinio_config = 129,1,1,1
pinio_config set to 129,1,1,1
# set pinio_box = 0,255,255,255
pinio_box set to 0,255,255,255
# set gyro_2_spibus = 1
gyro_2_spibus set to 1
# set gyro_2_sensor_align = CW90
gyro_2_sensor_align set to CW90
# set gyro_2_align_yaw = 900
gyro_2_align_yaw set to 900
# set name = Titan DC5
name set to Titan DC5
# profile 1
profile 1

# rateprofile 0
rateprofile 0

# # end the command batch

# batch end
###ERROR: ERRORS WERE DETECTED - PLEASE REVIEW BEFORE CONTINUING###

# #

etracer65, 2020-05-13:
Flash the proper firmware target. `IFLIGHT_F722_TWING` and be sure to enable the "full chip erase" switch.

gasrovescio, 2020-05-13:
thank you so much you saved my life !!! so since you have been so kind I can ask you if it is the same firmware for the drone version 6S because a friend of mine has the same problem. Eventually what does he have to change?

etracer65, 2020-05-13:
You need to contact the manufacturer and ask them what target you should use. We don't know what components they put into their products and we don't provide support for their products.

closed 2020-05-13 by etracer65