betaflight/betaflight #9802: Betaflight 4.1.6 Unified, Does not recognize QMC5883L compass on omnibus f4 pro v2.1 board

c172jeff, 2020-05-14:
**Describe the bug**
I have the Mateksys M8Q-5883 compass(QMC5883L)/gps installed on version 4.1.6 install of betaflight.  The board does not recognize any magnetometer connected.

**To Reproduce**
Using a banggood clone board, the one called OMNIBUS F4 Pro v 2.1, install betaflight 4.1.6.  

I am using the below betaflight firmware...
# Betaflight / STM32F405 (S405) 4.1.6 Apr 25 2020 / 05:11:18 (283bda8bf) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

**Expected behavior**
The firmware should recognize my magnetometer as connected.

**Flight controller configuration**
```

# diff all

# version
# Betaflight / STM32F405 (S405) 4.1.6 Apr 25 2020 / 05:11:18 (283bda8bf) MSP API: 1.42
# manufacturer_id: AIRB   board_name: OMNIBUSF4SD   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4SD
manufacturer_id AIRB
mcu_id 002700373536470534353030
signature 

# feature
feature GPS
feature RSSI_ADC

# serial
serial 0 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200

# master
set mag_declination = -14000
set serialrx_provider = SBUS
set failsafe_procedure = GPS-RESCUE
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_rescue_initial_alt = 80
set gps_rescue_ground_speed = 35
set osd_rssi_pos = 2262
set osd_rssi_dbm_pos = 232
set osd_gps_speed_pos = 2282
set osd_gps_sats_pos = 2282
set osd_home_dir_pos = 2282
set osd_home_dist_pos = 2282
set osd_altitude_pos = 2282
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vcd_video_system = NTSC
set gyro_1_align_yaw = 2700

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
```

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: ADC_RSSI
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: BARO_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: INVERTER 6
C09: INVERTER 3
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 3
TIM3:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: ADC 2
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

```
Can anyone help me get my compass working?  Software such as INAV can see the gps and compass.  Thanks

etracer65, 2020-05-14:
That target doesn't have I2C configured. Where are you connecting the SCL/SDA?

c172jeff, 2020-05-14:
I just found this thread....

https://github.com/betaflight/betaflight/issues/4531

I used the commands ....

resource SERIAL_TX 3 NONE
resource SERIAL_RX 3 NONE
resource I2C_SCL 2 B10
resource I2C_SDA 2 B11

and it seems to now be working. I am still testing though.

Thanks,
Jeff

closed 2020-05-15 by c172jeff