betaflight/betaflight #9804: servo motor mixer

JamesMKC, 2020-05-14:
Hello,
Firstly I'm really impressed with this software. 
I'm building a quad with four variable pitch propellers and would like to mix the four servos controlling the pitch on each propeller with the ROLL and PITCH signals sent out by the FC. Is that mix possible?

etracer65, 2020-05-14:
Not sure how it would work have the flight controller managing the motor speeds as a normal quad flight control along with variable pitch also affecting thrust. Unless you intend to drive all the motors from a single fixed throttle source and use the variable pitch for the flight control. Don't think the servos could move fast enough for this.

Also going to struggle I think to find a flight controller with 8 motor outputs (4 motors + 4 servos).

You might look here: https://github.com/betaflight/betaflight/blob/master/docs/Mixer.md

jflyper, 2020-05-15:
@JamesMKC 

If you are trying to come up with a quad rotor drone with props rotating at constant speed and use propeller pitch to control attitude of the drone, it isn't possible, as resulting craft would not be able to yaw.

JamesMKC, 2020-05-15:
Thanks Etracer65.
I had previously reviewed the information on mixer, mmix and smix at length. I tried to use it but it seems only one signal can be linked/mixed with each servo.
e.g. smix 0 3 0 100 0 0 100 0  # Roll / Aileron
 I will try outputting the motor roll and pitch signals to spare channels and then mix using the TX mixer. FYI the quad design transitions to forward flight. This is the primary reason for needing such a system.

JamesMKC, 2020-05-15:
Thanks for reviewing my post. This flight vehicle will retain yaw control through motor speed but it will also transition to forward flight.  Motor yaw control in hover will be modified in TX mixer during transition. 

Cheers 

James

mikeller, 2020-05-16:
@JamesMKC: Can I recommend that you join the Betaflight Slack (registration at https://slack.betaflight.com/), and look for user 'Wolfgang S'? He is working on experimental variable pitch aircraft, and he can potentially help you.

closed 2020-06-22 by stale[bot]