betaflight/betaflight #9805: channel_forwarding_start = 8 .... returns ###ERROR: UNKNOWN COMMAND, TRY 'HELP'###

JamesMKC, 2020-05-15:
**Describe the bug**
Betaflight 4.1.6 configurator 10.6.0

CLI returns the statement "unknown command" when applying "channel _forwarding_start = 8"

# channel_forwarding_start = 8
###ERROR: UNKNOWN COMMAND, TRY 'HELP'###

**To Reproduce**
in CLI enter command...    channel _forwarding_start = 4 (or 8)

**Expected behaviour**
command should be accepted

**Flight controller configuration**
tmotorF7  TMTR/TMOTORF7(STM32F7X2)

'''# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: LED 1
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: FREE
B09: CAMERA_CONTROL
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: PREINIT 4
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: MOTOR 1
C07: MOTOR 2
C08: FLASH_CS
C09: FREE
C10: FREE
C11: SERIAL_RX 3
C12: FREE
C13: BEEPER
C14: PINIO 1
C15: PREINIT 2
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE'''

**Setup / Versions**
 - Flight controller Taranis X9D+2019  Other components RX, X8R sbus to tmotorF7 FC using R3 uart

etracer65, 2020-05-15:
`set channel_forwarding_start = 8`

Please note that this not a support site.

Try our slack: 
https://slack.betaflight.com

Or Facebook group:
https://www.facebook.com/groups/betaflightgroup/

closed 2020-05-15 by etracer65

JamesMKC, 2020-05-15:
Hi Bruce
So that is a problem with my FC not a bug in the latest release of Betaflight
Sorry about that Bruce.