betaflight/betaflight #9806: Matek F405-STD - I2C 3 Unified Target Change - Motor 4 Not Spinning

SnowLeopardFPV, 2020-05-16:
**Describe the bug**
A recent unified target configuration change to the [MTKS-MATEKF405STD target](https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF405STD.config) to fix a reported issue with the I2C 3 resource mapping for legacy Matek F405-OSD boards has resulted in that change now breaking compatibility with the Matek F405-STD for which that target is actually intended. This breaking change was implemented in unified target pull request [betaflight/unified-targets#259](https://github.com/betaflight/unified-targets/pull/259).

With the new default I2C 3 resource mappings now added, motor 4 (connected to pad S4) on a Matek F405-STD board will no longer spin up if trying to use an ESC protocol other than one of the DSHOT protocols.

**To Reproduce**
Flash Betaflight 4.1.x to a Matek F405-STD board with "Full chip erase" enabled and apply the custom defaults when asked. Then set the ESC protocol to one of PWM, ONESHOT125, ONESHOT42 or MULTISHOT and notice that motor 4 will no longer spin up. If you set the ESC protocol to any of the DSHOT values then all motors spin up fine. I haven't tested the BRUSHED or PROSHOT1000 protocols because my ESCs don't support either of those.

Remap I2C_SCL 3 to NONE and I2C_SDA 3 to NONE as were the previous defaults prior to the change, and then notice that all 4 motors spin up regardless of which ESC protocol is selected.

**Expected behavior**
All motors should spin up regardless of the ESC protocol selected.

**Flight controller configuration**
This happens on a vanilla install of Betaflight 4.1.x with the custom defaults applied for the board and only the ESC protocol changed in order to demonstrate/reproduce the issue.

**Additional context**
Because the I2C 3 resource mapping changes were made specifically to fix an issue for the legacy Matek F405-OSD board as detailed in pull request [betaflight/betaflight#9750](https://github.com/betaflight/betaflight/pull/9750), it would make more sense to revert the changes made to the [MTKS-MATEKF405STD unified target](https://github.com/betaflight/unified-targets/blob/master/configs/default/MTKS-MATEKF405STD.config) and instead create a new unified target specifically for the Matek F405-OSD board to address the I2C 3 mapping issue on that particular board. Then any changes required to either target won't risk breaking the other one.

mikeller, 2020-05-16:
@SnowLeopardFPV: Can you please supply the information that you were asked to supply when opening this issue?

SnowLeopardFPV, 2020-05-16:
I'm not really sure what you are asking for. I've provided all the required information in the description and the exact steps to reproduce the issue. In fact I paid special attention to being as detailed and verbose as I could with a description and reproduction of the problem.

I flashed Betaflight 4.1.6 the day it was released and all was fine. If I flash it today, all is not fine. This is **without** applying any previous settings from a "diff all" backup. The only changes that have been made since the 4.1.6 firmware was released is the I2C 3 mappings in the unified config file, and reverting these immediately fixes the problem, so it proves those changes are the cause of the issue.

One of the Betaflight devs with a Matek F405-STD test board will immediately be able to reproduce this issue if they follow my steps.

It happens on a **vanilla** install of Betaflight 4.1.6 with **no changes** other than the ESC protocol as specified in my description of the issue, so the flight controller configuration is irrelevant. Everything is at **default** settings. The flight controller is specified in both the title and the description, and any other components are irrelevant because Betaflight is using the vanilla configuration with **nothing else** changed from the defaults.

I am not the only one to discover this issue. I just tested it to confirm the issue existed and then took it upon myself to report it so it can get fixed, or the changes that were applied which caused the issue be reverted.

mikeller, 2020-05-17:
@SnowLeopardFPV: Can you please test if the target I added to #9810 fixes your problem?

SnowLeopardFPV, 2020-05-18:
The fix is now tested and confirmed as working. I flashed the **betaflight_4.2.0_STM32F405_37209b4826.hex** firmware to my Matek F405-STD and then set the ESC protocol to ONESHOT125. All motors now spin up fine without any issues and I can see that MOTOR 4 has retained its mapping to C09 in the resource list. So this fix has resolved the original problem.

The only question now is that this is a code fix which has only been applied to the Betaflight 4.2 code base. Are there any plans to fix or revert the unified target changes that now (and still) cause issues under Betaflight 4.1.x? That version was working fine until the unified target configuration changes were made simply because someone wanted a change added for the legacy F405-OSD board which I doubt many people use any more compared to the newer F405-STD. So if someone simply re-flashes a version of Betaflight 4.1.x to their F405-STD which was previously working, they are going to be left scratching their head. There is no secondary I2C breakout pads on the F405-STD board so it doesn't need a second pair of I2C resources mapped.

PR #9810 (2020-05-27): Fixed pin ownership checks for I2C.
Fixes #9806.

This fixes the problem that currently, `IOInit()` does not check if a pin is already used (i.e. has an owner assigned) before setting the owner. This leads to cases where a pin is successfully assigned and a pin function started, but then the pin gets re-assigned to a different function, leaving artefacts:

Setup:

```
# resource

[...]

resource MOTOR 4 C09

[...]

resource I2C_SDA 3 C09

[...]
```

Before:

```
# resource show all

[...]

C09: I2C_SDA 3

[...]

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    C [... cut]
closed 2020-05-27 by mikeller