betaflight/betaflight #9809: Brain Radix stop connecting

DallyDDD, 2020-05-16:
**Describe the bug**
Brain Radix. Keeps disconnecting on 4.2. Flash ok, start setting up, and at random parts of reboots, eg, setting up in configuration,( reverse motor or beepers etc),  it will STOP being recognised by the configurator. FRSKY rx Fport being used. tried 2 boards, changed usb lead, still faulty.
Have to re flash, by putting into DFU mode. 
tried many times, 
Re flashed to older 4.1.3, and it all works fine.

**To Reproduce**
flash brain radix to 4.2 RC1, connect to config. 
go to configuration tab,
select frsky FPORT,
save and reboot.
now wont connect.

**Expected behavior**
A clear and concise description of what you expected to happen.
expected to setup correctly

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
wont connect so cant copy it.

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; Brain Radix
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up. rx FRSKY Fport

**Additional context**
Add any other context about the problem here.
tried a FC board which is not connected to anything else, just the board by itself, same problem

mikeller, 2020-05-17:
@DallyDDD: Your board is not supported by Betaflight - talk to the manufacturer about fixing the problem that you are having with the version of Betaflight that they are releasing for it.

closed 2020-05-17 by mikeller

DallyDDD, 2020-05-17:
thanks for the speedy reply, as far as I know, the board is now supported, as it has its release candidate, and other people with same board are using this version.

mikeller, 2020-05-17:
@DallyDDD: As I said in my last response, your board is not supported **by Betaflight** - you will have to contact the manufacturer to get support with the inofficial version of Betaflight that they are releasing.

jflyper, 2020-05-20:
@DallyDDD 
You may want to visit BrainFPV's FB https://www.facebook.com/groups/brainfpv/ if you haven't done so already.

Btw, Martin calls the software under question as "our version of Betaflight".
> We have updated our version of Betaflight to 4.2 RC1. The hex files are available on GitHub linked below.