betaflight/betaflight #9812: channel_forwarding is not working on BF 4.2 (RC1 and RC2)

brauliolorenzo, 2020-05-17:
**Describe the bug**
Hi, I was using BF 4.04, and put 4.2 (RC1 and RC2). The channel forwarding is not working.

**To Reproduce**
Steps to reproduce the behavior:
I´m setting channel_forward in Configuration, and on CMD I´m puttin CHANNEL_FORWARDING_START = 7. Like I was using on BF 4.04. Change the Motor 6 to NONE and Servo 1 to A08, and on Servo page, I put to change Aux4. I´m using exactly I was using on 4.04
**Expected behavior**
A clear and concise description of what you expected to happen.
I want to move a servo, but not moving

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
```
# version
# Betaflight / STM32F405 (S405) 4.2.0 Apr 29 2020 / 08:08:49 (b480103d4) MSP API: 1.43

# config: YES

# start the command batch
batch start

board_name MATEKF405CTR
manufacturer_id MTKS

# name: x5

# resources
resource MOTOR 6 NONE
resource SERVO 1 A08

# servo
servo 0 1000 2000 1500 100 7

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature GPS
feature CHANNEL_FORWARDING

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# aux
aux 0 0 0 900 1600 0 0
aux 1 1 0 1400 1625 0 0
aux 2 46 3 1750 2100 0 0
aux 3 13 0 1750 2100 0 0
aux 4 35 2 900 1300 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 1 2 3 4

# master
set acc_calibration = -3,21,-145,1
set min_check = 1000
set max_check = 2000
set rssi_channel = 9
set failsafe_procedure = GPS-RESCUE
set vbat_scale = 111
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set channel_forwarding_start = 7
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 100
set gps_rescue_descent_dist = 2
###ERROR: diff: CORRUPTED CONFIG: gps_rescue_descent_dist = 2###
set gps_rescue_ground_speed = 1000
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 6
set yaw_deadband = 6
set pid_process_denom = 2
set osd_warn_visual_beeper = OFF
set osd_warn_fail_safe = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_dbm_alarm = 156
###ERROR: diff: CORRUPTED CONFIG: osd_rssi_dbm_alarm = 156###
set osd_cap_alarm = 1080
set osd_alt_alarm = 200
set osd_vbat_pos = 2401
set osd_rssi_pos = 2273
set osd_tim_2_pos = 2497
set osd_remaining_time_estimate_pos = 2327
set osd_flymode_pos = 2177
set osd_throttle_pos = 2465
set osd_vtx_channel_pos = 2145
set osd_crosshairs_pos = 2285
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2369
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2423
set osd_gps_lon_pos = 2160
set osd_gps_lat_pos = 2128
set osd_gps_sats_pos = 2508
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2391
set osd_flight_dist_pos = 2359
set osd_compass_bar_pos = 2099
set osd_altitude_pos = 2453
set osd_pidrate_profile_pos = 2113
set osd_disarmed_pos = 2410
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set vtx_band = 3
set vtx_channel = 3
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5665
set gyro_1_align_yaw = 2700
set name = x5

profile 0

# profile 0
set p_pitch = 52
set d_pitch = 51
set p_roll = 70
set d_roll = 70
set p_yaw = 70

rateprofile 0

# rateprofile 0
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 20
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 60
set tpa_rate = 35

# end the command batch
batch end

# ```

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.
```# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: SERVO 1
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: FREE
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: SERVO 1
TIM2: FREE
TIM3:
    CH1: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: TIMUP 3
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.
Matek F405 CTR

**Additional context**
Add any other context about the problem here.
I downgrade to 4.04 and it´s working. But when I try in 4.2 (RC1 and RC2) not working. 

Thanks,
Bráulio Lorenzo

mikeller, 2020-05-17:
@brauliolorenzo:

> `###ERROR: diff: CORRUPTED CONFIG: gps_rescue_descent_dist = 2###`

Your configuration is corrupted. Can you please reset your configuration to defaults (on the 'Setup' tab), and then try again?

brauliolorenzo, 2020-05-17:
Hi. I fix it.

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 May 13 2020 / 14:26:41 (6aeac3caa) MSP API: 1.43

# config: YES

# start the command batch
batch start

board_name MATEKF405CTR
manufacturer_id MTKS

# name: x5

# resources
resource MOTOR 6 NONE
resource SERVO 1 A08

# servo
servo 0 1000 2000 1500 100 7

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature GPS
feature CHANNEL_FORWARDING

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 3 2 115200 57600 0 115200

# aux
aux 0 0 0 900 1600 0 0
aux 1 1 0 1400 1625 0 0
aux 2 46 3 1750 2100 0 0
aux 3 13 0 1750 2100 0 0
aux 4 35 2 900 1300 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 1 2 3 4

# master
set acc_calibration = -3,21,-145,1
set min_check = 1000
set max_check = 2000
set rssi_channel = 9
set failsafe_procedure = GPS-RESCUE
set vbat_scale = 111
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set channel_forwarding_start = 7
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_rescue_initial_alt = 100
set gps_rescue_descent_dist = 30
set gps_rescue_ground_speed = 1000
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set deadband = 6
set yaw_deadband = 6
set pid_process_denom = 2
set osd_warn_visual_beeper = OFF
set osd_warn_fail_safe = OFF
set osd_warn_rssi = ON
set osd_warn_link_quality = ON
set osd_rssi_dbm_alarm = -20
set osd_cap_alarm = 1080
set osd_alt_alarm = 200
set osd_vbat_pos = 2401
set osd_rssi_pos = 2273
set osd_tim_2_pos = 2497
set osd_remaining_time_estimate_pos = 2327
set osd_flymode_pos = 2177
set osd_throttle_pos = 2465
set osd_vtx_channel_pos = 2145
set osd_crosshairs_pos = 2285
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2369
set osd_craft_name_pos = 2081
set osd_gps_speed_pos = 2423
set osd_gps_lon_pos = 2160
set osd_gps_lat_pos = 2128
set osd_gps_sats_pos = 2508
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2391
set osd_flight_dist_pos = 2359
set osd_compass_bar_pos = 2099
set osd_altitude_pos = 2453
set osd_pidrate_profile_pos = 2113
set osd_disarmed_pos = 2410
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_flight_dist = ON
set osd_stat_total_time = ON
set vtx_band = 3
set vtx_channel = 3
set vtx_power = 4
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5665
set gyro_1_align_yaw = 2700
set name = x5

profile 0

# profile 0
set p_pitch = 52
set d_pitch = 51
set p_roll = 70
set d_roll = 70
set p_yaw = 70

rateprofile 0

# rateprofile 0
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 20
set roll_srate = 50
set pitch_srate = 50
set yaw_srate = 60
set tpa_rate = 35

# end the command batch
batch end

Thanks

mikeller, 2020-06-21:
@brauliolorenzo: Sorry, your last comment was not entirely clear - did fixing the configuration fix your problem, or is channel forwarding still broken with this configuration?

closed 2020-08-01 by stale[bot]