betaflight/betaflight #9815: Custom F411 board: no VCP after Flashing

albydnc, 2020-05-17:
Hi everyone,
I'm designing a Poor Man's FC to build a fast prototyping and testing platform.
I've found a chinese product called Black Pill that uses a F411CEU6 [here the documentation](https://github.com/WeActTC/MiniF4-STM32F4x1). 
Since this MCU is widely used in commercial FCs like MATEKF411, I wanted to port Betaflight on it.

I have tried to flash MATEKF411 firmware directly, but no VCP was found after the flashing.
The only thing I am missing is the USB_DETECT, but I tried to trigger it by hand on the specified pin with no luck.

I have rebuilt it then without USB_DETECT pin and reducing components to a minimum (removing outputs, timers, OSD, baro and softserial), but I got no luck again.

How do I get the VCP working?

Thanks

mikeller, 2020-05-17:
@albydnc: It _should_ work even without USB_DETECT connected - weird that it does not. One thing that could be is that the boot address is set to something different than 0x08000000 which is what is the default, and where Betaflight expects it to be - you can check this with the ST Link utility.
The other possiblity is that this board uses a non-standard crystal, and USB is not recognised as it comes up with the wrong signal frequency.

albydnc, 2020-05-17:
The board is using a 25 MHz crystal on PH0 PH1 and a 32.768kHz on PC14 PC15.

The boot address is set correctly to 0x08000000 and VCP is working with a simple demo program on CubeIDE.

jflyper, 2020-05-17:
@albydnc
> The board is using a 25 MHz crystal on PH0 PH1

This is problematic.

Betaflight support for F411CE assumes the external crystal is 8MHz.

You have three options

1. Modify `src/main/startup/system_stm32f4xx.c` so that it supports 25MHz HSE source.

2. Replace the crystal to 8MHz.

3. Set `HSE_VALUE = 0` in `.mk` file in your target directory.
This will force the clock code to use HSI as system clock source.
It will not be as accurate as an external oscillator device, but should work to some extent.

albydnc, 2020-05-17:
@jflyper thank you for the clarification.
Since the board is off-the-shelf and I would like to keep it original, I did not try to replace crystal.
I tried the second options and it worked, but I do not think this could be the default since HSI is not accurate enough for USB standard.
I would like to follow the third option, but why do you not recommend it?

albydnc, 2020-05-17:
I have put HSE_VALUE = 25000000 in the target.mk and it worked fine without the need of any adjustment. Is it the correct approach?

mikeller, 2020-05-17:
@albydnc: If it's working then apparently yes. ;-) And USB is one of the things that is sensitive to clocking issues, so that's a good sign.

@jflyper: We can look into making HSE settable for F411, so this can be used with Unified Targets - having access to a development board that is only USD 5 is attractive.

mikeller, 2020-06-21:
Pinned while waiting for a board to arrive.

albydnc, 2020-07-04:
We are now testing the board design, I'll keep you posted

haslinghuis, 2022-05-08:
Closing as inactive

closed 2022-05-08 by haslinghuis