betaflight/betaflight #9817: Zero GPS satellites detected until BF rebooted

flythebluesky, 2020-05-17:
**Describe the bug**

OSD shows 0 satellites detected until BF is rebooted via OSD. 

**To Reproduce**

1. Install BN-880, KAKUTEF7 BF 4.2.0RC2.
2. Configure GPS & Magnetometer and complete configuration.
3. Power up the quad.
4. Observe that OSD shows 0 satellites detected after waiting a long time (e.g. 10 mins)
5. Reboot Betaflight from the OSD.
6. OSD now shows that more than 0 satellites detected (typically 10+)

**Expected behavior**

Expect to see # of sats detected without having to reboot BF. 

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F745 (S745) 4.2.0 May 13 2020 / 14:27:23 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id HBRO

# name: TBS7_2

# resources
resource MOTOR 1 E11
resource MOTOR 2 E09
resource MOTOR 3 B01
resource MOTOR 4 B00

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature GPS
feature TELEMETRY

# serial
serial 0 2048 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# aux
aux 0 0 0 1875 1950 0 0
aux 1 0 0 1825 1900 0 0
aux 2 46 1 1875 1950 0 0
aux 3 13 0 1925 2000 0 0
aux 4 13 0 1825 1900 0 0
aux 5 35 0 1725 1800 0 0
aux 6 35 0 1625 1700 0 0
aux 7 35 0 1575 1650 0 0
aux 8 36 0 1925 2000 0 0
aux 9 36 0 1825 1900 0 0

# master
set acc_calibration = -50,101,-162,1
set align_mag = CW180FLIP
set mag_align_pitch = 1800
set mag_align_yaw = 1800
set mag_hardware = AUTO
set mag_declination = -14460
set mag_calibration = 40,127,17
set baro_hardware = AUTO
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 7
set serialrx_provider = CRSF
set blackbox_p_ratio = 0
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set vbat_warning_cell_voltage = 360
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_auto_baud = ON
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_initial_alt = 60
set gps_rescue_descent_dist = 30
set gps_rescue_allow_arming_without_fix = ON
set osd_vbat_pos = 2497
set osd_rssi_pos = 2105
set osd_link_quality_pos = 2137
set osd_remaining_time_estimate_pos = 361
set osd_flymode_pos = 377
set osd_craft_name_pos = 2504
set osd_gps_speed_pos = 2487
set osd_gps_lon_pos = 2113
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2519
set osd_home_dir_pos = 2480
set osd_home_dist_pos = 2512
set osd_compass_bar_pos = 2095
set osd_altitude_pos = 2453
set osd_pidrate_profile_pos = 409
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2347
set osd_core_temp_pos = 2145
set osd_stat_endbatt = ON
set gyro_1_align_yaw = 2700
set gyro_rpm_notch_q = 700
set name = TBS7_2

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set ff_spike_limit = 70
set ff_smooth_factor = 20

rateprofile 0

# end the command batch
batch end
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: LED 1
A03: FREE
A04: SDCARD_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 4
B01: MOTOR 3
B02: FREE
B03: FREE
B04: FREE
B05: FREE
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: ADC_CURR
C03: ADC_BATT
C04: FREE
C05: FREE
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: SDCARD_DETECT
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: BEEPER
E00: FREE
E01: GYRO_EXTI
E02: SPI_SCK 4
E03: FREE
E04: GYRO_CS 1
E05: SPI_MISO 4
E06: SPI_MOSI 4
E07: FREE
E08: FREE
E09: MOTOR 2
E10: FREE
E11: MOTOR 1
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 5
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 5
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**

 - OS: MacOS, Chrome: 80.0.3987.116, Configurator: 10.7.0-RC1
 - Kakute F7 v1.5
 - UART 1 -- TBS Unify Pro HV, SmartAudio
 - UART 4 -- TBS Crossfire Nano RX
 - UART 6 -- BN-880 GPS w/magenetometer configured
 - UART 7 -- ESC telemetry, Tekko32 Metal 4-in-1 ESC
 - VIfly2 buzzer

mikeller, 2020-05-18:
@flythebluesky:

> `set gps_auto_baud = ON`

Try disabling the GPS auto baud setting, and using a fixed baud rate instead - this setting seems to be causing problems with some GPS / flight controller combinations:

https://github.com/betaflight/betaflight/issues/9652#issuecomment-628987910

flythebluesky, 2020-05-19:
Thanks @mikeller 

The resolution for BN-880 GPS is as follows:
```
set gps_auto_baud = OFF
save
```
and then go to the Ports tab and set Sensor Input baud rate to 38,400 for the GPS.

closed 2020-05-19 by flythebluesky