betaflight/betaflight #9818: GPS missing data on OSD

havanacore, 2020-05-18:
Today I completed the installation of the on-board electronics but I have a problem: in the OSD menu I cannot see some GPS parameters (eachine BN220), in particular distance, home point, speed. In addition, the height marks me with busted values (I attach photos with zero satellites received but the speech is similar to when it receives 11). Except for the busted height, simple lines appear instead of the missing data. Some idea?
[image]

this is my diff:

Entering CLI Mode, type 'exit' to return, or 'help'

# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / MATEKF405 (MKF4) 4.0.3 Jun  1 2019 / 11:36:38 (094cfc956) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF405
manufacturer_id 

# name

# resources
resource MOTOR 1 C08
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C07

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2 115200 9600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1300 1600 0 0
aux 1 1 2 1800 2100 0 0
aux 2 27 1 1700 2100 0 0
aux 3 13 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -11,76,-191
set mag_hardware = NONE
set mid_rc = 1499
set dshot_idle_value = 150
set motor_pwm_protocol = DSHOT600
set failsafe_throttle = 1260
set small_angle = 180
set gps_provider = UBLOX
set gps_rescue_descent_dist = 30
set gps_rescue_min_sats = 6
set osd_vbat_pos = 2401
set osd_rssi_pos = 122
set osd_tim_2_pos = 2263
set osd_remaining_time_estimate_pos = 2465
set osd_flymode_pos = 2369
set osd_throttle_pos = 2337
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_gps_speed_pos = 2486
set osd_gps_lon_pos = 2128
set osd_gps_lat_pos = 2113
set osd_gps_sats_pos = 2168
set osd_home_dir_pos = 2158
set osd_home_dist_pos = 2222
set osd_flight_dist_pos = 2455
set osd_altitude_pos = 2422
set osd_warnings_pos = 385
set osd_avg_cell_voltage_pos = 2433
set osd_pit_ang_pos = 2231
set osd_rol_ang_pos = 2199
set osd_battery_usage_pos = 2496
set osd_nvario_pos = 2305
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON

# profile
profile 0

set acro_trainer_angle_limit = 50

# rateprofile
rateprofile 0

set yaw_rc_rate = 120
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 81

# end the command batch
batch end

mikeller, 2020-05-18:
@havanacore: The reason you don't get any GPS related information is because you don't have a GPS fix ( see the '0' in the top right corner for number of satellites) - and the reason for this is, judging from the picture, that you are trying indoors. Try outdoors, or at least very close to a window.

havanacore, 2020-05-18:
I solved a problem.  once the engines are armed the distance data appear.  the problem is height: when the engines are armed, the measurement is reset but then the data collected are not precise to the point that once landed, it does not mark zero.  is there a way to calibrate the quota also through CLI?

mikeller, 2020-05-19:
@havanacore: The problem you describe is inherent to the use of a barometric pressure sensor - these devices are sensitive to temperature changes and wind, so on-board on a multirotor is a bad place for them. You can mitigate some of this by placing open cell foam over the sensor.

closed 2020-06-27 by stale[bot]