betaflight/betaflight #9821: Mamba F405 mk2 - Calibrating MSC

JoshWithrow, 2020-05-18:
[NEW.txt](https://github.com/betaflight/betaflight/files/4646331/NEW.txt)
[NEW_1.txt](https://github.com/betaflight/betaflight/files/4646332/NEW_1.txt)

**Describe the bug**
- Using Betaflight Configurator 10.7.0-RC1.
- Mamba F405 Mk2.  
- Target is documented as FuryF4OSD (Legacy).  
- Flashed Betaflight 4.1.1 Legacy.  Full chip erase, reset settings multiple times both prior to and after flashing to 4.1.1.  Receiving arming flag "Calibrating MSP".  Can't figure out how to resolve.

In a related note, I had this problem on another Mamba F405 mk2 FC, but I had flashed the wrong image.  Once I used the correct image (FuryF4OSD Legacy), the problem went away.  Not sure why this one won't resolve.

**To Reproduce**
With Mamba F405 mk2, flash betaflight 4.1.1 using FuryF4OSD Legacy image.

**Expected behavior**
I expected to flash the FC with updated firmware, to be able to finish configuring the drone, and to be able to fly it.

**Flight controller configuration**
```
# version
# Betaflight / FURYF4OSD (FY4O) 4.1.1 Nov 15 2019 / 12:26:24 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# feature
feature TELEMETRY

# serial
serial 0 0 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 35 1 1700 2100 0 0

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set serialrx_provider = CRSF
set blackbox_device = NONE
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set current_meter = ADC
set yaw_motors_reversed = ON
set deadband = 4
set yaw_deadband = 4
set pid_process_denom = 1
set gyro_2_spibus = 1

profile 0

rateprofile 0

# end the command batch
batch end
```
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: USB_DETECT
C06: SERIAL_TX 6
C07: SERIAL_RX 6
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1:
    CH3: MOTOR 3
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 1
TIM3:
    CH3: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 3
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Mamba F405 mk2 purchased from Pyroflip RC.  Purchased as a stack combo, F405 mk2 F50 4in1 BLHeli_S ESC
 - Caddx Ratel, ImmersionRC Tramp HV, TBS Crossfire Nano RX, 1000uf 35v Cap, iFlight Xing-E 2208 1800kv
 - All soldered

**Additional context**
Add any other context about the problem here.

mikeller, 2020-05-19:
@JoshWithrow:

> Flashed Betaflight 4.1.1 Legacy. Full chip erase, reset settings multiple times both prior to and after flashing to 4.1.1. Receiving arming flag "Calibrating MSP"

The 'calibrating' message is indicative of a broken gyro that is producing too much noise to make it possible to calibrate - if this was the case from new then this is a case for a warranty return / refund.

JoshWithrow, 2020-05-19:
Good to know.  Thank you!

illmillrig, 2020-05-27:
did you accept the default configuration after flashing?  I think if you did, then the gyro_1_align_yaw would be set to 1800, but it doesn't appear in your diff.

I fly this FC, and I see it in my diff `set gyro_1_align_yaw = 1800`

JoshWithrow, 2020-05-28:
Mine didn't ask me to set defaults.  Honestly, I was feeling a bit weird
about everything.  The vendor seems to think the gyro is bad and is sending
me another board.  I just hope it doesn't have the same problem.  But I'll
try and see if it nets any behavioral differences.

illmillrig, 2020-05-28:
> 
> Mine didn't ask me to set defaults. Honestly, I was feeling a bit weird about everything. The vendor seems to think the gyro is bad and is sending me another board. I just hope it doesn't have the same problem. But I'll try and see if it nets any behavioral differences.
> […](#)

what does the gyro say in betaflight? go into the motors tab, and change it to 'gyro'.  play around with the scale a bit. do all three axes appear similar in regards to noise? what happens when you tilt the quad forward or to the side without yaw?  It should be pretty easy to tell if the gyro is damaged.  if not, I'd try aligning it to 180 like I mentioned.  I have this in my diff, and my quad flies great.  It would not fly at all if this setting was wrong.

closed 2020-07-04 by stale[bot]