betaflight/betaflight #9822: Constant prop explosions

cmmoser, 2020-05-18:
Hi guys

I am flying a Transtec Laser HD with the following configuration:
DJI digital FPV
Hobbywing X-Race Pro 2207, 2650kV
T-Motor F4, F55 HD stack
Beitian Micro M8N GPS
Gemfan Hurricane 51466 or 51499
Tattu 4S 1300

Since today I have made approximately 150 flights.
Freestyle and also cruising flights.

The problem started when I was using Gemfan 5152 Flash and Gemfan 5055 Hulkie props. I wanted to test these props to get higher topspeed. 

At the second flight with the 5252 props all of a sudden one prop exploded and I crashed. Only the center of the prop was still at the motor. 
Then the same happend with the 5055. Not at the same position. 

I went back to my 51466 props. With these props I was flying approximately 130 flights without any problems. Eventhough the props where already looking like o dog would have chewed on them. 
Then the same happened with these props also. Explosion, crash.

The motors are running smooth and the bearings are good. No loose motors.

I put a brand new pair of 51466 and it happened again.
And the strange thing is that it mostly does not happen when I am giving full throttle.

I have not yet analyzed the blackbox files but will upload them as soon as I put them together with the viedeo.

I was bending a bit at the arm of the Quad and had the impression that they are quite soft on torsion. Is it possible that the 6mm carbon gets worn out after 130 flights already???

Does anyone of you made the same experience?

Which parameters should I show when I mix the blackbox file with the video to find the problem?

Thank you very much for your inputs

mikeller, 2020-05-19:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2020-05-19 by mikeller

cmmoser, 2020-05-20:
Hi Michael
I thougt that you could think that way.I opened the tread anyway because my main question which is definetly a betaflight topic is how I can analyze the blackbox to find out where the problems come from.
Thank you very muchChristian

_______________________________

DISCLAIMER
The information in this email and any attachments is confidential and intended only for use by the intended recipient(s). If you are not the intended recipient of this message, please notify the sender immediately, and do not disclose or make copies of this message.

mikeller, 2020-05-20:
@cmmoser: This is not about this being Betaflight or not - GitHub is for discussion around **development** of Betaflight - for everything else my comment above applies.

cmmoser, 2020-05-20:
Hi Michael
Ok, sorry, I was not aware that github is only about the development.
Cheers Christian

_______________________________

DISCLAIMER
The information in this email and any attachments is confidential and intended only for use by the intended recipient(s). If you are not the intended recipient of this message, please notify the sender immediately, and do not disclose or make copies of this message.