betaflight/betaflight #9823: Mamba F405 Mk 2 No OSD in Unified target

MaloyOram, 2020-05-18:
**Describe the bug**
- No OSD in the unified targets. 
- Tried versions 4.1.6 and 4.2.0
- OSD appears in the legacy target on Betaflight 4.1.1

**Expected behaviour**
- No Betaflight logo on power up or OSD elements when added.
- I've flashed between legacy and unified target several times - always an OSD on legacy with nothing in the unified target.

**Flight controller configuration**
**Legacy 4.1.1 diff**
```
# diff

# version
# Betaflight / FURYF4OSD (FY4O) 4.1.1 Nov 15 2019 / 12:26:24 (1e5e3d369) MSP API: 1.42

# start the command batch
batch start

board_name FURYF4OSD

# name: Mamba

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 68 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200

# aux
aux 0 0 0 1775 2100 0 0
aux 1 1 5 1800 2100 0 0
aux 2 13 3 1300 1700 0 0
aux 3 35 3 1775 2100 0 0
aux 4 36 5 1300 2100 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 800

# master
set gyro_sync_denom = 2
set gyro_lowpass2_hz = 350
set dyn_lpf_gyro_min_hz = 280
set dyn_lpf_gyro_max_hz = 700
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 7
set blackbox_p_ratio = 128
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set ibatv_scale = 255
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set osd_rssi_pos = 2522
set osd_tim_2_pos = 2528
set osd_flymode_pos = 2553
set osd_throttle_pos = 2489
set osd_crosshairs_pos = 2253
set osd_craft_name_pos = 2540
set osd_avg_cell_voltage_pos = 2496
set osd_disarmed_pos = 2315
set osd_flip_arrow_pos = 2286
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_used_mah = OFF
set osd_stat_bb_no = OFF
set debug_mode = GYRO_FILTERED
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 3
set vtx_freq = 5843
set name = Mamba

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 98
set dyn_lpf_dterm_max_hz = 238
set dterm_lowpass2_hz = 210
set idle_min_rpm = 20

rateprofile 0

# end the command batch
batch end

#
```

**Legacy 4.1.1 Resource All**
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D02: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 1
    CH2: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 1
DMA2 Stream 3: DSHOT_BITBANG 2
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

**Unified 4.2.0 Diff**
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 May 13 2020 / 14:26:41 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: FURYF4OSD, version: 3a35e73b, date: 2019-09-30T05:46:12Z

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id DIAT

# master
set osd_tim_2_pos = 2412
set osd_crosshairs_pos = 2253
set osd_disarmed_pos = 2282
set gyro_1_align_yaw = 1800

profile 0

rateprofile 0

# end the command batch
batch end

#
```

**Unified 4.2.0 Resource All**
```
resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 1
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: BEEPER
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: MOTOR 2
B01: MOTOR 3
B02: FREE
B03: FLASH_CS
B04: LED 2
B05: LED 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: CAMERA_CONTROL
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH3: MOTOR 3
TIM2:
    CH3: MOTOR 4
    CH4: MOTOR 1
TIM3:
    CH3: MOTOR 2
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: MOTOR 4
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: MOTOR 1
DMA1 Stream 7: MOTOR 2
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: MOTOR 3
DMA2 Stream 7: FREE

#
```

**Setup / Versions**
 - Flight controller Diatone Mamba F405 Mk 2

**Additional context**
Appreciated if someone could have a look - looking forward to getting 4.2 on this quad as it's amazing on my others. Thanks

Nekrofago, 2020-05-19:
Could you try to "load font" on OSD tab with battery plugged in. 

I have had read somewhere that Mamba FC doesnt power up OSD chip by USB only and for BF 4.1 and above you need to load font in order to see again OSD. 

You have told that it works on 4.1.1 legacy but i had the same problem with my Diatone R239 and that was de solution. Maybe you could give it a shot

MaloyOram, 2020-05-19:
> Could you try to "load font" on OSD tab with battery plugged in.
>
> I have had read somewhere that Mamba FC doesnt power up OSD chip by USB only and for BF 4.1 and above you need to load font in order [...]

Thanks for the suggestion Nekrofago. 

Just tried in configurator 10.7 and on the 4.2 firmware but the font manager button remains greyed out - even with the battery plugged in. The configurator is telling me that no OSD chip has been recognised. Changed the font style just to see if that provoked it into appearing but it didn't work either.

mikeller, 2020-05-19:
@MaloyOram: I just tested with a MAMBAF405 MK2, and OSD works just fine with the FURYF4OSD target on 4.2.0-RC2 when a battery is connected to Vcc. If yours does not work then this is most likely due to a hardware defect (potentially a bad voltage regulator).

MaloyOram, 2020-05-19:
> @MaloyOram: I just tested with a MAMBAF405 MK2, and OSD works just fine with the FURYF4OSD target on 4.2.0-RC2 when a battery is connected to Vcc. If yours does not work then this is most likely due to a hardware defect (potentially a bad voltage regulator).

Thanks for the message @mikeller. If you say that it works on your FC then I'll happily close the issue and accept that it's my FC. 

Why would you think that the OSD works on the legacy target though but not the unified target if it's a hardware issue. Wouldn't you expect it work on neither? Just trying to expand my knowledge.

mikeller, 2020-05-19:
@MaloyOram:

> Why would you think that the OSD works on the legacy target though but not the unified target if it's a hardware issue. Wouldn't you expect it work on neither? Just trying to expand my knowledge.

No idea honestly. Since my board does not show this behaviour it does not look it has got something to do with the design of the board (unless multiple different designs were sold under the 'Mamba F405 MK2 FC' label). So all that is left is a defect in your board causing the OSD chip to malfunction in 4.2 due to some otherwise inconsequential change.

DedSecV, 2020-05-20:
I have the same issue. OSD is not recognized, even when battery is connected. A rollback on 4.1.2 made it work again.

Target: DIAT/FURYF4OSD(STM32F405)

MaloyOram, 2020-05-20:
> I have the same issue. OSD is not recognized, even when battery is connected. A rollback on 4.1.2 made it work again.
> 
> Target: DIAT/FURYF4OSD(STM32F405)

Ah looks like we do have the same issue. 

Rolled back to 4.1.6 but still no OSD. Plugged a battery in and uploaded a font. OSD appears. 

So it doesn't look to be a legacy vs unified target issue. 

It does look to be a 4.2 issue - whether we've got dodgy voltage regulators like mikeller suggests or something else.

vicewize, 2020-05-22:
Happens with me as well. No OSD using the MambaF405 target on 4.2 if I roll back to any version of 4.1 it works fine.

vicewize, 2020-05-23:
I fixed it by changing this setting from AUTO to MAX7456.

 set osd_displayport_device = MAX7456

MaloyOram, 2020-05-23:
> I fixed it by changing this setting from AUTO to MAX7456.
> set osd_displayport_device = MAX7456

Fingers crossed then. I'll have a look in a little bit and see if this makes a difference.

MaloyOram, 2020-05-23:
@vicewize Sweet. That's worked.

I set the osd_display_device to max7456 but the font manager button remain inactive. Set the osd_display_device back to auto and this bumped it back to life. 

Once this was done uploading the font was easy.

Much appreciated for the heads up. 

I can see through the issues that this has been a recent item being worked on.

vicewize, 2020-05-23:
Glad it worked out!

mikeller, 2020-05-24:
@vicewize:

> I fixed it by changing this setting from AUTO to MAX7456.

Good point, thanks. Fixed in betaflight/unified-targets#277

@MaloyOram: This probably also explains why this worked for me and did not work for you - without `set osd_displayport_device = MAX7456` the OSD will work if the battery is connected _before_ USB, but not if it is connected _after_.

closed 2020-05-24 by mikeller

Thegame-ai, 2020-09-04:
I know this is closed, but just wanted to say thank you for this. I ended up going back and rewiring my rx and tx on the caddx vista. osd popped right up. hopefully if someone stumbles upon this later it might help :D

CodesOfGabriel, 2024-09-30:
> Ah looks like we do have the same issue.
>
> Rolled back to 4.1.6 but still no OSD. Plugged a battery in and uploaded a font. OSD appears.
>
> So it doesn't look to be a legacy vs unified target issue. [...]

I have a Mamba f405 mk2 v2 and i was with the same problem in 4.5.1 version. I just downgrade a update to 4.3 and worked! (2024)

ozrs, 2024-11-04:
flashed 4.5.1 to a new (old stock) Mamba Mini Mk2, and had no OSD (font manager greyed out). Used:

set osd_displayport_device = MAX7456

and that fixed it.

haslinghuis, 2024-11-04:
For analog builds leave out `OSD_HD` in other options while flashing.