betaflight/betaflight #9825: Problems with secondary MSP through UART

adragoset, 2020-05-19:
**Describe the bug**
Having a secondary MSP through UART appears to be broken when using bluetooth through a crossfire receiver or generic bluetooth module.

**To Reproduce**
Follow setup here for crossfire.
https://www.youtube.com/watch?v=ZgPOEiP9U_w

Generic bluetooth setup secondary msp on uart with proper baud rate for bluetooth module.

**Expected behavior**
I should be able to pair bluetooth to windows or android and establish a serial bridge.

**Flight controller configuration**
```
A00: CAMERA_CONTROL
A01: LED_STRIP
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: GYRO_EXTI
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: GYRO_CS 1
B00: FREE
B01: FREE
B02: GYRO_EXTI
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: FREE
B07: FREE
B08: PINIO 3
B09: FLASH_CS
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: ADC_CURR
C03: GYRO_CS 2
C04: LED 1
C05: FREE
C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3
C10: SERIAL_TX 4
C11: SERIAL_RX 4
C12: FREE
C13: PINIO 1
C14: PINIO 2
C15: BEEPER
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2:
    CH2: LED_STRIP
TIM3: FREE
TIM4: FREE
TIM5:
    CH1: CAMERA_CONTROL
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: MOTOR 2
    CH2: MOTOR 4
    CH3: MOTOR 1
    CH4: MOTOR 3
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: LED_STRIP
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: TIMUP 8
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```
**Setup / Versions**
 - FC: GEP-STABLE PRO F7 DUAL BL32 35A - Betaflight 4.1.6
 - TX: Frsky Horus x12s - OPENTX:2.3.7
 - TX: TBS crossfire:3.2.8
 - RX: TBS crossfire Nano v2:3.2.8
 - TBS CROSSFIRE nano CH1/CH2 to Serial RX UART4 CH3/CH4 wired to Uart1 Tx/rX respectively and set to serial bridge in the crossfire module and baudrate 57600
**Additional context**

mikeller, 2020-05-19:
@adragoset: Can you please supply a CLI `diff` as you were asked to do when opening the pull request?

So do I understand this right that you are trying to connect to the flight controller by connecting to the CRSF over bluetooth?

adragoset, 2020-05-19:
CLI diff

```
# 
# Building AutoComplete Cache ... Done!
# 
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.6 Apr 25 2020 / 05:12:26 (283bda8bf) MSP API: 1.42
# manufacturer_id: GEPR   board_name: EXF722DUAL   custom defaults: YES

# start the command batch
batch start

board_name EXF722DUAL
manufacturer_id GEPR

# name: CinePro F7

# feature
feature -SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature TRANSPONDER

# serial
serial 0 1 57600 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# led
led 0 5,4::CTOBIW:0
led 1 6,4::CTOBIW:0

# aux
aux 0 0 0 1725 1875 0 0
aux 1 1 1 1850 1950 0 0
aux 2 2 1 1700 1800 0 0
aux 3 27 0 1725 1800 0 0
aux 4 27 0 1875 1950 0 0
aux 5 13 1 1800 1875 0 0
aux 6 13 1 1675 1750 0 0
aux 7 13 1 1150 1225 0 0
aux 8 13 1 950 1025 0 0
aux 9 13 1 1550 1600 0 0
aux 10 35 1 1575 1675 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 BAND_D   D CUSTOM  5362 5399 5436 5473 5510 5547 5584 5621
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

# master
set gyro_sync_denom = 2
set acc_trim_pitch = -2
set acc_trim_roll = 2
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 11
set serialrx_provider = CRSF
set motor_pwm_protocol = DSHOT300
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255
set gyro_2_align_yaw = 900
set name = CinePro F7

profile 0

# profile 0
set p_pitch = 38
set i_pitch = 72
set d_pitch = 34
set p_roll = 35
set i_roll = 65
set d_roll = 31
set p_yaw = 49
set i_yaw = 65
set level_limit = 25

rateprofile 0

# end the command batch
batch end
```
Crossfire has functionality too use the crossfire link and the crossfire nano as a serial bridge. I can also replicate the failure with a standard bluetooth serial bridge module hooked to a uart on the FC as well though.

IB1387, 2020-05-21:
Hi. I don't know why it's not working for you, but at least I can tell you that what you are trying to do should work in 4.1. I'm running the large XF module and use it to connect BF Configurator to the matek F722 mini via bluetooth. 
If it helps, I uploaded my diff here: https://pastebin.com/W2UJZV8h

Maybe try updating your XF version, I just skipped through your video, but the version of XF you are using looked old. I know for a fact that there were problems with the mavlink stuff some time ago, maybe TBS fixed some stuff with the serial bridge too.

mikeller, 2020-05-24:
@adragoset:

> Crossfire has functionality too use the crossfire link and the crossfire nano as a serial bridge. I can also replicate the failure with a standard bluetooth serial bridge module hooked to a uart on the FC as well though.

There is nothing obviously wrong in your configuration. But if you are using the bluetooth serial bridge module you'll have to make sure you change the baud rate of the MSP port to match whatever is used by the module - they commonly come configured at 115200.

Also, @IB1387's suggestion to update the Crossfire firmware to the latest version seems to be worth a try.

CorruptName, 2020-05-30:
I'm having this same issue on Transtec F7 FC. I have no issues doing this with INAV, only Betaflight.

closed 2020-07-06 by stale[bot]