betaflight/betaflight #9826: Matek F722 mini target no Battery Telemetry

delightfpv, 2020-05-20:
I recently flashed the Betaflight 4.2 RC2 and started having problems with my current sensing.
I started troubleshooting. After quite some time I noticed that by setting the voltage sensor source to ESC I was getting voltage telemetry but no current telemetry. After some more troubleshooting I rolled back to BF 4.1 and everything was working fine (voltage and current sensoring was working when  the sources were set as "onboard sensors". So I confirmed that my FC and ESC weren't the problem. It was a software problem. I also checked the resource mapping and I found no issues there.

Here is the dump from my FC:

```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 May 13 2020 / 14:27:44 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722MINI, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# resources
resource MOTOR 5 NONE
resource PINIO 1 A15

# dma
dma ADC 1 NONE

# feature
feature ESC_SENSOR

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 1 19200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# aux
aux 0 0 0 1400 1600 0 0
aux 1 13 0 900 1100 0 0
aux 2 35 2 900 1100 0 0
aux 3 36 1 1900 2100 0 0
aux 4 49 2 1900 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 500 800

# master
set acc_calibration = 121,-8,344,1
set fpv_mix_degrees = 10
set serialrx_provider = CRSF
set dshot_idle_value = 450
set dshot_bidir = ON
set vbat_min_cell_voltage = 341
set vbat_scale = 117
set ibata_scale = 218
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set osd_warn_link_quality = ON
set osd_rssi_dbm_alarm = 5
set osd_cap_alarm = 1200
set osd_vbat_pos = 2497
set osd_rssi_pos = 2113
set osd_link_quality_pos = 2081
set osd_tim_1_pos = 460
set osd_tim_2_pos = 2507
set osd_current_pos = 2518
set osd_mah_drawn_pos = 2488
set osd_avg_cell_voltage_pos = 2465
set osd_disarmed_pos = 2283
set osd_esc_tmp_pos = 2135
set osd_core_temp_pos = 2103
set osd_stat_battery = ON
set osd_stat_min_link_quality = ON
set vtx_band = 5
set vtx_channel = 6
set vtx_power = 1
set vtx_freq = 5843
set pinio_config = 129,1,1,1
set pinio_box = 0,255,255,255

profile 1

# profile 1
set vbat_pid_gain = ON
set launch_angle_limit = 80
set idle_min_rpm = 20

rateprofile 0

# rateprofile 0
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_expo = 18
set pitch_expo = 18
set yaw_expo = 18
set roll_srate = 83
set pitch_srate = 83
set yaw_srate = 80

# end the command batch
batch end
```

```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: PINIO 1
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: SERIAL_RX 6
C08: FREE
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

Quad setup:
FC: Mateksys F722 mini
ESC: Holybro Tekko32F3 Metal 4in1 ESC(65A)
Cam: Ethix micro Cam
VTX: Rush Tank mini
Motors: Ethix Stout V3's

I really hope it gets fixed soon

mikeller, 2020-05-24:
@delightfpv: What happens if you use 4.2.0-RC2 and set 'Voltage Meter Source' and 'Current Meter Source' to 'Onboard ADC'? Do you get voltage and / or current readings in this case?

delightfpv, 2020-05-24:
When I set both to "onboard ADC" I get no readings at all... Only If I set the voltage source to ESC Telem.. Then I get a voltage reading at least.

mikeller, 2020-05-24:
@delightfpv: This is interesting. Can you please provide the version, target, and a `diff` and `resource show all` for the 4.1 target that has 'Onboard ADC' working for you?

delightfpv, 2020-05-24:
Sure. So, I flashed to BF 4.1.6 and everything (voltage and current) were working just fine with "Onboard ADC" selected.

Here are the BF 4.1.6 informations:

Board: MTKS/MATEKF722MINI(STM32F7X2)
Target: MATEKF722MINI
diff:
```
# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.1.6 Apr 25 2020 / 05:12:26 (283bda8bf) MSP API: 1.42
# manufacturer_id: MTKS   board_name: MATEKF722MINI   custom defaults: YES

# start the command batch
batch start

board_name MATEKF722MINI
manufacturer_id MTKS

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 64 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# master
set serialrx_provider = CRSF
set dshot_bidir = ON
set beeper_dshot_beacon_tone = 3
set small_angle = 180
set pid_process_denom = 1

profile 0

rateprofile 0

# end the command batch
batch end
```

resource show all:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: USB
A12: USB
A13: LED 2
A14: LED 1
A15: FREE
B00: MOTOR 3
B01: MOTOR 4
B02: GYRO_CS 1
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: FREE
B07: FREE
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: PINIO 1
C09: PINIO 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: BEEPER
C14: FREE
C15: GYRO_CS 2
D00: FREE
D01: FREE
D02: FLASH_CS
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

mikeller, 2020-05-24:
@delightfpv: One more thing to confirm: With 4.2, when using 'Onboard ADC', when you have got a battery connected while connected to configurator, you don't see the voltage / current in the 'Setup' or 'Battery' tab? Or is it just the telemetry output on the TX that is affected?

delightfpv, 2020-05-24:
I don't see ANY battery information. The most confusing thing is that if I select ESC as the source only the voltage is working. Not the current. In either case, the current is not working in BF 4.2 for me. But if I roll back to 4.1.6 everything is working fine.

TheIsotopes, 2020-05-24:
I have two matekf7mini with bf 4.2 rc2 in use and no problems with onboard adc.

delightfpv, 2020-05-24:
But why isn't mine working? I couldn't be a hardware problem, because it works just fine in BF 4.1.6?

mikeller, 2020-05-24:
@delightfpv:

> The most confusing thing is that if I select ESC as the source only the voltage is working. 

This can probably be explained by the fact that your ESC supplies voltage information, but not current information over ESC telemetry, as it does not have the 4 independent shunts required to measure current.

delightfpv, 2020-05-24:
But how do I get it working then? And why is it working in BF 4.1? I mean, IF the ESC doesn't deliver current information through the wire....then why was it working?

delightfpv, 2020-05-24:
Here are some pictures: 
My configurator is in german ;)

[image]
[image]
[image]

TheIsotopes, 2020-05-24:
have you ever tried to configure from scratch without reading a config file?

delightfpv, 2020-05-24:
I got the problem. I flashed BF 4.2 RC2 again and tried to configure the quad again and take my time. And it was working. Normally I past the osd Information and the general config in the cli and it was working fine and all... But apperantly it isn't working in BF 4.2.
I feel kinda dumb. I'll close this issue.

closed 2020-05-24 by delightfpv

pliashkou, 2020-09-26:
Happens the same on MATEK F722 STD, Flashing back to 4.1.6 helped