betaflight/betaflight #9829: dont let gyro_rpm_notch_q be set LESS than default of 500

krunked-fpv, 2020-05-21:
Gyro_rpm_notch_q shouldn't be allowed to be set below the default of 500. I've had it on three occasions now where either I'm typing too quickly in the CLI or saving too quickly but the last zero isn't recognized and I accidentally save this parameter as 80....

This causes an instant fly away and will almost melt your motors... I found that out the hard way...

krunked-fpv, 2020-05-21:
@mikeller how is this missing information..... It's straight forward...... The gyro_rpm_notch_q being able to be set to low is dangerous and should not be allowed. Sigh...

mikeller, 2020-05-21:
Please do as you were asked to do and provide the information you were asked to provide when opening this issue.

krunked-fpv, 2020-05-21:
in my opinion the information is there....  here it is even more clearly for you seeing English is hard. 

**Is your feature request related to a problem? Please describe.**
 I've had it on three occasions now where either I'm typing too quickly in the CLI or saving too quickly but the last zero isn't recognized and I accidentally save this parameter as 80....

**Describe the solution you'd like**
Gyro_rpm_notch_q shouldn't be allowed to be set below the default of 500. 

**Describe alternatives you've considered**
typing slower...

**Additional context**
context being its unsafe and dangerous to be ABLE to set the gyro_rpm_notch_q this low.

ctzsnooze, 2020-05-22:
Thanks @krunked-fpv for this report.  It's very useful.

We can, and I think should, limit the lowest allowed rpm Q value to 500 in 4.2.  

In 4.1, 250 was needed on some setups, due to loootime jitter limitations.

I think this should be considered a bug report since you indicate it can cause immediate flyaway on arming.

krunked-fpv, 2020-05-22:
@ctzsnooze no problemo, Chris..  shall i move it over to bug report?  the damn thing fluttered like a butterfly on crack and i was able to basically do a bunch of loops and get myself close enuf to the ground and disarm softly in grass at the bottom of the fluttering flyaway -- motors were so hot almost burned me!   definitely coulda toasted them!  either i type too fast, or the configurator lags and missed the last zero before i hit enter and typed save and hit enter.  but ive basically done this to myself 3x now, lol.

butterSnake, 2020-05-22:
I suspect at 80 it was disabled which would explains the fly away

mikeller, 2020-05-24:
Closing due to refusal of reporter to collaborate.

closed 2020-05-24 by mikeller

ctzsnooze, 2020-05-24:
see https://github.com/betaflight/betaflight/pull/9841