betaflight/betaflight #9830: Add log for usable reelsteadygo gyro information

sevet, 2020-05-21:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

Currently only gopro 6 and 8 really support rsg, 6 needs an outdated version to work. You also loose other features like the gopro gps, the gopro session which is best for fpv kinda works but you need to do a very good soft mouny for the camera, other cameras simply dont work.....

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

Why not log a reelsteadygo compatible gyro data on flash if possible or sd logger? 
It will let us even stabilize every camera like gopro session, gp7, dji digital, split cams and even DVR recordings  ;).....

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

Buy many many gp6 cheap, decase them and use as naked gp6 but its an expensive solution.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

delightfpv, 2020-05-22:
That Idea is kinda sick ngl. I hope this get's implemented!

delightfpv, 2020-05-22:
I'm not sure, but I think you need to add that to the "pull-requests" and not the "issues" section

IllusionFpv, 2020-05-22:
@delightfpv with a pull request you add code in betaflight, this one is a `feature request` (IIRC already discussed) and goes here

delightfpv, 2020-05-22:
Ok, now I got it ;)
Thanks @IllusionFpv

mikeller, 2020-05-24:
@sevet: I think the information that is required (essentially the gyro information) is already logged in blackbox.

There already is a request discussing the options to convert blackbox logs into the format required for reelsteady: betaflight/blackbox-log-viewer#418 Closing this as duplicate

closed 2020-05-24 by mikeller