betaflight/betaflight #9831: Just Curious as to Why

rekkelley, 2020-05-21:
Just curious as to why this must be done on a Mac with Catalina. 

Changes to the security model used in the latest versions of MacOS X 10.14 (Mojave) and 10.15 (Catalina) mean that the operating system will show an error message ('"Betaflight Blackbox Explorer.app" is damaged and can’t be opened. You should move it to the Trash.') when trying to install the application. To work around this, run the following command in a terminal before installing: sudo xattr -rd com.apple.quarantine /Applications/Betaflight\ Blackbox\ Explorer.app.

mikeller, 2020-05-24:
@rekkelley:

> Changes to the security model used in the latest versions of MacOS X 10.14 (Mojave) and 10.15 (Catalina) mean 

There you go for the 'why'. If you want to know why Apple changed this you will have to contact them.

rekkelley, 2020-05-24:
Why should I have to contact Apple. This seems to be the only program I have loaded that has this problem. Seems like the developersof these programs should make this work correctly like all the other developers have.

mikeller, 2020-05-24:
@rekkelley: As I said above, it is Apple, not us, who introduced the policy that you have to pay them annually to have your software not being blocked by their operating system. If you want to take over the cost of USD 99 annually for the required fee to prevent this then this can be arranged.

rekkelley, 2020-05-24:
That is what I thought was the problem.
Thanks

djhashh, 2020-05-26:
Tact much? Please be kind. I'm not suggesting users sugar coat themselves. Being 'short' can undermine mutual report. It is certainly another example of Apple and other corporate tyranny. Excellent project none the less. thanks

edit: bruh u sound like a dick. u hangry huh

rekkelley, 2020-05-26:
Examples of unacceptable behavior by participants include:

The use of sexualized language or imagery and unwelcome sexual attention or advances
Trolling, insulting/derogatory comments, and personal or political attacks
Public or private harassment
Publishing others' private information, such as a physical or electronic address, without explicit permission
Other conduct which could reasonably be considered inappropriate in a professional setting

closed 2020-05-26 by rekkelley