betaflight/betaflight #9833: OSD ESC Temp not showing in 4.2.0RC2

dzett, 2020-05-22:
**Describe the bug**
After moving from 4.0.6 to 4.2.0-RC2, ESC Temperature is no longer shown on the OSD although configured and shown in preview in Configurator (10.7.0-RC1). 
For testing I added the RPM values coming through telemetry to the OSD - they are visible and get updated.

Thanks for looking into this!

**To Reproduce**
Add ESC Temperature to the OSD configuration.

**Expected behavior**
To see the ESC temperature in the OSD.

**Flight controller configuration**
```
# 

# diff

# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 May 13 2020 / 14:27:44 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: DIAT, board_name: MAMBAF722, version: a5269145, date: 2020-04-09T22:32:24Z

# start the command batch
batch start

board_name MAMBAF722
manufacturer_id DIAT

# name: 3in

# feature
feature -RX_PARALLEL_PWM

# serial
serial 1 1024 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 3 0 19200 57600 0 115200
serial 4 32 115200 57600 0 115200

# aux
aux 0 0 0 1400 2100 0 0
aux 1 1 0 1650 1850 0 0
aux 2 2 0 1875 2100 0 0
aux 3 13 3 1875 2100 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 1
vtxtable powervalues 14
vtxtable powerlabels 25

# master
set acc_calibration = -123,70,-176,1
set rssi_channel = 6
set rc_smoothing_auto_smoothness = 7
set rx_min_usec = 910
set dshot_bidir = ON
set motor_poles = 12
set bat_capacity = 700
set vbat_scale = 106
set osd_rssi_alarm = 30
set osd_cap_alarm = 700
set osd_tim2 = 1281
set osd_vbat_pos = 420
set osd_rssi_pos = 2104
set osd_tim_2_pos = 2133
set osd_flymode_pos = 2082
set osd_current_pos = 2498
set osd_mah_drawn_pos = 2468
set osd_craft_name_pos = 2060
set osd_display_name_pos = 11
set osd_avg_cell_voltage_pos = 2435
set osd_esc_tmp_pos = 2518
set osd_esc_rpm_pos = 216
set osd_core_temp_pos = 2486
set vtx_band = 4
set vtx_channel = 6
set vtx_power = 1
set vtx_low_power_disarm = UNTIL_FIRST_ARM
set vtx_freq = 5840
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = 3in

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 7
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set ff_spike_limit = 70
set ff_smooth_factor = 20

rateprofile 0

# rateprofile 0
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
set tpa_rate = 80
set tpa_breakpoint = 1800

# end the command batch
batch end

# 
```
Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

Resource Alloc:
```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: FREE
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: MOTOR 3
A09: MOTOR 4
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FLASH_CS
B00: PINIO 1
B01: FREE
B02: BEEPER
B03: FREE
B04: FREE
B05: SPI_MOSI 3
B06: FREE
B07: SERIAL_RX 1
B08: CAMERA_CONTROL
B09: FREE
B10: SERIAL_TX 3
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: ADC_BATT
C02: FREE
C03: ADC_CURR
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: MOTOR 1
C09: MOTOR 2
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SERIAL_TX 5
C13: FREE
C14: LED 2
C15: LED 1
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: CAMERA_CONTROL
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 3
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 3
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
MAMBA F722 mini stack. TBS nano pro32. Foxeer Micro Falcor 2 camera. FrSyk R-XSR. 
Wired up as recommended by FC and been flying without issues on 4.0.6.

etracer65, 2020-05-22:
You failed to enable the ESC_SENSOR feature.

dzett, 2020-05-22:
Thank you, confirmed. I somehow were under the impression that this data would be coming through the bidirectional dshot as well.

closed 2020-05-22 by dzett