betaflight/betaflight #9834: No LED strip and bi-directional DSHOT on HGLRCF411

JoeDante, 2020-05-22:
I flashed Betaflight 4.2 RC2 on a HGLRCF411 FC. I flashed JazzMaverick 16.77 24khz ESCs. 
Everything runs great right out of the box, untill I installed a buzzer/LEDs today. 
When I turn on the LEDs tap in configurations, bi-directional DSHOT will not run anymore. Motors show 100% errors, when trying to arm, Betaflight will show an "dshot bbang rpm filter" error. 
When I turn the LEDSTRIP switch off, everything runs great again. Except the LEDs, of course. 

Used hardware was a HGLRC Petrel 132 4s and an HGLRC Buzzer/LED module. 

# diff

# version
# Betaflight / STM32F411 (S411) 4.2.0 May 13 2020 / 14:27:01 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: HGLRCF411, version: b71f0372, date: 2019-12-18T07:39:59Z

# start the command batch
batch start

board_name HGLRCF411
manufacturer_id HGLR

# name: Petrel 132

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 7,8:S:CW:10
led 1 8,8:S:CW:10

# color
color 10 222,0,255

# mode_color
mode_color 6 0 2

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 1 900 1050 0 0
aux 2 2 1 1375 1600 0 0
aux 3 13 2 900 1075 0 0
aux 4 35 2 1900 2100 0 0
aux 5 39 3 900 1075 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set acc_calibration = 28,-46,251,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set rc_smoothing_derivative_type = PT1
set serialrx_provider = SBUS
set blackbox_device = NONE
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 440
set small_angle = 180
set pid_process_denom = 4
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2049
set osd_tim_2_pos = 2455
set osd_current_pos = 22
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 2401
set osd_esc_tmp_pos = 56
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set gyro_rpm_notch_q = 700
set name = Petrel 132

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 6
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 41
set i_pitch = 81
set d_pitch = 34
set f_pitch = 86
set p_roll = 38
set i_roll = 77
set d_roll = 32
set f_roll = 81
set p_yaw = 41
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 21
set d_min_pitch = 23
set ff_spike_limit = 70
set ff_smooth_factor = 20

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67

# end the command batch
batch end

#

mikeller, 2020-05-23:
@JoeDante: Can you please supply the information that you were asked to supply when opening this issue:

> Use `resource show all` to create a resource allocation list and post it here in a code block. Put > ``` (three backticks) at the start and end of the output block.

Also, the configuration you are posting does not have the LED_STRIP feature enabled - the LED_STRIP will not work in this configuration.

JoeDante, 2020-05-23:
Yes, I had turned off the LEDSTRIP, because motors won't work when enabled. I turned in on again, motors now show 100% error rate. 

# diff

# version
# Betaflight / STM32F411 (S411) 4.2.0 May 13 2020 / 14:27:01 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: HGLR, board_name: HGLRCF411, version: b71f0372, date: 2019-12-18T07:39:59Z

# start the command batch
batch start

board_name HGLRCF411
manufacturer_id HGLR

# name: Petrel 132

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature TELEMETRY
feature LED_STRIP

# beeper
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN

# beacon
beacon RX_LOST
beacon RX_SET

# serial
serial 0 8192 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 7,8:S:CW:10
led 1 8,8:S:CW:10

# color
color 10 222,0,255

# mode_color
mode_color 6 0 2

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 1 900 1050 0 0
aux 2 2 1 1375 1600 0 0
aux 3 13 2 900 1075 0 0
aux 4 35 2 1900 2100 0 0
aux 5 39 3 900 1075 0 0

# vtxtable
vtxtable bands 6
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665    0 5885 5905    0    0
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable band 6 IMD6     I CUSTOM  5732 5765 5828 5840 5866 5740    0    0
vtxtable powerlevels 4
vtxtable powervalues 25 100 200 400
vtxtable powerlabels 25 100 200 400

# master
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set acc_calibration = 28,-46,251,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set rc_smoothing_derivative_type = PT1
set serialrx_provider = SBUS
set blackbox_device = NONE
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set motor_poles = 12
set vbat_max_cell_voltage = 440
set small_angle = 180
set pid_process_denom = 4
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2049
set osd_tim_2_pos = 2455
set osd_current_pos = 22
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 2401
set osd_esc_tmp_pos = 56
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 4
set vtx_channel = 1
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5740
set vcd_video_system = NTSC
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_harmonics = 1
set gyro_rpm_notch_q = 700
set name = Petrel 132

profile 0

# profile 0
set dyn_lpf_dterm_curve_expo = 6
set vbat_sag_compensation = 100
set iterm_relax_cutoff = 20
set yaw_lowpass_hz = 100
set throttle_boost = 7
set throttle_boost_cutoff = 25
set p_pitch = 41
set i_pitch = 81
set d_pitch = 34
set f_pitch = 86
set p_roll = 38
set i_roll = 77
set d_roll = 32
set f_roll = 81
set p_yaw = 41
set i_yaw = 81
set f_yaw = 81
set d_min_roll = 21
set d_min_pitch = 23
set ff_spike_limit = 70
set ff_smooth_factor = 20

rateprofile 0

# rateprofile 0
set rates_type = ACTUAL
set roll_rc_rate = 20
set pitch_rc_rate = 20
set yaw_rc_rate = 20
set roll_expo = 54
set pitch_expo = 54
set yaw_expo = 54
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67

# end the command batch
batch end

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: FREE
A03: SERIAL_RX 2
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: LED_STRIP
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: PREINIT 2
B00: ADC_BATT
B01: ADC_CURR
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: CAMERA_CONTROL
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: LED_STRIP
TIM2: FREE
TIM3: FREE
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: LED_STRIP
DMA2 Stream 7: FREE
```

mikeller, 2020-05-24:
@JoeDante:

> TIM1:
    CH1: LED_STRIP

Looks like there is a resource conflict between LED_STRIP and bitbanged Dshot on this board - can you please try `set dshot_bitbang = off` and report back? If it does not work please include another `resource show all` with this change.

JoeDante, 2020-05-25:
You are right. With dshot_bitbang off, LED_STRIP and biderectional DHSOT works!

mikeller, 2020-05-25:
@JoeDante: Yeah, if this works for you and does not have any other conflicts (i.e. stuff not working) then this is all good.

closed 2020-05-27 by mikeller