betaflight/betaflight #9835: Current offset (both ADC and virtual): Cannot set negative values in BF OSD

RipperDrone, 2020-05-22:
Latest BFC 10.7, BF 4.2RC1, FC=GHF411AIO (MATEKF411) --> You can set negative current offset values in BFC no problem. However, trying to do the same thing on BF OSD seems wrongly restricted to positive values only: When you roll down (stick left) below zero, it reads 655xx instead of negative, and when saving these values, current meter on OSD shows some 230+ Amps. --> Impossible to set negative current offset values over OSD... RC is Taranis X9D+ (2013).

Looks like unsigned integer vs. positive/negative values needed bug...

Toothpicks - BetaFPV Knight 4 Betaflight Diff.txt

resource show all:
```
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: GYRO_EXTI
A02: SERIAL_TX 2
A03: FREE
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: CAMERA_CONTROL
A09: SERIAL_TX 1
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: ADC_BATT
B01: FREE
B02: BEEPER
B03: FREE
B04: MOTOR 1
B05: MOTOR 2
B06: MOTOR 3
B07: MOTOR 4
B08: I2C_SCL 1
B09: I2C_SDA 1
B10: FREE
B11: FREE
B12: OSD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: FREE
C13: LED 1
C14: LED 2
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE

Currently active Timers:
-----------------------
TIM1:
    CH1: CAMERA_CONTROL
TIM2: FREE
TIM3:
    CH1: MOTOR 1
    CH2: MOTOR 2
TIM4:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: MOTOR 3
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: MOTOR 4
DMA1 Stream 4: MOTOR 1
DMA1 Stream 5: MOTOR 2
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

RipperDrone, 2020-05-22:
I have checked variable ranges in cms_menu_power.c and they seem consistent. Bugged somewhere else...
[image]

Asizon, 2020-05-23:
Ok, i have playing a bit with this, seems that the negative value after 0 is reenplaced by power battery capacity max `65535`. In the other hand, the values and configuration for this feature in cms are wrong, but not related to this issue.

PR #9842 (2020-05-27): Fixed display / editing of int16 in CMS.
Fixes #9835.

Test firmware:

[betaflight_4.2.0_STM32F745_4b52c819e9.zip](https://github.com/betaflight/betaflight/files/4673682/betaflight_4.2.0_STM32F745_4b52c819e9.zip)
[betaflight_4.2.0_STM32F7X2_4b52c819e9.zip](https://github.com/betaflight/betaflight/files/4673683/betaflight_4.2.0_STM32F7X2_4b52c819e9.zip)
[betaflight_4.2.0_STM32F411_4b52c819e9.zip](https://github.com/betaflight/betaflight/files/4673684/betaflight_4.2.0_STM32F411_4b52c819e9.zip)
[betaflight_4.2.0_STM32F405_4b52c819e9.zip](https://github.com/betaflight/betaflight/files/4673685/beta [... cut]
closed 2020-05-27 by mikeller