betaflight/betaflight #9837: Uncontrolled max throttle with bf 4.1.6

Dnarm, 2020-05-22:
Hi friends,

I am using armattan marmotte w CL Racing F4S. I flashed bf 4.1 (same was with bf 4.0) and used the stock pids and filtering settings. When I increase the throttle above certain level, motors start to spin uncontrollably fast (basically, to the max) and with that making throttle zero already doesn't help. What is the issue and how can I potentially fix it? Thanks a lot!

haslinghuis, 2020-05-23:
Bench testing by arming the quad with no props will only serve to confuse you. Don't do it!

Dnarm, 2020-05-23:
> Bench testing by arming the quad with no props will only serve to confuse you. Don't do it! Could burn your motors...

Thanks! Unfortunately, that happens not only on bench. I put my props on, armed, started to increase throttle and at some time the motors just go to the max, without responding to the throttle stick...

Dnarm, 2020-05-23:
Here are a couple of videos with **props on** that I made, to give you an impression of what’s going on. In both cases I cut the throttle, but as you can see, the motors don’t respond

https://www.dropbox.com/sh/epk1k6oqe9wowcn/AACDLpGzQKvUBMyXZGNRQrFPa?dl=0

RipperDrone, 2020-05-23:
Have you calibrated gyro properly before testing? If not it might be intended behavior: After surpassing min_check limit, flight controller tries to compensate for off-level alignment and hits throttle to do so...

Dnarm, 2020-05-23:
> Have you calibrated gyro properly before testing? If not it might be intended behavior: After surpassing min_check limit, flight controller tries to compensate for off-level alignment and hits throttle to do so...
> 
> Plus: Looks in test like you mounted props upside down to avoid a lift-off. This will force the quad to generate down forces on throttle, thereby not being able to level the quad or follow stick inputs (setpoint changes). Go test outside with props mounted correctly. Disarm if it still does fly to the moon... :-)

Hey RipperDrone,
I haven’t calibrated the gyro myself. Isn’t it calibrated automatically each time I connect the battery? Tried to do a quick search on how one does it - have you encountered any instructions?

Props are in fact connected the way I usually fly (not upside down). The abs bench is pressing on the quad to prevent a lift off :-))

RipperDrone, 2020-05-23:
@Dnarm calibration is NOT done pre-flight automatically, values stored once are maintained, that's it. Pls calibrate in Betaflight Configurator, make sure the quad is level when doing so. Plenty of desciptions available on the web.

haslinghuis, 2020-05-23:
You can use `gyro_cal_on_first_arm = ON` to force automatic calibration before you arm the quad the first time.

Dnarm, 2020-05-23:
> You can use `gyro_cal_on_first_arm = ON` to force automatic calibration before you arm the quad the first time.

Thanks! Tried it, but it doesn't help. Still motors accelerate to max.
One thing I didn't mention - it didn't occur with BF 3.5.7.

RipperDrone, 2020-05-23:
pls try 4.2.0RC2, configure from defaults, don't load any diff from previous versions and double check. Shouldn't occur then... 4.2 is super safe with regards to conservative filters and rock solid parameter settings as default.

marekwiecek, 2020-05-23:
Can you try disabling AIRMODE (if on) and see any if there is any
difference?

Dnarm, 2020-05-23:
> Can you try disabling AIRMODE (if on) and see any if there is any difference?
> […](#)

Hey! Yes, tried it. Without airmode (say in angle) it is working fine. On zero throttle it cuts the motors (I have motor stop option turned on).

Will try 4.2 now

Dnarm, 2020-05-23:
Just updated to 4.2 now. Also calibrated the gyro with the gyro_cal_on_first_arm function in CLI.
Unfortunately in Airmode it still shoots the motors high.

Check again - both in Angle and Horizon it works ok - throttle remains under control and I can cut decrease/cut it with the stick...

marekwiecek, 2020-05-23:
@Dnarm I would say this is normal behavior. You typically want airmode on
when in the air. But not on the bench. I would suggest you configure a mode
on a switch where you control airmode being on or off.

Dnarm, 2020-05-23:
> @Dnarm I would say this is normal behavior. You typically want airmode on when in the air. But not on the bench. I would suggest you configure a mode on a switch where you control airmode being on or off.
> […](#)

That’s how I have it - air mode on a switch. The thing is when I switch to air mode and pump the throttle a bit - it shoots to the moon. I can’t bring it down with the sticks and have to disarm. Tried it once outside, scary shit :)

mikeller, 2020-05-24:
@Dnarm: Can you please supply the information that you were asked to supply when opening this issue? Diagnosing this based on guesswork is not likely to result in success.
Also, what fligth mode are you trying to use? If it is _not_ acro mode, then this behaviour is normal, and probably exacerbated by a bad accelerometer calibration.

@RipperDrone:

> @Dnarm calibration is NOT done pre-flight automatically, values stored once are maintained, that's it. Pls calibrate in Betaflight Configurator, make sure the quad is level when doing so. Plenty of desciptions available on the web.

You have to make a distinction between _accelerometer_ calibration and _gyro_ calibration here - accelerometer calibration is done as a one - off in configurator (or in CMS or with stick commands) and then stored. Gyro calibration is done on every power up (and before first arming if `gyro_cal_on_first_arm` is enabled).

Dnarm, 2020-05-24:
> @Dnarm: Can you please supply the information that you were asked to supply when opening this issue? Diagnosing this based on guesswork is not likely to result in success.
> Also, what fligth mode are [...]

Hi mikeller!
I am very happy to add additional information, but please help me with what exactly should I supply (sorry, I didn't see anything that I was asked of during issue opening. I am new here, so please help to understand).

Just as a quick recap:
- The issue is drone flies away as motors go max RPM as I increase throttle. After that throttle stick doesn't work anymore and one has to disarm
- I am using **Airmode** flight mode (although at angle and horizon modes all is good)
- The issue is both **with and without props**
- I tried both **BF 4.1 and 4.2** - both don't work, but BF 3.5.7 works fine
- I am using **stock PIDs and filters**, no changes

mikeller, 2020-05-24:
@Dnarm:

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

Use `resource show all` to create a resource allocation list and post it here in a code block. Put ``` (three backticks) at the start and end of the output block.

Dnarm, 2020-05-24:
Appreciate your help a lot, @mikeller !
Please, find below.
Diff all:
```
# 

# diff all

# version
# Betaflight / STM32F405 (S405) 4.2.0 May 13 2020 / 14:26:41 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF4, version: b3487114, date: 2019-10-19T00:26:29Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CLRACINGF4
manufacturer_id CLRA
mcu_id 00560047324d501720303256
signature 

# name: Marmotte

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature MOTOR_STOP

# beeper
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# serial
serial 3 64 115200 57600 0 115200

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 900 1200 0 0
aux 2 2 1 1325 1675 0 0
aux 3 13 2 1700 2100 0 0
aux 4 19 4 1700 2100 0 0
aux 5 35 3 1700 2100 0 0

# master
set acc_calibration = -19,-23,-81,1
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 10
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT300
set small_angle = 180
set pid_process_denom = 2
set debug_mode = GYRO_RAW
set name = Marmotte

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
#
```

Resource show all:
```
# 

# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: MOTOR 4
A03: MOTOR 3
A04: GYRO_CS 1
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: PINIO 1
A15: OSD_CS
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: PREINIT 3
B04: BEEPER
B05: LED 1
B06: FREE
B07: SDCARD_DETECT
B08: FREE
B09: CAMERA_CONTROL
B10: FREE
B11: FREE
B12: SDCARD_CS
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: INVERTER 1
C01: ADC_CURR
C02: ADC_BATT
C03: FREE
C04: GYRO_EXTI
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: BEEPER
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH1: DSHOT_BITBANG 2
    CH2: DSHOT_BITBANG 1
TIM9: FREE
TIM10: FREE
TIM11:
    CH1: CAMERA_CONTROL
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: DSHOT_BITBANG 2
DMA2 Stream 3: DSHOT_BITBANG 1
DMA2 Stream 4: ADC 1
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

#
```

I also did a couple of Blackbox logs: https://www.dropbox.com/sh/epk1k6oqe9wowcn/AACDLpGzQKvUBMyXZGNRQrFPa?dl=0
2 in Angle mode (when throttle can control the motors) and 2 in Airmode (where the problem is and motors are running out of control). I did it in GYRO_RAW debug mode, but always happy to repeat in other mode if helpful.

My observations are that motors 1 and 4 are running to the max, not exactly all motors...

Looking forward to your help, folks. Thanks a lot in advance!

ctzsnooze, 2020-05-25:
In the logs, you are holding the quad down with something.
It cannot fly while held down.
You have a small offset on yaw of 4 deg/s, so the FC wants to make the quad turn at 4deg/s.  But because you are holding the quad down, the yaw I term builds up in an attempt to turn it as commanded, which leads to the two yaw drive motors slowly going faster and faster.
This is entirely normal behaviour.
Quadcopter always do this when held down.
You need to go to a grassy field and actually fly it.
Have you ever flown a Betaflight quadcopter like this before, or is this your first one?
[image]

Dnarm, 2020-05-25:
> In the logs, you are holding the quad down with something.
> It cannot fly while held down.
> You have a small offset on yaw of 4 deg/s, so the FC wants to make the quad turn at 4deg/s. But because you [...]

Hi @ctzsnooze ! Thanks a million for this explanation. It makes sense completely. Let me go out and try it.
Why is this offset happening? Is it normal?
Yeah, I have some experience with betaflight drones, I flew this particular machine a lot on BF 3.5.7.

Dnarm, 2020-05-26:
Guys, thanks a lot again for the help.
I tested today quickly the quad in an open field, indeed, it doesn't fly away! Super happy about that.

Now the big questions is what to do with the small yaw rotation?

ctzsnooze, 2020-05-27:
Closed. No problem with software. Quad held down by user.

closed 2020-05-27 by ctzsnooze