betaflight/betaflight #9840: Intermittent RPMFILTER arming flag on 4.2 RC2

granolaFPV, 2020-05-23:
**Describe the bug**
I'm getting an intermittent RPMFILTER arming flag when attempting to arm. After unplugging and re-plugging the battery pack a few times, the quad arms as normal. This doesn't occur on every arming attempt, only about 50% of the time. This is on a new build, and I haven't tried any other Betaflight versions yet. I don't see any obvious hardware issues, but I can't confirm it's NOT a hardware issue. There are at least two other people reporting similar issues on the Betaflight Facebook group and Bardwell's Discord. All are running RC2.

**To Reproduce**
Steps to reproduce the behavior: Plug in. Arm quad. Observe "RPMFILTER" warning. Disarm. Try again. RPMFILTER flag persists. Unplug battery. Try arming again. Eventually the quad arms without the RPMFILTER error.

**Expected behavior**
Quad arms without throwing the RPMFILTER error.

**Flight controller configuration**
```# diff all

# version
# Betaflight / STM32F745 (S745) 4.2.0 May 13 2020 / 14:27:23 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: HBRO, board_name: KAKUTEF7, version: cc8e118f, date: 2020-03-04T02:37:20Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id HBRO
mcu_id 001a003a3538510a31373730
signature 

# name: GRANOLA

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature LED_STRIP

# beeper
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 3 32 115200 57600 0 115200
serial 5 64 115200 57600 0 115200
serial 6 1024 115200 57600 0 115200

# led
led 0 5,6::C:12
led 1 6,6::C:12
led 2 7,6::C:12
led 3 8,6::C:12
led 4 9,6::C:12
led 5 10,6::C:12

# aux
aux 0 0 0 1800 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 2 1925 1975 0 0
aux 4 28 1 900 1700 0 0
aux 5 30 1 1400 2100 0 0
aux 6 35 2 1825 1925 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 3
vtxtable powerlabels 25 200 600 1K

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 500
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = -14,-24,87,1
set baro_hardware = NONE
set min_check = 1020
set max_check = 2000
set rssi_channel = 16
set rc_smoothing_auto_smoothness = 20
set fpv_mix_degrees = 13
set serialrx_provider = SBUS
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set motor_poles = 16
set bat_capacity = 1300
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set vbat_scale = 111
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_vbat_pos = 407
set osd_rssi_pos = 2081
set osd_tim_1_pos = 55
set osd_tim_2_pos = 2103
set osd_flymode_pos = 2084
set osd_throttle_pos = 321
set osd_vtx_channel_pos = 2433
set osd_craft_name_pos = 2443
set osd_warnings_pos = 2377
set osd_avg_cell_voltage_pos = 2455
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 3
set vtx_low_power_disarm = ON
set vtx_freq = 5880
set gyro_1_align_yaw = 2700
set gyro_rpm_notch_q = 800
set name = GRANOLA

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 126
set dyn_lpf_dterm_max_hz = 306
set dyn_lpf_dterm_curve_expo = 10
set dterm_lowpass2_hz = 270
set vbat_sag_compensation = 100
set anti_gravity_gain = 5000
set feedforward_transition = 100
set iterm_relax_cutoff = 10
set iterm_windup = 75
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 70
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set idle_min_rpm = 20

profile 1

# profile 1
set dyn_lpf_dterm_min_hz = 126
set dyn_lpf_dterm_max_hz = 306
set dyn_lpf_dterm_curve_expo = 10
set dterm_lowpass2_hz = 270
set vbat_sag_compensation = 100
set anti_gravity_gain = 5000
set feedforward_transition = 100
set iterm_relax_cutoff = 10
set iterm_windup = 75
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 70
set p_pitch = 51
set i_pitch = 70
set d_pitch = 53
set f_pitch = 133
set p_roll = 46
set i_roll = 65
set d_roll = 49
set f_roll = 126
set p_yaw = 70
set i_yaw = 75
set f_yaw = 126
set d_min_roll = 39
set d_min_pitch = 43
set d_min_boost_gain = 55
set d_min_advance = 100
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set idle_min_rpm = 20

profile 2

# profile 2
set dyn_lpf_dterm_min_hz = 126
set dyn_lpf_dterm_max_hz = 306
set dyn_lpf_dterm_curve_expo = 10
set dterm_lowpass2_hz = 270
set vbat_sag_compensation = 100
set anti_gravity_gain = 5000
set feedforward_transition = 100
set iterm_relax_cutoff = 10
set iterm_windup = 75
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set yaw_lowpass_hz = 70
set p_pitch = 55
set i_pitch = 70
set d_pitch = 53
set f_pitch = 115
set p_roll = 50
set i_roll = 65
set d_roll = 49
set f_roll = 110
set p_yaw = 70
set i_yaw = 75
set f_yaw = 100
set d_min_roll = 39
set d_min_pitch = 43
set d_min_boost_gain = 55
set d_min_advance = 100
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set idle_min_rpm = 20

# restore original profile selection
profile 2

rateprofile 0

# rateprofile 0
set thr_mid = 43
set thr_expo = 15
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

rateprofile 1

# rateprofile 1
set thr_mid = 43
set thr_expo = 15
set yaw_rc_rate = 210
set yaw_expo = 15
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 10

rateprofile 2

# rateprofile 2
set thr_mid = 43
set thr_expo = 15
set roll_rc_rate = 1
set pitch_rc_rate = 1
set yaw_rc_rate = 210
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 10

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save```

**Setup / Versions**
 - Holybro Kakute F7 V1.5 FC
 - Tekko32 F3 40A 4in1 ESC
 - FC and ESC are connected with the supplied cable (w. plugs on both ends)
 - Eaglepower 2306 2345Kv motors (these are old and crappy)

**Additional context**
Quad flies great once it arms successfully. In the Motors tab in the configurator, errors show 0% for all ESCs — _except_ one instance where it was showing ESC #3 at 100% error. Unplugging and re-plugging the battery cleared the issue — all ESCs again showed 0% error.

This is my first time submitting an issue for Betaflight, so please let me know if additional information is needed.

Thanks Betaflight devs! 4.2 is awesome.

Bwsbmw, 2020-05-23:
I am also experiencing intermittent RPM filter arming flag causing a no arm condition. The way I have been resolving this issue for now is unplugging and replugging. I've double checked my soldering, no issues found. I have verified proper operation of the rpm filters via betaflight. Confirming there is 0% errors from idle to 1500us. Didnt have any issues with any of the dev versions, didn't have any issues after the flag was introduced. Has only happened on 4.2rc2. Did have to re plug twice. Am going flying later, will add results.

Ferrett1231, 2020-05-23:
I have the same stack as GranolaFPV - and experience the same RPMFiltering issue. Power cycle my quad and it goes away. BF4.2 RC2

mikeller, 2020-05-24:
@granolaFPV, @Bwsbmw, @Ferrett1231: The 'RPMFILTER' arming disabled reason is used when RPM filtering is active, but there is no valid RPM feedback received from one or more of the ESCs.
Can you please update all of your ESCs to the latest firmware version that supports RPM filtering (16.71 in the case of BLHeli_S) and try again? If it does not work please report the ESC firmware version that you are using, and the output of `dshot_telemetry_info` in CLI taken in a case when 'RPMFILTER' is showing?

Ferrett1231, 2020-05-24:
@mikeller: I already have the latest version of BLHeli32 - v.32.7

granolaFPV, 2020-05-24:
@mikeller — I'm on BLHeli32 32.7 as well.

Occasionally ESC #3 doesn't read in BLHeliSuite32xm, which mirrors what I see in the Betaflight Config Motors tab — intermittent issues with ESC/Motor #3.

I've taken photos of all of the hardware components and dropped them here: https://www.dropbox.com/sh/atpcmfeqjbwa2y7/AAB-VRqB060fAQIjxuWtMrT1a?dl=0

Bwsbmw, 2020-05-25:
I'm also on the most current Blheli32, 32.7 I have only got to fly 3 batteries since experiencing the flag, and have not had it again. Still the occasional unsynced startup tones (last 2 beeps). I will be flying it again tomorrow though. Not sure what to be look for when i am having the intermittent rpmfilter warning arming flag. I guess I will comment if this is still open.

mikeller, 2020-05-25:
@granolaFPV, @Bwsbmw, @DronerFPV: I sure can't see a commonality in potential cause between you guys' problem - and 1000s of users are using 4.2.0-RC2 with RPM feedback without these problems, so not sure this is a problem in Betaflight.

granolaFPV, 2020-05-25:
@mikeller Thanks for looking into this. I’m going to swap out motors and continue troubleshooting later this week.

Bwsbmw, 2020-05-25:
Yes, thanks for looking into it. I have run 5 batteries so far, I had one sketchy sync where motor 3 was a good 2 seconds behind the others. So I power cycled, again dont believe thats BF related . Cannot duplicate condition at this time. Thanks again, be safe.

granolaFPV, 2020-05-29:
Flew a few test packs this evening (with fresh motors) and got the RPMFILTER arming flag on one of four packs. Unplugging and re-plugging the battery resolved the issue, and it didn't appear again for the rest of the session. Still running 4.2 RC2. Everything looks normal in Betaflight and BLHeli when on the bench. 

I didn't grab the `dshot_telemetry_info` output this time, but I'll get it next time this comes up.

dentex, 2020-05-30:
I had the same problem with this arm prevention warning. I also have one "lazy" ESC from the 4in1 (it's an Airbot Furling32, brand new; also new motors and build in general). It happened yesterday with the RC2 and today with the RC3.

vicewize, 2020-05-30:
Im also experiencing the same issue so much so as I thought it was an ESC issue (un-synced start up tones and having to re-cycle power until it works). Kakute F7 HDV. I bought a new ESC tested doing same thing. Power cycling it clears it up until next battery swap, it doesn't always occur but for me quite often.

Bwsbmw, 2020-05-31:
rpmfilter flag happened 2 times again within the last 8 batteries on RC2. Question: How am I supposed to get the dshot_telemetry_info?

DronerFPV, 2020-05-31:
I have been at least 10 lipos now on BF4.2_RC3 without seeing the 'RPMFILTER'
Holybro Kakute F7 V1.5 FC
Tekko32 F3 40A 4in1 ESC

vicewize, 2020-05-31:
> rpmfilter flag happened 2 times again within the last 8 batteries on RC2. Question: How am I supposed to get the dshot_telemetry_info?

Go to motor tab run your motors at like idle or so say 1050 (or 5% digital idle for example) leave them running then go to CLI type dshot_telemetry_info and if should show you all related d-shot eRPM, frequency, error rate etc on all 4 motors. Problem is when it works it works great (for me at least) and always shows perfect telemetry data, I haven’t tried it when it happens but I know it would show one or 2 motors with a high error rate, or sometimes missing 1 motor (I have ran into that in the motors tab when it occurred but never tried it in CLI but worth a shot, for me it’s shows perfect days when I tried this in CLI with it not occuring of course). I’ve tried reflashing and going back to 4.1.x and it never occurs, it’s very odd. It still fly the new firmware because it’s very good and fly’s great and I can live with power cycling if it occurs for now but we’re definitely not imagining this issue.

Bwsbmw, 2020-05-31:
Yeah I just ran it in the cli so I could also get my rpm for dynamic idle.
I got nothing. Everything looks fine, no errors. 0% at 5.5% I would have to
sit here and try it over and over on thr bench?

DronerFPV, 2020-05-31:
Another 11 lipos today - no RPMFILTER arming flag on BF4.2_RC3

Bwsbmw, 2020-05-31:
I have not been able to fly due to weather, and having to take care of
things around the house. I'm now on RC3 also, I have yet to fly though.
Within the next couple of days I will have some input. Sorry guys.

vicewize, 2020-06-01:
> Yeah I just ran it in the cli so I could also get my rpm for dynamic idle. I got nothing. Everything looks fine, no errors. 0% at 5.5% I would have to sit here and try it over and over on the bench? Bardwell has seen it happen on a quad with rpm filtering turned off...
> […](#)

You don’t need to go to the CLI to get your eRPMs for dynamic idle you just spin them up in the motors tab and it will show the eRPMs just make sure you have it set to your idle value  i.e. 1050 for 5% idle etc. (I don’t know what your idle value is). As far as seeing the data in the CLI the motors have to be spinning before getting into the CLI before it will show eRPMs and such. So go to motors tab leave them running then go to CLI and type command.

vicewize, 2020-06-01:
> Yeah I just ran it in the cli so I could also get my rpm for dynamic idle. I got nothing. Everything looks fine, no errors. 0% at 5.5% I would have to sit here and try it over and over on the bench? Bardwell has seen it happen on a quad with rpm filtering turned off...
> […](#)

Interesting so Bardwell came across this issue as well you stated? On YouTube or somewhere else (forum etc)?

dentex, 2020-06-01:
Yesterday 7 LiPOs, no warnings. But often one of the ESC came up later than the others. So now I don't know if the things are related or not. I'll keep you posted about what I'll found in the upcoming days.

Bwsbmw, 2020-06-01:
Yes, I understand that. I did use the motors tab to get my dynamic idle
rpm. I proceeded into the cli with the motors running, because I figured
that's what I want to look at, dshot_telemtery_info and was not seeing
anything. Rpm is working properly on the bench while pluged into bf via
USB, but not in the field.

vicewize, 2020-06-01:
> Yes, I understand that. I did use the motors tab to get my dynamic idle rpm. I proceeded into the cli with the motors running, because I figured that's what I want to look at, dshot_telemtery_info [...]

Gotcha misunderstood you. Most of the time I noticed I don’t always get the RPM flag but get the un-synced startup tones which happen quite often. More than just having the flag and the un-synced tones together.

capposteve, 2020-06-06:
i personaly think this is a build issue, and im on the understanding its noise near the wires that joint the fc to the esc, really if they where shielded then i think no one would have these issues.
steve

DronerFPV, 2020-06-06:
I have been through dozens of lipos in RC3 and no more RPMFiLTER issues
since RC2...

capposteve, 2020-06-06:
i mean build issue as in actual build, not firmware build. some quads and fcs and esc are just noisier electricly. this is my understanding. i currently have a mamba f711 mini refusing to work repeatedly on 4.17 rc1

Bwsbmw, 2020-06-06:
Update: I have also flown quite a bit of packs also threw the quad that was
getting RPM WARNING on 4.2RC2. Has not happened on RC3, or RC4.

haslinghuis, 2020-06-06:
Already suggested to @capposteve to upgrade before further issuing because betaflight == betaflight :trollface:

capposteve, 2020-06-06:
> Already suggested to @capposteve to upgrade before issuing because betaflight == betaflight :trollface: it is an ongoing project improving in time. It will never mature because it's bleeding edge. Although it is pretty stable before I offend anyone.

Well bf 4.17 isn’t exactly old. Are you suggesting I set it up with bf 4.2 rc4 and see what happens? 

Thanks.

haslinghuis, 2020-06-06:
There is no point in testing a 4.1.7 `release candidate` version of a previous betaflight version if there is a 4.1.7 `final`.

But @Bwsbmw states the problem does not occur with `latest` so I would test that claim 😉

capposteve, 2020-06-06:
Where is the final? I haven’t seen that. All I’ve seen is the 4.17 rc1 and rc2. I must be missing something

haslinghuis, 2020-06-06:
[final](https://github.com/betaflight/betaflight/releases/tag/4.1.7)
[latest](https://github.com/betaflight/betaflight/releases/tag/4.2.0-RC4)

Bwsbmw, 2020-06-06:
Both I, and @DronerFPV  have not seen the rpmwarning issue since 4.2rc2, causing a no arm condition.

capposteve, 2020-06-06:
> [final](https://github.com/betaflight/betaflight/releases/tag/4.1.7)
> [latest](https://github.com/betaflight/betaflight/releases/tag/4.2.0-RC4)

sorry my bad. yes im running the final build of 4.17, its the configurator that im running 10.7.0 RC1  in. someone has already reported the same issue with BF 4.2 rc4 so dont really want to update. 
what was the fix on other posts as it was fixed aparently?

Bwsbmw, 2020-06-06:
There hasnt been a fix. We are still trying to figure out what is causing
the intermittent rpm warning flag.

capposteve, 2020-06-07:
Ok I fixed it for me anyway. My issue was (or what seemed to fix it for me anyway) was in blheli32 was the start up tone. I had it set to do a music tone using each motor to do slightly diff tones. Turning this off and putting it back to default fixed my issue and not had one issue since.

mikeller, 2020-06-07:
@capposteve: Good point - other users have reported as well that custom startup tones can interfere with bidirectional RPM being properly recognised by the ESC.

capposteve, 2020-06-07:
Maybe there is a timing issue? If say each motor doesn’t finish it’s start up tone at the same time it throws an error? As when plugging into blheli32 it was coming up saying esc3and 4 not synced. As soon as I removed the custom start up tones the problem went away. Just letting you know my findings for my build anyway. Steve

dentex, 2020-06-07:
I've never used startup tones and I've been having this issue since day one (I started using RPM filters and bl_heli32 from BF 4.2 RC2). I noticed though that on RC 3 and now on 4, it's happening less often.

capposteve, 2020-06-07:
Have you tried this on a diff setup or is this happening on just the one quad?

Bwsbmw, 2020-06-08:
> I've never used startup tones and I've been having this issue since day one (I started using RPM filters and bl_heli32 from BF 4.2 RC2). I noticed though that on RC 3 and now on 4, it's happening less often.

You've had the warning flag for RPMFILTER on 4.2rc3 and rc4? I deffenetly got it on rc2 and @granolaFPV @DronerFPV and a few others having the flag and not being able to arm.. The startup tones being out of sync doesnt seem to be related. I have 1 blheli s and one blheli32 that startup tones are out of sync but they fly fine. The issue is the intermittent rpmfilter warning arming flag causing the no arm condition.

capposteve, 2020-06-08:
This is what I was getting when it was playing up. 
[image]
[image]

capposteve, 2020-06-08:
Also this. 
[image]

DronerFPV, 2020-06-08:
I have not seen the RPMFILTER flag on RC3 or RC4.

granolaFPV, 2020-07-11:
I got one or two instances of this on the final 4.2 release, but have since moved my VTX wiring away from my 4-in-1 ESC. Since making this change I haven't seen the issue reappear. This supports @mikeller's original comments that this is likely hardware-related (at least in my case).

Bwsbmw, 2020-07-11:
I also have not had the issue occur since going to 4.2rc3 only had the
issue with rc2. since 4.2 rc3,4 and full have been fine. thanks

granolaFPV, 2020-08-02:
I had another instance of this last week with 4.2.0. Is it possibly related to this bidirectional DSHOT issue with F3 ESCs?

https://github.com/bitdump/BLHeli/issues/468

Bwsbmw, 2020-08-02:
> I had another instance of this last week with 4.2.0. Is it possibly related to this bidirectional DSHOT issue with F3 ESCs?
> 
> [bitdump/BLHeli#468](https://github.com/bitdump/BLHeli/issues/468)

The quad I was having the same RPM Filtering warning was a non F3, so I dont think its specifically related to F3 esc's as it was a non F3. I do have another quad running an F3 4in1, no issues so far besisdes the occasional unsynced intialization. Which doesnt seem to effect the way it flys, I'm also running test version 32.7.2 on that. Needless to say, I havent had it come up at all since rc2 on the quad showing the flag, and thats still on 32.7 I honestly don't know a fix besides power cycle. You also could try blheli_32 32.7.2. You have to download the hex as its not in blheli32suite.

droneperson42, 2020-08-14:
I also have this issue running release 4.2.1 although two other setups with the same ESC & FC (T-motor F45A & Lumeneir Lux F7) work perfectly using the same 'diff' settings. ~50% of the time the error appears and the quad will not arm.

**As posted in another thread, plugging the FC into 5V with a USB cord then plugging in the battery and removing the USB cord consistently fixes the issue.** 

It's like the FC takes too long to boot up and initialise the motor input pins on some occasions and misses the initial telemetry communications to/from the ESC when it boots up.

SJChannel, 2020-09-01:
I have this issue under Betaflight 4.2.2.  My FC is the Holybro Kakute F7 AIO and the ESCs (not 4-in-1) are Holybro Tekko32 F3 35 amp.  I'm currently using DSHOT600, but will try DSHOT300 and see if it helps.

Bwsbmw, 2020-09-02:
> I have this issue under Betaflight 4.2.2. My FC is the Holybro Kakute F7 AIO and the ESCs (not 4-in-1) are Holybro Tekko32 F3 35 amp. I'm currently using DSHOT600, but will try DSHOT300 and see if it helps.

The quad I was having the same RPM Filtering warning was a non F3, tekko32 35amp individuals.  I dont think its specifically related to F3 esc's, as it was a non F3. I do have another quad running an F3 4in1, no issues so far besisdes the occasional unsynced intialization. Which doesnt seem to effect the way it flys. I'm also running test version 32.7.2 on that. Needless to say, I havent had it come up at all since rc2 on the quad showing the flag, and thats still on 32.7 I honestly don't know a fix besides power cycle. You also could try blheli_32 32.7.2. You have to download the hex as its not in blheli32suite.

Bwsbmw, 2020-09-02:
> I had another instance of this last week with 4.2.0. Is it possibly related to this bidirectional DSHOT issue with F3 ESCs?
> 
> [bitdump/BLHeli#468](https://github.com/bitdump/BLHeli/issues/468)

I'm not to sure, but I dont think so. I have had the warning pop up on non F3 esc's, and ive heard of one occurance on a blhelis esc.  Can try 32.7.2, I havent been having any issue on that so far.

SJChannel, 2020-09-02:
@Bwsbmw I forgot to mention that I'm running BLHeli32 32.7.2 in the ESCs.  I have edited my previous comment to state that.  I'll report back after I've tried switching to DSHOT300.

SJChannel, 2020-09-02:
Here's something interesting.  I changed my quad from DSHOT600 to DSHOT300 (and also changed the loop times from 8k/8k to 8k/4k as required).  With DSHOT300 it gets the RPMFILTER warning **every** time.  I tried power-cycling it at least 15 times, and never once succeeded in arming the quad.

I changed it back to DSHOT600 and it is back to the previous behavior.  I.e., it gets the RPMFILTER warning roughly 25% of the time.

Bwsbmw, 2020-09-03:
> Here's something interesting. I changed my quad from DSHOT600 to DSHOT300 (and also changed the loop times from 8k/8k to 8k/4k as required). With DSHOT300 it gets the RPMFILTER warning **every** [...]

Man, that's beyond me. You have 0.00% error when you spin your motors up in betaflight? Props off. You can also look at your dshot_telemetry_info in the cli. I find it odd cause im running a kakutef7mini v2 bf 4.2.2, and a tekko32 F3 C 45amp 4in1, on 4s, 2306.5-2580kv smoox, 8k8k dshot600 CPU load is like 3-4%, and tasks in the cli is 7.7% on the gyro, 34% on the pid, and 68% total. So dshot600 8k8k on a f7 should be fine.  Esc's set at demag high, timing 23. I have not had any issues on this quad at all. Which is really interesting. I would try turning it off just to see if it goes away totally. If it does, there might be something going on between the esc's and the fc, which makes no sense really.. I wish I could help. Sorry.

vicewize, 2020-09-03:
Sounds like possibly an ESC related issue. (Hardware possibly settings)

-Encrypted via Nord VPN-
vicewize.com

SJChannel, 2020-09-03:
> Sounds like possibly an ESC related issue. (Hardware possibly settings)

It seems like it might be a timing issue when BF tries to detect whether bidirectional DShot is working at power-up time.  When I go into Betaflight Configurator and run the motors, the RPM values read correctly and it shows 0.00% errors on all four motors.  Then if I disconnect the USB cable (leaving the LiPo still connected), I can arm it from my radio without any problems.  But when I power it up by plugging in the LiPo without connecting the USB port, it fails to arm every time with the RPMFILTER warning.  This is all with the ESC protocol set to DSHOT300.

I tried flashing it back to BF 4.1.7, and it failed the same way.

I was hoping to try and debug it with some custom Blackbox logging, but unfortunately the SD card on the FC is dead.  So for now, I'll be running BF 4.2.2 without RPM filters.

SJChannel, 2020-09-06:
I have some new information about this problem.  As I mentioned before, I have been running the beta version 32.7.2 of BLHeli32 on my ESCs.  When using DSHOT300, I got the RPMFILTER warning every single time I tried to arm the quad.  When using DSHOT600 I got the same thing, but only about 8-10% of the time.

Today I flashed my ESCs with BLHeli32 32.7, the latest official release.  Running that firmware, everything seems to be working fine at both DSHOT300 and DSHOT600.  So my problem was apparently caused by something in the beta ESC firmware.

Again, these are Holybro Tekko32 F3 35 amp individual ESCs (BLHeli32 target Furling32) and Betaflight 4.2.2.

@sskaug Just let me know if you'd like any additional information, or if you'd like me to run some other tests.

SJChannel, 2020-10-06:
I have been trying to debug this problem, but haven't had any success.  With this HW/SW setup, it fails **every** time:

- Holybro Kakute F7 AIO
- Tekko32 F3 35 amp individual ESCs
- Betaflight 4.2.3
- BLHeli32 32.7.2
- DSHOT300
- Bidirectional DSHOT enabled
- 8k/4k PID loop

I have examined the DSHOT signal line on an oscilloscope, and the packets being sent by the flight controller look 100% valid.  At power-up, the signal line goes high (the idle state for inverted DSHOT) for about 142 usec, then the periodic DSHOT packets (0x000f = motor stop) start coming out.  The timing of the pulses is perfect for DSHOT300, but the ESC does not respond by sending back DSHOT telemetry in the reverse direction.  I tried modifying the BF code to make the initial high-level idle state last longer before the DSHOT packets start coming out; but it made no difference.

If I power the FC from USB prior to connecting the LiPo, bidirectional DSHOT works fine and the Motors tab shows 0% errors from all four motors.  But if I power the quad first from the LiPo before plugging in the USB cable, the Motors tab shows 100% errors on all motors and they do not spin when I move the sliders.

If I switch to DSHOT600, it works most of the time, but it still fails occasionally.  Ditto for DSHOT150.

When I flash the ESCs back to BLHeli32 32.7, all DSHOT speeds work fine.  But 32.7 with bidirectional DSHOT is an unusable combination for F3 ESCs because of the well-known BLHeli32 bug that causes the DSHOT replies to come back too soon on occasion.

mikeller, 2020-10-11:
@SJChannel:

> When I flash the ESCs back to BLHeli32 32.7, all DSHOT speeds work fine. But 32.7 with bidirectional DSHOT is an unusable combination for F3 ESCs because of the well-known BLHeli32 bug that causes the DSHOT replies to come back too soon on occasion.

This sounds very much like an issue with the ESC firmware - this should be closed and an issue with [BLHeli_32](https://github.com/bitdump/BLHeli/issues) opened.

[3 more replies]

closed 2021-02-08 by github-actions[bot]