betaflight/betaflight #9846: BF 4.2 - GPS Sat fix issue

peteoz, 2020-05-24:
**Describe the bug**
number of users are reporting an issue with BF4.2 where they can't get a gps sat fix, in fact they can't even pick up a single satellite (this was posted on FB LR Hooligans group), while at the same time identical setup works fine on earlier versions of BF.
for some of the affected users it helped when they recycled power, or plugged the FC back to laptop and made any random change in GPS settings on configuration tab but then same issue pops up on the next flight.

I've flashed BF 4.2 RC2 on one of my quads today and went out to test. my success rate was about 50% and fair enough when I didn't get a fix then after I recycled power then I would have a fix immediately (on one occasion I had to recycle power 2x).

this is like a world of 2 extremes : either I get a fix instantly (honestly under 2 seconds with close to 20 satellites), or the other extreme I don't a fix at all and not even a single satellite no matter how long I wait.

the good news is that usually one power recycle fixes the problem (until the next flight), then all GPS related OSD elements work as they should and tested GPS Rescue works perfectly fine.

same quad on BF 3.5.7 & BF 4.1.6 doesn't behave this way, I get a fix every time, I don't need to recycle power.

**To Reproduce**
Install BF4.2 on a quad with GPS

**Expected behavior**
should work as in BF 3.5.7, BF 4.1.6 as explained above

**Flight controller configuration**
```
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 May 13 2020 / 14:26:41 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: CLRA, board_name: CLRACINGF4, version: b3487114, date: 2019-10-19T00:26:29Z

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id CLRA

# name: Stingy 6

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B08

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature SOFTSERIAL
feature GPS
feature TELEMETRY

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER1234

# serial
serial 3 64 115200 57600 0 115200
serial 5 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# aux
aux 0 0 0 900 1700 0 0
aux 1 1 1 900 1150 0 0
aux 2 46 1 1150 1400 0 0
aux 3 13 1 1850 2100 0 0
aux 4 28 0 900 1300 0 0
aux 5 35 1 1600 1850 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A FACTORY 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B FACTORY 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E FACTORY 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F FACTORY 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R FACTORY 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues 0 1 2 4
vtxtable powerlabels 25 200 500 800

# rxfail
rxfail 3 h

# master
set gyro_lowpass2_hz = 0
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_notch_min_hz = 90
set dyn_notch_max_hz = 350
set acc_calibration = 59,16,-112,1
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_derivative_type = PT1
set rc_smoothing_auto_smoothness = 25
set max_aux_channels = 4
set serialrx_provider = CRSF
set dshot_bidir = ON
set failsafe_delay = 2
set failsafe_throttle = 1055
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 440
set ibata_scale = 221
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_set_home_point_once = ON
set gps_rescue_angle = 40
set gps_rescue_sanity_checks = RESCUE_SANITY_OFF
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_link_quality = ON
set osd_rssi_alarm = 35
set osd_rssi_dbm_alarm = -95
set osd_alt_alarm = 1500
set osd_vbat_pos = 450
set osd_rssi_pos = 441
set osd_link_quality_pos = 2480
set osd_rssi_dbm_pos = 2487
set osd_tim_1_pos = 461
set osd_tim_2_pos = 2508
set osd_flymode_pos = 333
set osd_g_force_pos = 505
set osd_throttle_pos = 2083
set osd_vtx_channel_pos = 432
set osd_ah_pos = 200
set osd_current_pos = 2113
set osd_mah_drawn_pos = 2499
set osd_craft_name_pos = 2540
set osd_gps_speed_pos = 2121
set osd_gps_lon_pos = 2064
set osd_gps_lat_pos = 2050
set osd_gps_sats_pos = 2104
set osd_home_dir_pos = 2127
set osd_home_dist_pos = 2129
set osd_flight_dist_pos = 2135
set osd_compass_bar_pos = 2089
set osd_altitude_pos = 2550
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 480
set osd_power_pos = 2531
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2466
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 424
set osd_disarmed_pos = 138
set osd_nheading_pos = 2099
set osd_nvario_pos = 2519
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_core_temp_pos = 2473
set osd_efficiency_pos = 2145
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_min_link_quality = ON
set osd_stat_flight_dist = ON
set osd_stat_max_fft = ON
set osd_stat_min_rssi_dbm = ON
set vtx_band = 5
set vtx_channel = 1
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5658
set vcd_video_system = PAL
set gyro_rpm_notch_q = 800
set name = Stingy 6

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 84
set dyn_lpf_dterm_max_hz = 204
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 180
set vbat_sag_compensation = 100
set anti_gravity_gain = 7500
set feedforward_transition = 60
set iterm_relax_cutoff = 10
set iterm_windup = 75
set throttle_boost_cutoff = 10
set p_pitch = 65
set d_pitch = 50
set f_pitch = 145
set p_roll = 60
set d_roll = 45
set f_roll = 140
set p_yaw = 60
set f_yaw = 120
set d_min_roll = 40
set d_min_pitch = 40
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 42
set ff_boost = 5

rateprofile 0

# rateprofile 0
set roll_rc_rate = 117
set pitch_rc_rate = 117
set yaw_rc_rate = 150
set roll_expo = 25
set pitch_expo = 25
set yaw_expo = 25
set roll_srate = 74
set pitch_srate = 74
set yaw_srate = 40
set tpa_breakpoint = 1250
```

**Setup / Versions**
CLRacing F4, HGLRC UBlox M8N GPS, Crossfire Rx, Unify Pro HV VTx, Spedix GS35 ESCs 

**Additional context**
nice sunny day, unobstructed open sky, the same GPS unit and quad works perfectly fine with any other BF version.

this doesn't seem to be same issue as reported here : https://github.com/betaflight/betaflight/issues/9194

because when I get a successful fix then my number of sats is high (usually close to 20) and the number of sats throughout the flight is consistent, not dropping.

mikeller, 2020-05-25:
@peteoz: Interesting observation about the intermittent issues when power cycling. Maybe the experts from room #9194 (@TonyBlit, @atomiclama, ...) can make sense of this?

TonyBlit, 2020-05-25:
4.2 has introduced the SBAS_NONE setting and is now the default. It doesn't make sense to me that this parameter could avoid sat locking, and even less in a random fashion, but you can set gps_sbas_mode=SBAS_AUTO to get what the same setting as in previous BF versions.

peteoz, 2020-05-25:
> 4.2 has introduced the SBAS_NONE setting and is now the default. It doesn't make sense to me that this parameter could avoid sat locking, and even less in a random fashion, but you can set gps_sbas_mode=SBAS_AUTO to get what the same setting as in previous BF versions.

@TonyBlit I'm using gps_sbas_mode = EGNOS (on 3.5.7, 4.1.6 and now 4.2) so in my case this setting has not changed between the different BF versions

TonyBlit, 2020-05-25:
oops, I looked for the sbas_mode param in your config and for some reason I didn't see it, sorry O:-)
Hmm.. now I'm remembering I had a similar problem during development. In order to implement sbas_none I refactored how the sbas command was sent (the old hardcoded way was very difficult to scale), and at some point I was not calculating the CRC properly so the state machine couldn't progress and the GPS didn't work at all. I fixed the CRC calculation and that was it. But maybe there's yet another issue that I'm not aware of which is causing the same problem. So I think I should be more conservative and ignore NAKs (as previous code did). Worst case is that SBAS setting won't change from its default, but won't prevent the GPS from working.
@mikeller am I on time to change this?

TonyBlit, 2020-05-25:
Minor correction after checking the code: I'm currently ignoring NAKs, so a badly calculated CRC won't stop the init. What is different from older versions is that I'm waiting for an ACK or a NAK without a timeout. This is a terrible idea, because if the RX line is noisy or if it's not even connected, the GPS won't work. I should have figured that for myself, sorry. :-(

mikeller, 2020-05-25:
@peteoz: Can you please test #9850?

peteoz, 2020-05-25:
> @peteoz: Can you please test #9850?

@mikeller sure, I have flashed the new test firmware, and hopefully I get to test it tomorrow (it's already past midnight over here)

TimCiarlette, 2020-05-25:
@mikeller I tried it on a stmF722 board and works good on bench, raining now or I'd go fly. #9850?

mikeller, 2020-05-25:
@peteoz: 

> ...it's already past midnight over here

Heh, based on the ...oz moniker in your name I had assumed until now that you are in Australia. :grinning:

peteoz, 2020-05-26:
> @peteoz:
> 
> > ...it's already past midnight over here
> 
> Heh, based on the ...oz moniker in your name I had assumed until now that you are in Australia. 😀

@mikeller normally yes, but I got stuck in Europe :) (since March) as my returned flight got cancelled and the airline will restart the service only in July

peteoz, 2020-05-26:
> @peteoz: Can you please test #9850?

@mikeller I tested today with the new target downloaded from https://github.com/betaflight/betaflight/pull/9850

this was only short session with 5 batteries, I was unable to replicate the issue so it's looking promising and every sat fix was fast.
I was in a built up area and couldn't fly far enough to retest GPS rescue, but that was anyway working fine in all of my tests last time on RC2 so I don't expect there to be any new issue.

sometimes later this week I'm planning to do some long range flights in the mountains, if I find any new issues I'll report back.

mikeller, 2020-05-26:
Thanks @peteoz. Hope you'll be able to find your way back to our side of the globe soon!

closed 2020-05-27 by mikeller

kingmot, 2020-10-06:
Hi, I have exactly the same problem with BetaFlight 4.2. Has anyone not been able to solve this problem yet?

kingmot, 2020-10-06:
[image]

The number of satellites is incorrect, how could the signal of 4 satellites be received while the number of satellites is 0?

TonyBlit, 2020-10-06:
If you don't have a 3D fix, nothing is expected to work. ie: on the OSD you will see the sat count blinking, but it won't show the coords, speed, etc.

JFly02, 2020-10-30:
Set gps_provider = NMEA 
And then it should work again, at least that's what worked for me.

PR #9850 (2020-05-27): Added timeout for Ublox commands
Fixes #9846 

Fixes an issue with a bad or not connected RX line to the GPS.
Possibly the cause of this issue: https://github.com/betaflight/betaflight/issues/9846