betaflight/betaflight #9847: new mode that control each motor separately

Nitzanbendavid, 2020-05-25:
Hi everyone,
I would like to add a new mode in my quad where I can control each motor separately, without the intervention of the flight controller. The reason is, that I want my quad also to drive on walls, and for this I need to have the option than only two motors will spin the other direction, while the other two motors are connected to the wheels and in charge of the steering and driving. (it means that two motors making thrust in the opposite direction in order to stick the quad to the wall, and the other two motors are driving the wheels.) Tottaly there are 4 motors.
The bottom line, how can I add a new mode, where I can control on each motor separate?

Thanks for helping,
Nitzan

mikeller, 2020-05-25:
@Nitzanbendavid: It is hard to understand what you actually want and how this should work. Can you please use the template that you were given when opening this feature request:

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

Nitzanbendavid, 2020-05-26:
the problem is that i want to have a new mode, besides the flying mode, for ground mode. in the ground mode i would like to control on each 2 motors separately, it means that i want that motors 1 and 3 will be controlled by the same channel, and motors 2 and 4 will be controlled on other channel. is it possible?
the solution that i thought about is to have a code in the betaflight that adding this new mode.

mikeller, 2020-05-26:
@Nitzanbendavid: I think iNav (https://github.com/iNavFlight/inav/) has got some support for rovers - you should look into this.

closed 2020-07-03 by stale[bot]