betaflight/betaflight #9851: Displaying altitude in dji for fc without baro

loutwice, 2020-05-26:
hi guys - 
not really a bug from betaflight but i found that dji use the information for altitude in MSP_ALTITUDE - and as you see it is only set when USE_BARO or USE_RANGEFINDER is defined.
Is it possible to add defined(USE_GPS) for fc without baro but gps or that is not a good choice?

case MSP_ALTITUDE:
#if defined(USE_BARO) || defined(USE_RANGEFINDER)
        sbufWriteU32(dst, getEstimatedAltitude());
#else
        sbufWriteU32(dst, 0);
#endif

thanks guys for the work btw

mikeller, 2020-05-26:
@loutwice: Thanks for the report. This should not be a big problem in 4.2 as 98% of the targets have `USE_BARO` and `USE_RANGEFINDER` defined.

But we can fix this.

loutwice, 2020-05-27:
@mikeller ok i understand, btw in the past fc with gps and USE_BARO but no physical baro had problem with altitude calc . i see that it was  managed to correct it forcing the source - brillant

PR #9864 (2020-06-02): Fixed altitude output in MSP.
Fixes #9851.
closed 2020-06-02 by mikeller