betaflight/betaflight #9852: CrazybeeF4 wont bind to my Taranis X9 lite

Piece0fPaper, 2020-05-27:
Hello, i saw a few threads that described kinda the problem that i have, but still found no solution 

**1. Describe your problem;**
I cannot bind my Taranis x9 Lite (EU LBT) to my crazybeeF4 FC (EU FrSky)

**2. Include ways to reproduce the problem;**

- updated the firmware to the latest one . CRAZYBEEF4FR (HAMO)

- try to bind throuhg CLI and with the BIND button on the FC itself.

- Set the reciever configuration in Betaflight of the FC to SPI RS support and FRSKY D16 mode -> FRSKY_X

- tried the bind_rx command followed by set frsky_x_rx_num=1 and also set the rx on the TX on 01.

3. Provide as much information as possible about your hardware and software, including:
- what hardware / software you are using;
Taranis x9 Lite EU LBT (internal module ISRM-N, 1.1.0/1.1.3. EU , External module OFF)
Crazybee F4 (with CRAZYBEEF4FR HARMO) also the latest one.
Betaflight 10.6.0

- how things are connected / wired up.
i changed nothing, everything is still in factory mode.

4. DIFF
``` 
# 

# diff all

# version
# Betaflight / STM32F411 (S411) 4.1.6 Apr 25 2020 / 05:11:39 (283bda8bf) MSP API: 1.42
# manufacturer_id: HAMO   board_name: CRAZYBEEF4FR   custom defaults: YES

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name CRAZYBEEF4FR
manufacturer_id HAMO
mcu_id 0044001a3437511635313736
signature 

# feature
feature -RX_PARALLEL_PWM

# master
set gyro_1_align_yaw = 900

profile 0

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

rateprofile 1

rateprofile 2

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
# 
``` 

when i try to bind through CLI or with the Button on the board, the lights stop flashing and are lighting permanent. 
it just looks like my TX wont find it...
i have this feeling that i made a mistake wiht all that EU and non EU things .

I hope you guys can help me out! 
I'd would like to put this little thing into air asap :)

Best Regards

MJ666, 2020-05-27:
You may take a look here and give it a try: https://github.com/betaflight/betaflight/pull/9697

Piece0fPaper, 2020-05-27:
thanks alot for the fast answer!

sorry i have to disappoint you, i have no idea how i implement htis code you sent me ...
so i have to use just another firmware on my frsky taranis x9 lite ?

mikeller, 2020-05-27:
@Piece0fPaper: See https://github.com/betaflight/betaflight/pull/9697#issuecomment-634907604

Closing this as duplicate of #8603.

closed 2020-05-27 by mikeller

Piece0fPaper, 2020-05-27:
Thanks alot!
fixed it with the solution from @MJ666