betaflight/betaflight #9853: scale value of current sensor betaflight 4.2 RC inconsitent

cusglider, 2020-05-27:
**Describe the bug**
Using a Matek F405 CTR i tried to recalibrate with betaflight 4.2. RC1/2 as the old value of 179 did produce incorrect results. However after each calibration run (recording flown mAh OSD value, recharging, mathematical recalculation of scale value) I get a new scale value and never reach a consisten value.

**To Reproduce**
1. fly 1000maH according to OSD
2. recharge battery and document recharged maH
3. calculate scale value

**Expected behavior**
I need to get a scale value, that stays stable and is not changing with every set of flights
A clear and concise description of what you expected to happen.

**Flight controller configuration**
``` 
# diff

# version
# Betaflight / STM32F405 (S405) 4.2.0 May 13 2020 / 14:26:41 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF405CTR, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

board_name MATEKF405CTR
manufacturer_id MTKS

# name: CUS Alien

# feature
feature -SOFTSERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 0 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
serial 3 64 115200 57600 0 115200

# aux
aux 0 0 1 1300 2100 0 0
aux 1 1 0 900 1300 0 0
aux 2 13 3 1300 2100 0 0

# master
set gyro_lowpass2_hz = 500
set dyn_lpf_gyro_min_hz = 400
set dyn_lpf_gyro_max_hz = 1000
set acc_calibration = -1,94,-199,1
set rssi_channel = 12
set rc_smoothing_auto_smoothness = 20
set serialrx_provider = CRSF
set motor_pwm_protocol = MULTISHOT
set failsafe_throttle = 1100
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set ibata_scale = 157
set auto_disarm_delay = 60
set deadband = 5
set yaw_deadband = 5
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_warn_rssi = ON
set osd_rssi_alarm = 70
set osd_cap_alarm = 1200
set osd_vbat_pos = 376
set osd_rssi_pos = 2458
set osd_tim_2_pos = 439
set osd_remaining_time_estimate_pos = 472
set osd_flymode_pos = 25
set osd_throttle_pos = 121
set osd_vtx_channel_pos = 32
set osd_crosshairs_pos = 2317
set osd_current_pos = 2508
set osd_mah_drawn_pos = 2496
set osd_craft_name_pos = 2094
set osd_altitude_pos = 2464
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 344
set osd_disarmed_pos = 2380
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_total_time = ON
set gyro_1_align_yaw = 2700
set gyro_rpm_notch_q = 800
set name = CUS Alien

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 140
set dyn_lpf_dterm_max_hz = 340
set dyn_lpf_dterm_curve_expo = 7
set dterm_lowpass2_hz = 300
set feedforward_transition = 40
set iterm_relax_cutoff = 10
set yaw_lowpass_hz = 70
set throttle_boost_cutoff = 10
set ff_interpolate_sp = AVERAGED_3
set ff_spike_limit = 55
set ff_smooth_factor = 40
set ff_boost = 0

rateprofile 0

# rateprofile 0
set roll_expo = 45
set pitch_expo = 45
set yaw_expo = 45

# end the command batch
```

```
# resource show all
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: SERIAL_TX 4
A01: SERIAL_RX 4
A02: FREE
A03: FREE
A04: FREE
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: LED 2
A15: FREE
B00: FREE
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: FREE
B09: LED 1
B10: OSD_CS
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE
C01: SDCARD_CS
C02: GYRO_CS 1
C03: GYRO_EXTI
C04: ADC_CURR
C05: ADC_BATT
C06: MOTOR 1
C07: MOTOR 2
C08: MOTOR 3
C09: MOTOR 4
C10: SERIAL_TX 3
C11: SERIAL_RX 3
C12: FREE
C13: BEEPER
C14: FREE
C15: FREE
D00: FREE
D01: FREE
D02: FREE
D03: FREE
D04: FREE
D05: FREE
D06: FREE
D07: FREE
D08: FREE
D09: FREE
D10: FREE
D11: FREE
D12: FREE
D13: FREE
D14: FREE
D15: FREE
E00: FREE
E01: FREE
E02: FREE
E03: FREE
E04: FREE
E05: FREE
E06: FREE
E07: FREE
E08: FREE
E09: FREE
E10: FREE
E11: FREE
E12: FREE
E13: FREE
E14: FREE
E15: FREE
F00: FREE
F01: FREE
F02: FREE
F03: FREE
F04: FREE
F05: FREE
F06: FREE
F07: FREE
F08: FREE
F09: FREE
F10: FREE
F11: FREE
F12: FREE
F13: FREE
F14: FREE
F15: FREE

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3:
    CH1: MOTOR 1
TIM4: FREE
TIM5: FREE
TIM6: FREE
TIM7: FREE
TIM8:
    CH2: MOTOR 2
    CH3: MOTOR 3
    CH4: MOTOR 4
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**
 - Flight controller Matek F405 CTR;
 - Other components [
RX TBS Crossfire Nano, 
VTX,  Caddx Vista for DJI FPV
Matek beeper
Betaflight 4.2 RC2 

 - how are the different components wired up.
directly soldered

**Additional context**
with Betaflight 4.1.6 the scale value of 179 was consitant and did not produce different readings

alfag4, 2020-06-13:
Hello, same issue with Matek 722SE

Enurian, 2020-07-11:
Same problem here. LUX F7 Ultimate (target is MATEKF722SE) with an Airbot Furling32 4in1 ESC.
Fuel is inconsistent and it started with BF 4.2.0.

One flight it showed me 671mah consumed and I loaded 1200mah back in.
Another one it was like 840mah consumed and 900mah back in, so the readings are just wrong.

However, the displayed value of the latest current value seems to be normal. It looks like just the accumulation is off.

CapVille, 2020-09-02:
I am having the same problem on HGLRC F722 and the Aikon F7 mini.

kreidler, 2020-09-03:
> I am having the same problem on HGLRC F722 and the Aikon F7 mini.
Which ESC do you use. MATEK F405 CTR do have a current sensor on the FC. Your FCs do not have any sensor onboard and need the data from the ESC.

cusglider, 2020-09-03:
Matek F4 and F7 have a current sensor on the FC, just check the Matek documentation.

CapVille, 2020-09-03:
> > I am having the same problem on HGLRC F722 and the Aikon F7 mini.
> 
> Which ESC do you use?
> MATEK F405 CTR do have a current sensor on the FC. Your FCs do not have any sensor onboard and need the data from the ESC.

For the HGLRC is the HGLRC Fordward 45A and for the Aikon is the Aikon Race Dragon RD32 4-in-1 45A

kreidler, 2020-09-03:
> For the HGLRC is the HGLRC Fordward 45A and for the Aikon is the Aikon Race Dragon RD32 4-in-1 45A

HGLRC Forward 45A ESC does not contain current sensor. So, inconsistent data normal and no BF issue.
Aikon issue can be caused by many reasons. First test other BF releases to exclude actual release and open a new issue if still needed.

CapVille, 2020-09-03:
> HGLRC Forward 45A ESC does not contain current sensor. So, inconsistent data normal and no BF issue.
> Aikon issue can be caused by many reasons. First test other BF releases to exclude actual release [...]

Thank you. I didn't check for the HGLRC sorry.