betaflight/betaflight #9855: i2c gyro mpu6050 high cpu load F4

Sekilsgs2, 2020-05-28:
**Describe the bug**

I'm using stm32f405 with i2c mpu6050 - build with config ALIENFLIGHT4
In configurator i'm change pid freq to 4khz.
RX - is spi nrf2402 - v202 protocol
With this i'm getting 70% cpu loading

```
Task list             rate/hz  max/us  avg/us maxload avgload  total/ms
00 - (         SYSTEM)      9       2       1    0.5%    0.5%         1
01 - (         SYSTEM)    916       7       0    1.1%    0.0%       103
02 - (           GYRO)   6410     122     109   78.7%   70.3%    185881
03 - (         FILTER)   3205      10       3    3.7%    1.4%      2468
04 - (            PID)   3195      48      31   15.8%   10.4%     27036
05 - (            ACC)    915   21717     109 1987.6%   10.4%     26653
06 - (       ATTITUDE)     97      14       8    0.6%    0.5%       301
07 - (             RX)     29      58      30    0.6%    0.5%       268
08 - (         SERIAL)     97  481006      76 4666.2%    1.2%      1289
09 - (       DISPATCH)    917       7       0    1.1%    0.0%       138
10 - (BATTERY_VOLTAGE)     47       4       2    0.5%    0.5%        20
11 - (BATTERY_CURRENT)     47       7       1    0.5%    0.5%        17
12 - ( BATTERY_ALERTS)      5       4       3    0.5%    0.5%         3
13 - (         BEEPER)     97       9       2    0.5%    0.5%        32
14 - (            GPS)     97      73      12    1.2%    0.6%       322
17 - (       ALTITUDE)     36       7       2    0.5%    0.5%        13
23 - (            CMS)     17       3       1    0.5%    0.5%         6
24 - (        VTXCTRL)      5       2       1    0.5%    0.5%         0
25 - (        CAMCTRL)      5       2       0    0.5%    0.0%         0
27 - (    ADCINTERNAL)      1       2       2    0.5%    0.5%         0
28 - (       PINIOBOX)     17       2       1    0.5%    0.5%         3
RX Check Function                  10       6                        47
```

```
MCU F40X Clock=168MHz (PLLP-HSE), Vref=2.91V, Core temp=55degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3692, Max available config: 16384
Gyros detected: gyro 1
GYRO=MPU6050, ACC=MPU6050.n
OSD: NONE
System Uptime: 791 seconds, Current Time: 2020-05-28T12:45:54.171+00:00
CPU:64%, cycle time: 126, GYRO rate: 7936, RX rate: 33, System rate: 9
Voltage: 675 * 0.01V (2S battery - WARNING)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP GPS
```

**To Reproduce**

**Expected behavior**

**Flight controller configuration**
```
# version
# Betaflight / ALIENFLIGHTF4 (AFF4) 4.2.0 May 28 2020 / 13:38:07 (norevision) MSP API: 1.43

# start the command batch
batch start

board_name ALIENFLIGHTF4

# mixer
mixer QUADX1234

mmix reset

mmix 0  1.000 -0.414  1.000 -1.000
mmix 1  1.000 -0.414 -1.000  1.000
mmix 2  1.000  0.414  1.000  1.000
mmix 3  1.000  0.414 -1.000 -1.000
mmix 4  1.000 -1.000 -0.414 -1.000
mmix 5  1.000  1.000 -0.414  1.000
mmix 6  1.000 -1.000  0.414  1.000
mmix 7  1.000  1.000  0.414 -1.000

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature -DYNAMIC_FILTER
feature GPS
feature RX_SPI

# map
map AETR1234

# serial
serial 1 2 115200 9600 0 115200

# aux
aux 0 2 6 900 2100 1 0
aux 1 2 0 900 1100 1 0
aux 2 47 0 1500 2100 0 0

# master
set acc_calibration = 52,-8,63,1
set mag_hardware = NONE
set baro_hardware = NONE
set mid_rc = 1498
set min_check = 1101
set max_check = 1898
set rx_spi_protocol = V202_1M
set min_throttle = 1070
set use_unsynced_pwm = OFF
set failsafe_procedure = GPS-RESCUE
set battery_meter = ADC
set vbat_scale = 58
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set pid_process_denom = 2
set enable_stick_arming = ON
set gyro_1_bustype = I2C
set gyro_1_i2cBus = 1
set gyro_1_sensor_align = CW180
set gyro_1_align_yaw = 1800

profile 0

# profile 0
set p_pitch = 180
set p_roll = 180
set p_yaw = 180

rateprofile 0

# rateprofile 0
set roll_rc_rate = 30
set pitch_rc_rate = 30
set yaw_rc_rate = 60

# end the command batch
batch end
```

```
A00: MOTOR 3
A01: MOTOR 4
A02: SERIAL_TX 2
A03: SERIAL_RX 2
A04: RX_SPI_CS
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: SWD
A14: SWD
A15: FREE
B00: RX_SPI_CS 1
B01: FREE
B02: FREE
B03: SPI_SCK 3
B04: SPI_MISO 3
B05: SPI_MOSI 3
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: MOTOR 1
B09: MOTOR 2
B10: FREE
B11: FREE
B12: FREE
B13: SPI_SCK 2
B14: FREE
B15: FREE
C00: ADC_BATT
C01: FREE
C02: SPI_MISO 2
C03: SPI_MOSI 2
C04: FREE
C05: FREE
C06: FREE
C07: FREE
C08: FREE
C09: FREE
C10: FREE
C11: FREE
C12: LED 1
C13: BEEPER
C14: FREE
C15: INVERTER 2
D02: LED 2

Currently active Timers:
-----------------------
TIM1: FREE
TIM2: FREE
TIM3: FREE
TIM4:
    CH3: MOTOR 1
    CH4: MOTOR 2
TIM5:
    CH1: MOTOR 3
    CH2: MOTOR 4
TIM6: FREE
TIM7: FREE
TIM8: FREE
TIM9: FREE
TIM10: FREE
TIM11: FREE
TIM12: FREE
TIM13: FREE
TIM14: FREE

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: FREE
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: ADC 1
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: FREE
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE
```

**Setup / Versions**

**Additional context**

mikeller, 2020-05-28:
@Sekilsgs2: This is expected - the I2C gyro implementation is busy waiting for I2C responses, soaking up huge amounts of CPU times - this is not a problem since there are no officially supported boards with I2C gyro. Switch to a board with an SPI gyro.

Sekilsgs2, 2020-05-28:
Ok thank

closed 2020-05-28 by Sekilsgs2

jflyper, 2020-05-28:
@Sekilsgs2 
Yes, high CPU load with I2C connected acc/gyro chip is the expected behavior.

Betaflight I2C bus driver is based on busy waiting. That is, while a bus transaction (i.e. gyro data read) is active, CPU is waiting for the transaction to finish.

Now, with I2C bus running at 800KHz, it will take at least 9 [bits/byte] * 6 [byte] * 1/800 [KHz] = 67.5 [us] to read 6 byte gyro data off the chip.

Since 4.2 reads gyro at it's native speed, MPU6050 is read at 8KHz.
This means gyro data is read every 125 [us] of which 67.5[us] (54%) is used waiting for I2C read to finish.

Add up other task usages and you get 70%.

Sekilsgs2, 2020-05-28:
But with spi how cpu load?

jflyper, 2020-05-29:
You mean the typical CPU load of F405 based boards with SPI gyro chip?

It would be under 10% with 4K pid loop and bidir DSHOT300.

Sekilsgs2, 2020-05-29:
Yes i mean this. Thank.
But if using async reading from i2c with interrupt - maybe this better way ?

mikeller, 2020-05-29:
@Sekilsgs2: Due to the slower speed of the I2C bus gyro reading over it will always be slower, which is why support for I2C gyros was deprecated when Betaflight moved to the faster loop speeds used today, and the I2C driver code has not received a lot of attention since. But feel free to open a pull request to improve it, and we'll be considering it.