betaflight/betaflight #9856: [4.2.0 RC2] Truncated BBL header

0crap, 2020-05-28:
It seems I'm getting some BBL logs which does not contain the complete header.
I know this has been fixed in the past.
Maybe the header is creeping and a bit bigger now, might we run into this issue again?

Problem is intermitted, half of my logfiles are perfectly fine.

**Flight controller configuration**
```
# version
# Betaflight / STM32F7X2 (S7X2) 4.2.0 May 13 2020 / 14:27:44 (6aeac3caa) MSP API: 1.43
# config: manufacturer_id: MTKS, board_name: MATEKF722, version: f173f9bc, date: 2019-10-02T14:43:13Z

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722
manufacturer_id MTKS
mcu_id 00<cut>
signature 

# name: REVERB5

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# feature
feature -AIRMODE
feature -DYNAMIC_FILTER
feature GPS
feature LED_STRIP

# beeper
beeper -ON_USB

# serial
serial 0 2 115200 57600 0 115200
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led
led 0 11,15::AB:0
led 1 11,14::AB:0
led 2 4,14::AB:0
led 3 4,15::AB:0

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 1 900 1100 0 0
aux 2 46 3 1900 2100 0 0
aux 3 13 2 1900 2100 0 0
aux 4 28 1 1400 2100 0 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 3
vtxtable powervalues 25 200 400
vtxtable powerlabels 25 200 400

# rxfail
rxfail 3 h
rxfail 5 s 1000

# master
set gyro_lowpass_hz = 0
set gyro_lowpass2_hz = 350
set gyro_notch1_hz = 235
set gyro_notch1_cutoff = 225
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -6,20,10,1
set baro_hardware = NONE
set min_check = 1015
set max_check = 1990
set serialrx_provider = CRSF
set crsf_use_rx_snr = ON
set dshot_burst = OFF
set dshot_bidir = ON
set dshot_bitbang = OFF
set motor_pwm_protocol = DSHOT300
set vbat_max_cell_voltage = 420
set vbat_warning_cell_voltage = 340
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = AUTO
set gps_ublox_use_galileo = ON
set gps_ublox_mode = PEDESTRIAN
set gps_set_home_point_once = ON
set gps_rescue_angle = 50
set gps_rescue_initial_alt = 20
set gps_rescue_ground_speed = 1600
set gps_rescue_min_sats = 6
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_alt_mode = FIXED_ALT
set yaw_deadband = 4
set pid_process_denom = 2
set telemetry_disabled_acc_x = ON
set telemetry_disabled_acc_y = ON
set telemetry_disabled_acc_z = ON
set telemetry_disabled_pitch = ON
set telemetry_disabled_roll = ON
set telemetry_disabled_heading = ON
set telemetry_disabled_altitude = ON
set telemetry_disabled_vario = ON
set telemetry_disabled_ground_speed = ON
set osd_warn_no_gps_rescue = OFF
set osd_rssi_alarm = 45
set osd_link_quality_alarm = 70
set osd_cap_alarm = 1200
set osd_link_quality_pos = 6517
set osd_rssi_dbm_pos = 6485
set osd_tim_2_pos = 6548
set osd_throttle_pos = 2082
set osd_current_pos = 6500
set osd_mah_drawn_pos = 6540
set osd_gps_speed_pos = 4182
set osd_gps_lon_pos = 112
set osd_gps_lat_pos = 100
set osd_gps_sats_pos = 4147
set osd_home_dir_pos = 4139
set osd_home_dist_pos = 4141
set osd_altitude_pos = 4132
set osd_debug_pos = 353
set osd_warnings_pos = 14569
set osd_avg_cell_voltage_pos = 6533
set osd_core_temp_pos = 321
set osd_stat_tim_1 = ON
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_min_link_quality = ON
set osd_stat_min_rssi_dbm = ON
set osd_profile = 2
set osd_gps_sats_show_hdop = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_freq = 5732
set vcd_video_system = NTSC
set camera_control_key_delay = 120
set gyro_1_align_yaw = 1800
set gyro_rpm_notch_q = 700
set name = REVERB5

profile 0

# profile 0
set dyn_lpf_dterm_min_hz = 91
set dyn_lpf_dterm_max_hz = 221
set dterm_lowpass2_hz = 195
set vbat_sag_compensation = 100
set feedforward_transition = 10
set iterm_relax_cutoff = 10
set p_pitch = 50
set f_pitch = 180
set p_roll = 44
set f_roll = 180
set f_yaw = 100
set d_min_roll = 0
set d_min_pitch = 0
set d_min_advance = 0
set ff_max_rate_limit = 120
set idle_min_rpm = 24

profile 1

profile 2

# restore original profile selection
profile 0

rateprofile 0

# rateprofile 0
set rates_type = QUICK
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 50
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 60
set pitch_srate = 61
set yaw_srate = 45

rateprofile 1

# rateprofile 1
set rates_type = QUICK
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 50
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set pitch_srate = 71
set yaw_srate = 45

rateprofile 2

# rateprofile 2
set rates_type = QUICK
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 50
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 80
set pitch_srate = 81
set yaw_srate = 45

rateprofile 3

rateprofile 4

rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

0crap, 2020-05-28:
[LOG00114.BFL.txt](https://github.com/betaflight/betaflight/files/4696121/LOG00114.BFL.txt)

McGiverGim, 2020-05-28:
The original PR that fixed that was: https://github.com/betaflight/betaflight/pull/8284

Maybe it needs a little modification incrementing the cache size.

etracer65, 2020-05-28:
@0crap Can you please test #9857

Note that it requires having GPS enabled to make the header big enough to cause the problem. The fix is an improvement to the way the logs are written rather than simply increasing the cache (which is not sustainable).

0crap, 2020-05-29:
Awesome, I'll do!

And I think you are spot on with the GPS remark.
I have a similar quad, but without GPS, also on 4.2.0 RC2 and NOT showing this issue.

0crap, 2020-05-29:
I'm sorry to say, but I did 4 packs today of which 2 are truncated.
See attached.
[LOG00118.BFL.txt](https://github.com/betaflight/betaflight/files/4703700/LOG00118.BFL.txt)
[LOG00119.BFL.txt](https://github.com/betaflight/betaflight/files/4703701/LOG00119.BFL.txt)

PR #9857 (2020-06-02): Blackbox flush SD card sector cache after writing header and before logging starts
Fixes #9856 

The normal "flush" for SD card only queues a cache sector for writing and the actual sync to the media happens asynchronously. During this period the cache entry is not available until the write completes sometime later. So as the blackbox header fields were written they end up consuming the majority of the cache. A "flush" was made before actual logging starts, but the async writes were not completing fast enough to ensure available cache sectors for the actual logging. This resulted in the cache  [... cut]
closed 2020-06-02 by mikeller