betaflight/betaflight #9858: Cell voltage in summary at end of flight

Kowitzki, 2020-05-29:
I think the standard for displaying the battery voltage should be the cell voltage and not the voltage of the pack. Whoops use 1s, 2s, 3s and 5inch quads use 4s, 5s, and 6s. I wont remember every state of charge of 6 different lipo packs. 
In the OSD there is an option to display the average cell voltage. Unfortunately this option is not available for the summary after the flight.

**Request:**
Please make an option for the summary to display the minimum cell voltage and the current cell voltage.

0crap, 2020-05-30:
I agree, pack voltage means nothing to me and I don't have a feel for it.
Average cell voltage tells me everything to know without even thinking. :-)

dkustec, 2020-06-01:
I done this before, so I can make the pull request.
It's easy to configure via CLI flag "osd_stat_show_cell_value". If it's "ON", average cell is showed at the stats screen. "OFF" is default and it's showing the batt value.

[image]
[image]

PR #9878 (2021-02-17): Added option to show cell value instead of batt value at stats screen
According to issue [#9858](https://github.com/betaflight/betaflight/issues/9858).
After-flight stats for a cell instead of the stats of a whole battery are more appropriate for some users (including me :).
Here is the feature to enable it. All the user needs is to enable this flag:
`osd_stat_avg_cell_value = ON`

Fixes: #9858

| `Variable`                                    | Description/Units                                                                                                                                       [... cut]
closed 2021-02-17 by mikeller